mu-editor/mu #249: It isn't easy to get the latest version of Mu on a Raspberry Pi

jaustin, 2017-02-14:
Hi,

(forking the discussion about installation at #185 into a separate issue)

Currently teachers or group leaders who want to use Mu on Windows, Mac OSX or non-ARM Linux can get binaries that 'just work' for Mu very easily.

For people using the Pi, it's a bit more complicated.

It would be really nice if we could solve this problem. Given that for a technical user the 'normal' install steps using the command line are acceptable, this issue is to track a much less technical solution - for example a single binary that can be downloaded or distributed via a USB stick that will simply 'just run' when offline, or something that can be copied onto the 'boot' partition of a standard RPi installation and launched from there.

I think the 'gold standard' would be having a version of Mu installed in standard RPi images that knows how to update itself, either via apt and a well-managed Mu-specific repository, or via an independent update mechanism, though this needs to be balanced agains a desire for stability in what is default in that distribution!

ntoll, 2017-02-14:
We have this in hand. When we next release Mu I'll be in touch with the Raspbian maintainers who'll create an apt-getable version. That will meet all the requirements you set out above (updates etc...).

gbaman, 2017-09-19:
@ntoll any update on getting a more up to date version of Mu available on Raspbian?

ntoll, 2017-09-19:
Yes. It'll be available *very soon*. We expect to release version 1.0 of Mu on all platforms (including Raspbian) at PyCon UK. ;-)

yonghuming, 2018-03-09:
@ntoll is raspbian apt-getable version ok? any docs?

rhubarbdog, 2018-05-18:
Hi 
these instructions which i put on this ubuntu site will work on raspberry pi running a raspbian linux operating system. both are debian.
 https://ubuntuforums.org/showthread.php?t=2377314

ntoll, 2018-06-04:
RPi have responsibility for packaging Mu. They update when we make a release. Should be just `apt-get install mu-editor` from now on. Closing.

closed 2018-06-04 by ntoll