mu-editor/mu #253: Error checking in mu - request

Isioviel, 2017-02-28:
Hi,

Firstly, sorry if my terminology is a bit off / what I'm asking makes no sense, I'm new to all this :).

I have been using mu with students / families in clubs - in general I love it, but have been struggling with the whitespace/new line errors - having to fix all of these errors is making it difficult for them to find 'real' errors in their code.

E.g. they are struggling with the idea of indentation, and how important it is that each line is indented the same amount - when they are told they have a general whitespace error they try to fix it by putting in random spaces in the wrong place, making their code worse.  I can't get around them all quick enough to explain before they've got themselves extremely mixed up, and they can't process the importance of correct indentation, because it isn't treated any differently in mu to having new lines between functions for example.

Basically my request is to have the option to turn off the formatting errors in mu - I understand their purpose, and by following the rules their code is much easier to read, but it makes my job a lot harder as I spend loads of time helping them to fix these errors, when I could be explaining programming concepts, and they're getting a bit frustrated.

If the whitespace errors didn't show, using the 'check' button would be much easier for them, as they would (hopefully) have less errors to correct, and then I can help them with formatting at appropriate times instead of it being forced on us.

I hope that makes sense, thanks for all the work that goes into mu, it's a wonderful program!

ZanderBrown, 2017-02-28:
Err if mu's complaining about whitespace the program probably isn't valid Python

ntoll, 2017-02-28:
Hi @Isioviel,

Thank you for your feedback, you make some great points and I had a sneaking suspicion these sorts of things would happen.

Briefly, the "check" button uses flake and PEP8 to provide the feedback you see annotated to the code. These utilities can be configured and your feedback is *invaluable* in that you're someone providing real-world evidence of ways in which beginner programmers get caught out by this sort of thing.

Right now, all errors (no matter how trivial) are exposed, and some don't apply to MicroPython anyway (such as the newline at the end of the file). Some of the "errors" are, in fact, style guides for writing clean code in Python.

I wonder if we should allow people to change the level of error checking (I'd need to work out how to expose this to users in an easy to use / configure manner).

All feedback about this would be *really helpful*. Keep it coming!

N.

Isioviel, 2017-02-28:
Thanks, that's exactly it!

I understand that good style is important, but for beginners, just writing working code is difficult enough!
Having to correct errors that python doesn't care about it is making it harder to explain the *real* errors that will stop their code working when they flash it to the micro:bit - especially when they start ignoring the error messages when they realise their code still works (and stop trusting the 'check' button), and so can't debug their own code when it fails.

Being able to choose the level of error checking, would be really helpful - I could turn it off with the beginners, then as they get better turn it on to encourage well written / styled code.
I am not sure on the best way to use this (I love the clean look / lack of options in mu as it stops students accidentally/on purpose changing things), but would be happy to be a test subject!

ZanderBrown, 2017-02-28:
I think we are increasingly in need of a settings dialog

BillyLundh, 2017-06-26:
I'm glad I found this and will try to run the scripts anyway with the error messages. Since some don't apply to microPython. But I spent a lot of time until founding this trying to resolve it.
I think it would be better just to remove these then and only display errors if it won't run on the microBit in the meantime otherwise a settings options would be great. Another suggestion is to have auto formatting, a short command like in the Arduino IDE (cmd + a). To tab everything right. Also in order to get serial working in terminal I have to disconnect and connect the microBit again, otherwise the device will be busy and the REPL in the IDE doesn't work. I like the microBit better compared to Arduino plattform because of The python possibility but it's not easy to get started with.

ntoll, 2018-06-04:
OK... so the checker has been "toned down" when it comes to un-helpful whitespace warnings. Marking this as closed.

closed 2018-06-04 by ntoll

supuflounder, 2020-05-21:
I have a simple program I am translating from C++ to Python.  In it, there is a table telling me how to map bit positions for the transform I'm doing.  It looks like this:
[image]

(If I just copy and paste the code into a "code" environment, it gets reformatted into a single long line, so I have to show the screen image)
When I do the "Check", I get the following output:
[image]

I read the Python beginners' tutorial, and I saw spaces before the first element in a tuple, so I don't understand why I am getting an error message; it looks like mu is whimsically complaining about my style.  I tried deleting the space, which put two spaces after the =, and it complained that I had too many spaces after an operator.  This does not seem to conform to the Python syntax as I understood it.  Furthermore, it is suggested in the documentation that no explicit operating is needed when splitting tuples across multiple lines.  I tried adding a "\" at the end of the line, and mu told me that was redundant; yet it complains that the lines are "UNDER-indented" (my emphasis added) even though they are indented a fair amount, and in any case, whitespace between values (including newline) is not specified in the Python grammar.  So it is hard to tell how much of this is actual Python requirements, and how much is the check code used by mu is just trying to tell me how to make my program "look pretty".  I would like to turn off all checking that is not detecting genuine Python source code errors.