mu-editor/mu #256: Flake issues are hidden by PEP8 issues

tjguk, 2017-03-01:
Because the Flake and PEP8 issue markers are applied one after the other in logic:check_code any PEP8 issues for a line will overwrite any Flake issues already applied for the same line.

To see this in action:
* Create a new file
* Enter the word "blah" (or some other nonsense)
* Press the [Check] button

The error which is indicated complains that there is no newline at the end of the file, and totally ignores that NameError!

ntoll, 2017-03-01:
Ooh... good catch! :-)

moshez, 2018-05-28:
Does not reproduce for me.

[image]

This is running on Linux, directly from master, git revision 6c6921f0fec3f9d36cf9a8e022fe254d7da6c83e

ntoll, 2018-06-04:
This has been fixed. Closing. Thanks for the report and check. :-)

closed 2018-06-04 by ntoll