mu-editor/mu #258: Remove comments before flashing program

webmaster442, 2017-03-25:
Comments should be removed before flashing program to the microbit, to save space.

ZanderBrown, 2017-03-25:
This has been brought up before.

Is it something you could add in a PR?

carlosperate, 2017-06-22:
I'd be against removing comments unless is absolutely necessary, and even when doing so there should be a warning to the user of what's about to happen. This is because the script can be recovered from the .hex file output.
This is also discussed in https://github.com/mu-editor/mu/issues/192.

kamar535, 2017-09-25:
Just got hit by this when developing an assignment for my students. In the source I've have plenty of comments for documenting the API's of functions etc.

rhubarbdog, 2017-11-25:
write a sed script to rip comments out `sed 's/\( *#.*\)//' my-file.py > comment-free.py`

kamar535, 2017-11-25:
@rhubarbdog The comments serves a purpose during reading/editing the source and I would prefer if Mu had an option to remove the comments prior to flashing the program to the Microbit.  Using an external tool, for example sed, to remove the comments before loading the file in Mu is not an ideal workflow for (novice) students.

carlosperate, 2018-04-18:
The master branch now offers the option in the settings to minify the micro:bit Python code (added in https://github.com/mu-editor/mu/commit/bf8bb7586e37f51e3f7434f17bb56e996eaf1f28), and will be included in the next release of Mu:
[image]
I'll close this issue, but please feel free to reopen if you have any questions or suggestions.

closed 2018-04-18 by carlosperate

kamar535, 2018-04-18:
This is great!