mu-editor/mu #259: How do we import modules stored in the filing system, successfully?

DavidWhaleMEF, 2017-03-26:
I'm struggling to get modules stored in the file system to work, due to what looks like a catch-22 user interface flow problem.

I want a main script in the .hex file that gets flashed when I drag and drop the .hex file (using the 8K application region). I then want some python modules in the filing system, that are imported by the application script.

When I copy the .hex file to the micro:bit it erases the whole memory, so my modules are then deleted. It then tries to auto run the script, which runs the import command, and because the flash has just been erased, by module is not present and I get a long scrolling message on the micro:bit display telling me as such.

I then press the FILES button to try to upload the dependent module to the filing system, the micro:bit seems to reset and re-run my main script again, and then I get a long scrolling message on the screen saying the same again, and Mu crashes with an unexpected error.

How do I get python modules into the filing system and then load a main application script into the standard 8K region, without falling into this trap?

I specifically want my modules in the filing system, as they are bigger than 8K, so my application has to be in the fixed 8K region and my modules in the filing system.

So, a reproducible test case is:

mod1.py has this and I want it in the filing system:
print("imported mod1")

app.py has this and I want it in the fixed 8K app region:
import mod1
print("app loaded")

So, what sequence of steps do I follow to successfully get mod1.py into the filing system, and app.py into the main 8K application region without Mu crashing along the way - or is this impossible?

carlosperate, 2017-11-03:
This article from @jezdean https://microbit-playground.co.uk/howto/add-python-module-microbit-micropython does a good job listing the steps to add files. We need to add something similar to the Mu documentation and see if there is anything we can do to make it a bit easier.

ntoll, 2018-06-04:
Hmm... this isn't a Mu problem. Rather, it's more to do with MicroPython on the micro:bit. I agree we should document this somewhere, but my advice for now is to flash a blank hex and use the convention of using `main.py` to run the script on start-up instead.

closed 2018-06-04 by ntoll

carlosperate, 2018-06-05:
Also, now that MicroPython can stop scrollable error messages via REPL Ctrl+C it means the files pane can be opened and used inmediately after flashing a user script with a missing import.
This used to be a huge pain before, as microFS coudn't connect to the REPL until the error message finished scrolling on the micro:bit.