mu-editor/mu #261: Microbit Micropython version is a year old?

plummet555, 2017-04-17:
Hi

I've been using Mu (mu-2017-01-29_15_38_04.exe for Windows) for MicroBit development, but the version of MicroPython embedded in it seems quite old:

MicroPython v1.7-9-gbe020eb on 2016-04-18; micro:bit with nRF51822

so it doesn't support some of the newer documented APIs such as setting GPIO pin modes.

Is there a reason why the version is so old? I'm wondering if it is because there don't actually seem to be any official release tags for MicroPython on MicroBit?

https://github.com/bbcmicrobit/micropython

Thanks

ZanderBrown, 2017-04-17:
MicroPython in Mu is provided by uFlash.
The bundled uFlash is from October however uFlash was last updated in Febuary

However why MicroPython is reporting being that old i don't know

plummet555, 2017-04-17:
Hi - that's helpful, thanks for the quick reply.

I see there is an issue raised against uFlash to request a newer MicroBit version and there are some comments there about version number display too:

https://github.com/ntoll/uflash/issues/34

So I guess we can wait for that - although I think a useful new feature for Mu would be for it to allow a different .hex file to be specified? This could be (for example) a command-line option: mu.exe [-hex <myhexfile>]

ZanderBrown, 2017-04-17:
@plummet555 that's definitely something worth considering

carlosperate, 2017-04-17:
Just a quick note that the version listed in the REPL is the upstream MicroPython version that the micro:bit MicroPython is based on.

tim1mw, 2017-04-19:
ZanderBrown: As far as I can tell the micopython Hex in uflash hasn't been updated since last october, the subsequent updates to uflash are unrelated changes.

I've done my own fork of uflash with the latest micopython hex here: https://github.com/tim1mw/uflash , however, there are some bugs in micropython code relating to the pins. I'll probably update the fork again once the bug is fixed

Even though I know that this is using the latest micropython code, the version still shows as "v1.7-9-gbe020eb on 2016-04-18". That could simply reflect the last "official" release version and it will only change when a new official release is made.

The '-hex' command line option is a nice idea. Looking at the code, it would be a simple task to add it to uflash (not sure about mu). However, we need to persuade the microbit micropython developers to put up regular hex builds for this to be of much use, I couldn't find a download site (if there is one, please point me to it). I did my own build of the micropython hex file, getting the correct toolchain installed to achieve this took forever.

ntoll, 2017-04-19:
uflash is only updated when there's a "stable" release. One will be forthcoming soon-ish. 

However, if you need an updated hex file the documentation shows how uflash can make this work for you:

https://uflash.readthedocs.io/en/latest/#command-usage

Specifically, use the `-r` flag. That's all you need.

Out of interest, why DON'T people use the `-r` flag? It's far simpler than forking and hacking your own version.

plummet555, 2017-04-20:
Hi

I notice here:

https://github.com/mu-editor/mu/blob/master/mu/logic.py#L394

that Mu instructs uFlash to use its current bundled version of the Micropython runtime like this:

micropython_hex = uflash.embed_hex(uflash._RUNTIME, python_hex)

where uflash._RUNTIME is a huge string consisting of the hex.

So, I think it would be fairly easy for Mu to optionally pass in a big string that is read from a file instead.

If I got that right, then supporting a '-hex <filename>' option would be a fairly straightforward change to the Mu code and would require no changes to uFlash.

carlosperate, 2017-04-20:
Yeah, that could definitely be an option. Perhaps an extra entry for the settings json file?

plummet555, 2017-04-20:
I agree - settings.json looks like a better option than a command-line parameter.

plummet555, 2017-04-20:
..I could have a bash at that change and contribute it if that's of interest.

ntoll, 2017-04-21:
Knock yourself out! All contributions most welcome around here... :-)

plummet555, 2017-04-21:
OK!. Here's the code:

https://github.com/plummet555/mu/commit/f33fbc32d35d8370985871e947b532cd5d9b6c18

I haven't submitted a pull request yet, probably need to think about documentation and whether a test needs to be added. Comments welcome though

ntoll, 2017-04-21:
Just follow the checklist in CONTRIBUTING.rst (https://github.com/mu-editor/mu/blob/master/CONTRIBUTING.rst).

plummet555, 2017-04-22:
Thanks. Done & pull request submitted.

carlosperate, 2017-11-03:
Fixed (a while ago) on https://github.com/mu-editor/mu/pull/264, thanks @plummet555!

closed 2017-11-03 by carlosperate