mu-editor/mu #262: Pop up help for radio.init() shows spi.init() help

DavidWhaleMEF, 2017-04-19:
Mu version: https://codewith.mu/help/0.9.13/

When I use the pop up help in Mu on the radio module, the init() incorrectly shows the help for the spi.init() method

[image]

DavidWhaleMEF, 2017-04-19:
Clearly I should be using radio.config()

But I didn't understand why it offered me spi help for the radio module in the first place?

ntoll, 2017-04-19:
We've known about this for a while, and when I investigated it found that it is a limit of the underlying Scintilla widget. Without re-writing it all from scratch there's nothing we can really do about it. :-(

closed 2017-04-19 by ntoll