mu-editor/mu #263: REPL button doesn't work on Windows, and no useful help comes up from pressing HELP

DavidWhaleMEF, 2017-04-20:
We have noticed that installing Mu (0.9.13) on a fresh Windows (10) laptop, the REPL button pops up a message saying that it could not find a BBC micro:bit and you have to flash MicroPython to it before the REPL will work.

However, the micro:bit has been detected, because we just flashed a program to it. So this is very confusing messaging.

We then press on HELP and it takes us to a page that does not give any help as to how to resolve this situation, or indeed what it all means and why it has happened.

The cause is that on windows the mbed serial driver has to be installed before the REPL will work - but it is not clear from the messages that come up in Mu that that is the case.

We have a well tested k-base article about installing the mbed serial driver here:

https://support.microbit.org/solution/articles/19000022103-how-do-i-use-the-serial-port-with-a-micro-bit-on-windows

I think we could make this workflow much more enjoyable for users with a few small tweaks

1) When that error message comes up, can you perhaps add some additional messaging about making sure that on windows the mbed serial driver is installed (and suggesting you press the HELP button to find out how?)

2) If the user is using Mu, they are probably using it because they are offline - in which case, pressing the HELP button and going to a web page won't be very helpful. So we need to think about the workflow there. Although clearly users have recently come from codewith.mu so presumably they have internet connectivity at the moment.

3) Can we have some text somewhere on the HELP page that explains about installing the mbed serial driver on windows - or at a minimum, explain what it is and why they need it and then link through to our k-base article?

We get a lot of tickets logged from users struggling with the existing workflow, it would be much nicer I think if we could signpost people at the point of the problem, rather than them either giving in, or having to specifically ask for help to solve what is a normal workflow.

I'm happy to accept alternative workflows that you might have ideas on, the above is just a suggestion. But if we can tidy this workflow up, it will make Mu a much better first experience for many users.

Thanks!

ntoll, 2017-04-20:
I hear you...

Right next to the download button on codewith.mu are instructions for how to install the driver.

A better fix would be for Windows to actually work properly. But we have no control over that. :-(

DavidWhaleMEF, 2017-04-20:
True. Although to be fair, it's nothing about windows working properly. mbed serial driver is a third party driver and there is no obligation on Microsoft to install all possible third party drivers that users may or may not need as part of the standard system image.

However, based on real user research, I observed a load of users installing Mu, and none of them saw that link, as they are pressing download and wanting to get going quickly. When they experience the actual error much later on, they are not on the download page so that link is not of any use to them and they don't even clock that it is there.

We've seen some nice work surface recently in some of the mobile apps, where when an error condition occurs due to some user mis-action, the box it pops up saying there is an error actually suggests what the fix could be. 

It feels to me that by adding a little bit more text in two places (dialog box and on the help page) users will be able to self help in an obvious way. Help page is higher priority, as that is easy to deploy. Dialog box clearly has to wait for a new version of Mu to push through the pipeline.

Could you consider this though, I think it would be really helpful for users.

Thanks

ntoll, 2017-04-20:
Regarding Windows... it works well enough with OSX and Linux. It's a relatively simple change for Microsoft to make and would make their OS much more friendly for embedded development. Everyone wins.

Totally agree with you that we should do as much to help people get things set up properly. It might just be as simple as an "on boarding" checklist if you're on Windows that's shown when you first start Mu for the first time.

carlosperate, 2017-04-20:
I think the driver should install itself under Windows 10, and possibly Windows 8, when the micro:bit is first plugged in. However, many institutions and schools are probably running Windows 7.

> It might just be as simple as an "on boarding" checklist if you're on Windows that's shown when you first start Mu for the first time.

I think most users will probably just close that without reading it and be as effective as the message in the download page. On the other they are more likely to read an error message when things stop working, so I think it'd be more effective to have a note about the driver on that alert message when it is opened under windows.

About the help pages... that reminds me about something I wanted to propose for quite a long time, to make it simpler and easier to submit PRs to just the help pages. I'll create an example repo within the next hour or so to illustrate how I would like to organise it (I'm pretty sure I wrote about it in one of the issues, but I cannot find it any more).

DavidWhaleMEF, 2017-04-20:
We tried windows 10 today, and no, it doesn't auto install the mbed serial driver when you plug the micro:bit in - it just comes up as an unknown device.

It's 100 times worse in schools, you can't install the driver without admin access, and kids accounts are not set as administrators. So this means your said school has to go round to 200 computers and manually plug in a micro:bit and install the mbed serial driver while logged in as an administrator, which is a real logistical nightmare.

I'm not making this stuff up either, we've had schools report to us that this is the only way they can get the REPL to work with Mu on their school system.

carlosperate, 2017-04-20:
Were you connected to the internet? It guess it might be possible for the network to have that feature deactivated as well, but the drivers are definitely available from the Windows 10 repositories.
Nervelessness, it is a PITA when it doesn't work, having to do that manually is definitely an inconvenience.

carlosperate, 2017-04-21:
The PR for the documentation build system: https://github.com/mu-editor/mu-editor.github.io/pull/9
It doesn't directly affect this issue, but if we like the PR I'd like to have it merged before we update the documentation itself.

ntoll, 2017-04-21:
*sigh*

The fundamental problem is school IT systems are broken and not fit for purpose.

Ergo, we *should* help people get around this, but, more importantly, this is an opportunity for education: configure your setup so you *do* have the right drivers at the touch of a button.

Nevertheless, I do not think Mu will be improved by pop-ups and hints to cover every possible misconfiguration. We should trust our users will read the docs.

It's the equivalent of assuming musicians will look at a conductor while playing in a symphony orchestra. It takes a bit of effort on the part of musician's but the end result is improved hugely. ;-)

DavidWhaleMEF, 2017-04-21:
@ntoll actually I disagree here. There is already a message that pops up erroneously saying 'could not find a BBC micro:bit' when pressing the REPL button, when it clearly has found a BBC micro:bit because it just flashed code to it successfully.

All I think that is being suggested here, is that the error message be made more correct and clearer and pointing the user to where to go to resolve the single problem that it has already detected.  I think it's *just* rewording some text in that existing dialog box that is all that is required to improve the workflow here.

It's not every possible mis configuration, it's one specific case. Mu has already in this case detected the problem, it's just that the words that pop up don't actually at the moment reflect correctly the cause (and the steps to resolve).

Thanks

ntoll, 2017-04-21:
Nope, nope, nope (sorry).

Mu has not found a micro:bit with which to connect via USB/serial. 

It has found a path on the filesystem that appears to be a micro:bit, and thus uses said path to flash the device. 

They're two completely different things. That this is not clear is down to the underlying complexity of the technology. 

I agree that we should change the wording (suggestions welcome) of an existing pop-up to help the user.

My original assertion still stands - a better fix would be for Windows to actually work properly. But we have no control over that. :-(

DavidWhaleMEF, 2017-04-21:
Implementation detail. Users are reporting this as being confusing.

I'll have a think of some suggested words and update this ticket, let me sleep on it :)

Which bit of windows is not working properly? We can log that issue with the Microsoft team and they can escalate it internally to get it fixed, but we just need to be clear what it is supposed to do, and what it is not doing. Is it the fact that you would expect windows to automatically install the third party mbed serial driver? I recall @carlosperate saying something about that?

ntoll, 2017-04-21:
I seem to remember various BBC people (Phil?) and also Jonny mentioning that Windows doesn't really need a USB serial driver, it's just a configuration option that Microsoft have defaulted to off (apparently it used to be on by default). I was told the head of ARM was going to take it up with CEO of MS. It's obviously not trickled down yet.

As always, wording suggestions would be most welcome..! :-)

ntoll, 2017-04-21:
My fundamental point is we're working around another problem rather than fixing the problematic case. While it's easier for us to fix Mu, a more useful outcome is for MS to fix Windows. Then *everyone* benefits.

martinohanlon, 2018-02-14:
Can you include the mbed driver in the new windows installer?

DavidWhaleMEF, 2018-02-14:
@ntoll might have a view on this.

The latest knowledge we have on the mbed serial driver is here:

https://support.microbit.org/support/solutions/articles/19000022103-how-do-i-use-the-serial-port-with-a-micro-bit-on-windows

Specifically note:

_Note that when installing the Windows mbed driver, you should make sure that the micro:bit is plugged into the USB first, before trying to install the driver - some users have experienced 'unknown install errors' of this driver, and it seems to be related to not having the micro:bit actually plugged in at the time you do the install. You will also need full admin rights on your PC to install the driver._

This might make it a bit challenging to do mass deployments via a packaged MSI (as last knowledge from ARM told us that you have to have admin access *and* the micro:bit plugged in when installing the driver).

DavidWhaleMEF, 2018-02-14:
Also, from this page:

https://docs.mbed.com/docs/mbed-os-handbook/en/latest/getting_started/what_need/#windows-serial-driver

Note this:

_Download the mbed Windows serial port driver.
Plug in your mbed device over USB. It mounts as an mbed drive.
Close all Explorer windows showing the mbed drive.
Run the installer. This may take some time or display a few “unsigned driver” warnings.
Troubleshooting
If you have multiple mbed devices but the serial port only appears for one of them: Make sure you run the installer for every device (plug in the device over USB and run the installer again); Windows loads the driver based on the serial number, so it needs to be run for each device individually.

If the installer fails because “No mbed Microcontrollers were found”: Check your device is plugged in properly over USB.

If the installer reports the message “mbedWinSerial_nnnnn.exe is not a valid Win32 application”: If you downloaded the installer using Internet Explorer, please try a different browser (Firefox, Chrome).

If the installer appears to hang: Check if Windows is displaying an “unsigned driver/permission” window; these often get hidden behind other windows with nothing to indicate so in the taskbar. The installer will continue to run as soon as you click OK._

DavidWhaleMEF, 2018-02-14:
@ntoll It's not a windows issue. The mbed serial driver was written specifically for ARM boards, and it was written in a way that requires it to be installed running as Admin with the appropriate board plugged in. @jaustin might be able to shed more light on this, but currently the only known way to install the driver (as it will not install automatically via windows driver update) is to follow the install instructions on the pages linked above).

If there is actually no code installed (i.e, just configuration settings in the registry) as part of the install process, it is possible for the manufacturer to create a .INF file with the appropriate CDC class GUID's in it and then digitally sign it, and that could be included in the Mu package. But we would need to loop in someone from ARM to confirm that position as to whether it is just a standard CDC class with appropriate VID/PID and GUID information, or whether the mbed serial driver really does include new kernel driver modules as part of the install process.

ntoll, 2018-06-04:
I'm marking this as closed since chatting with @jaustin a few weeks ago confirms the Windows missing the driver problem only happens if the micro:bit was one of those in the original drop (with old firmware). The solution to this problem is simply to update the firmware (i.e. it should be something Mu shouldn't really  particularly care about since I'd hope MBFoundation were already telling users to do this for all serial related problems with Windows 10).

closed 2018-06-04 by ntoll