mu-editor/mu #265: Time for a new release?

ZanderBrown, 2017-05-03:
It's been 6 months and 75 commits since 0.9.13

Perhaps when #257 & #264 have been merged we should go for a 0.9.2 release?

Since 0.9.13 we have added quite a bit to Mu.

Looking at the commit logs we have implemented:

* Atomic saving (@carlosperate)
* Log timestamps (@ntoll)
* PEP8 alterations (@tjguk)
* Various bug fixes and extra tests (@tjguk, @ntoll, @carlosperate (*thanks to @carlosperate for fixing the one i added in 0.9.13*))
* Add filename & version to titlebar (@ntoll)

All that along with a couple of the open PRs definetly deserves a new release

ntoll, 2017-05-03:
But the underlying version of MicroPython has problems. :-(

What do people think about releasing with a broken MicroPython rather than waiting for a fixed up version..?

ZanderBrown, 2017-05-03:
ah well not such i good idea.

Does @dpgeorge have an ETA on that?

ntoll, 2018-06-04:
The new (fixed) version of MicroPython is landed in master and will be out with beta.16 (soon).

closed 2018-06-04 by ntoll