mu-editor/mu #266: Make check returns errors

jabaa, 2017-05-06:
I cloned the project and checked it with `make check` without changing anything. I got following errors:

    $ make check

    rm -rf build
    rm -rf dist
    rm -rf mu.egg-info
    rm -rf .coverage
    rm -rf docs/_build
    find . \( -name '*.py[co]' -o -name dropin.cache \) -delete
    find . \( -name '*.bak' -o -name dropin.cache \) -delete
    find . \( -name '*.tgz' -o -name dropin.cache \) -delete
    find . \( -name _build -o -name var \) -type d -prune -o -name '*.py' -print0 | xargs -0 -r -n 1 pycodestyle --repeat --exclude=build/*,docs/*,mu/contrib*,mu/resources/api.py --ignore=E731,E402
    find . \( -name _build -o -name var -o -path ./docs -o -path ./mu/contrib \) -type d -prune -o -name '*.py' -print0 | xargs -0 -r pyflakes
    ./mu/interface.py:812:24: invalid syntax
            message_box.exec()
                           ^
    ./mu/logic.py:126: undefined name 'FileNotFoundError'
    ./mu/logic.py:144: undefined name 'FileNotFoundError'
    ./mu/logic.py:361: undefined name 'FileNotFoundError'
    ./mu/logic.py:583: undefined name 'FileNotFoundError'
    ./tests/test_interface.py:1014:17: invalid syntax
        mock_qmb.exec = mock.MagicMock(return_value=None)
                    ^
    ./tests/test_logic.py:156: undefined name 'FileNotFoundError'
    ./tests/test_logic.py:227: undefined name 'FileNotFoundError'
    ./tests/test_logic.py:942: undefined name 'FileNotFoundError'
    Makefile:25: recipe for target 'pyflakes' failed
    make: *** [pyflakes] Error 123

ZanderBrown, 2017-05-06:
What system are you using? Is python 2 or python 3 the default?

jabaa, 2017-05-06:
I'm working on Ubuntu

    $ python --version
    Python 2.7.13

jabaa, 2017-05-06:
Now it's working. I had to add some missing dependencies. PyQt5 and Qscintilla are missing in `requirements.txt`

ntoll, 2017-05-07:
Hi @jabaa Mu is a Python3 only project. If you look in the https://github.com/mu-editor/mu#development section of the README it has all the details on setting up Mu for local development. Contribution guidelines are here: https://github.com/mu-editor/mu/blob/master/CONTRIBUTING.rst

We :heart: feedback! :-)

ZanderBrown, 2018-06-04:
Move to close?

jabaa, 2018-06-04:
ok

closed 2018-06-04 by jabaa