mu-editor/mu #267: Opening two buffers with the same file

jmsoeiro, 2017-05-12:
(context: MacOS Sierra 10.12.4, mu  16/nov/2016)

Accidentally, I loaded the same file twice, and started editing both buffers independently. 

I think this should be programatically avoided, as it may lead  to  unintentional data loss.

To reproduce:
- Load the same file twice; edit one of the buffers; it will be marked modified; save it; 
- you will have two same_name different_content buffers marked unmodified; 
- the version on disk will be the one last saved, and you can save /  switch between them without any warning from mu; 

If you close mu, there is no warning about unsaved  work, but you effectively loose the contents of one buffer.

In the same setup, there is also some stranger behaviour detected, for instance:
- in some situations you  save  a modified buffer, but it stays  '*' marked;
- in some situations the 'save as' dialog is opened.

 I tried to reproduce both, but I couldn't, so ...

ZanderBrown, 2017-05-12:
Hmm interestingm, yes Mu does currently allow us to load the same file twice

matt-land, 2017-05-23:
https://github.com/mu-editor/mu/pull/269/files
I have a PR for this
- prevents loading twice
- tells the user what happened
- after the user ok's, focus the window corresponding to the correct buffer

carlosperate, 2017-06-22:
Implemented in #269, thanks @matt-land!

closed 2017-06-22 by carlosperate