mu-editor/mu #268: Open file from command line

peter-jensen, 2017-05-14:
Unable to open a file from the command line. Double clicking on a python script does therefore not work.
I.e ./run.py test.py does not open file.
Version of "mu".  37f4c7e365797151bfdd32b143098d0a7cd0f033
Platform Ubuntu 16.04.

ZanderBrown, 2017-05-14:
Good one, this should probably be a priority with our change of focus

peter-jensen, 2017-05-14:
If files can be opened by clicking on them, then there is a high probability that several invocations of the editor may exist. This could affect common resources such as .local/share/mu/settings.json and .cache/mu/log/mu.log.

ZanderBrown, 2017-05-14:
Single instance is probably worth looking into

carlosperate, 2017-06-22:
Implemented in #270, thanks @matt-land!

closed 2017-06-22 by carlosperate