mu-editor/mu #271: Issue installing microbit mu on pitop

GemmaMayLatham, 2017-05-31:
Hi, I have been installing the python mu on the raspberry pi on a set of pitops (running pitop OS based on raspbian I believe). There seems to be a class on installing this with installing the pitop speaker. For example, on one pitop I installed the pitop speaker first (sudo apt-get install -y pt-speaker) then when i tried t install mu (sudo apt-get install mu) I got this error 
[image]

On another machine I got the same error when trying to install the pitop speaker after installing mu successfully. I have done sudo apt-get install each time. Wondered if there is some kind of clash between the two
Thanks
Gemma

DavidWhaleMEF, 2017-05-31:
Do you mean you have done sudo apt-get update every time??

GemmaMayLatham, 2017-05-31:
let me check

GemmaMayLatham, 2017-05-31:
yes, sudo apt-get before each install attempt

DavidWhaleMEF, 2017-05-31:
sudo apt-get update is required before install, so that the package lists are updated, otherwise you may get unresolved dependency errors if the local package lists on your Raspberry Pi are out of date.

Can you confirm that you used sudo apt-get update before each install? Thanks

DavidWhaleMEF, 2017-05-31:
i.e.

```
sudo apt-get update
sudo apt-get install mu
```

ZanderBrown, 2017-05-31:
Looking at the screenshot (whatever you want to call it) `sudo apt[-get] update` has been run.

What does `sudo apt install -f` recommend?

Are packages being held back pending other upgrades? Try `sudo apt upgrade` or check the output of `sudo apt list --upgradable`

carlosperate, 2017-05-31:
Is it possible you are running raspbian wheezy? Could you run the following command and copy the output?
```
cat /etc/os-release
```

GemmaMayLatham, 2017-06-01:
sudo apt-get -f install gives me the following:

pi@pi-top:~ $ sudo apt install -f
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Correcting dependencies... Done
The following packages were automatically installed and are no longer required:
  libdrm-freedreno1 libdrm-nouveau2 libdrm-radeon1 libelf1 libffi-dev
  libllvm3.5 nodejs rlwrap
Use 'apt-get autoremove' to remove them.
The following extra packages will be installed:
  pt-aliases
The following NEW packages will be installed:
  pt-aliases
0 upgraded, 1 newly installed, 0 to remove and 316 not upgraded.
2 not fully installed or removed.
Need to get 0 B/2,278 B of archives.
After this operation, 2,048 B of additional disk space will be used.
Do you want to continue? [Y/n] y
(Reading database ... 139423 files and directories currently installed.)
Preparing to unpack .../pt-aliases_1.1.0-1_armhf.deb ...
Unpacking pt-aliases (1.1.0-1) ...
dpkg: error processing archive /var/cache/apt/archives/pt-aliases_1.1.0-1_armhf.deb (--unpack):
 trying to overwrite '/usr/bin/rpi-vol', which is also in package rpi-vol 1.0.0-1
Removing volume keyboard shortcuts from Openbox
Package 'pt-aliases' has been uninstalled, but requires openbox to be reconfigured. Please reconfigure openbox by running 'openbox --reconfigure' or restarting.
Errors were encountered while processing:
 /var/cache/apt/archives/pt-aliases_1.1.0-1_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

GemmaMayLatham, 2017-06-01:
sudo apt upgrade gives the following:

pi@pi-top:~ $ sudo apt upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
You might want to run 'apt-get -f install' to correct these.
The following packages have unmet dependencies:
 pt-speaker : Depends: pt-aliases (>= 1.0.0-1) but it is not installed
E: Unmet dependencies. Try using -f.

GemmaMayLatham, 2017-06-01:
sudo apt list --upgradble the following:

Listing... Done
3d-slash/jessie 3.0.0 armhf [upgradable from: 1.0.0-1]
alacarte/stable 3.11.91-2+rpi4 all [upgradable from: 3.11.91-2+rpi2]
apt/stable 1.0.9.8.4 armhf [upgradable from: 1.0.9.8.2]
apt-utils/stable 1.0.9.8.4 armhf [upgradable from: 1.0.9.8.2]
bash/stable 4.3-11+deb8u1 armhf [upgradable from: 4.3-11]
bind9-host/stable 1:9.9.5.dfsg-9+deb8u11 armhf [upgradable from: 1:9.9.5.dfsg-9+deb8u5]
binutils/stable 2.25-5+deb8u1 armhf [upgradable from: 2.25-5]
bluej/stable 3.1.7b all [upgradable from: 3.1.7]
ca-certificates/stable 20141019+deb8u3 all [upgradable from: 20141019+deb8u1]
ceed-universe/jessie 0.5.2-1 armhf [upgradable from: 0.3.2-1]
chromium-browser/stable 56.0.2924.84-0ubuntu0.14.04.1.1000 armhf [upgradable from: 45.0.2454.85-0ubuntu0.14.04.1.1097]
curl/stable 7.38.0-4+deb8u5 armhf [upgradable from: 7.38.0-4+deb8u3]
dbus/stable 1.8.22-0+deb8u1 armhf [upgradable from: 1.8.20-0+deb8u1]
dbus-x11/stable 1.8.22-0+deb8u1 armhf [upgradable from: 1.8.20-0+deb8u1]
debhelper/stable 10.2.2~bpo8+1 all [upgradable from: 9.20150101]
dmsetup/stable 2:1.02.90-2.2+deb8u1 armhf [upgradable from: 2:1.02.90-2.2]
dpkg/stable 1.17.27 armhf [upgradable from: 1.17.26]
dpkg-dev/stable 1.17.27 all [upgradable from: 1.17.26]
e2fslibs/stable 1.42.12-2 armhf [upgradable from: 1.42.12-1.1]
e2fsprogs/stable 1.42.12-2 armhf [upgradable from: 1.42.12-1.1]
eject/stable 2.1.5+deb1+cvs20081104-13.1+deb8u1 armhf [upgradable from: 2.1.5+deb1+cvs20081104-13.1]
file/stable 1:5.22+15-2+deb8u3 armhf [upgradable from: 1:5.22+15-2+deb8u1]
firmware-atheros/stable 0.43+rpi5 all [upgradable from: 0.43+rpi4]
firmware-brcm80211/stable 0.43+rpi5 all [upgradable from: 0.43+rpi4]
firmware-libertas/stable 0.43+rpi5 all [upgradable from: 0.43+rpi4]
firmware-ralink/stable 0.43+rpi5 all [upgradable from: 0.43+rpi4]
firmware-realtek/stable 0.43+rpi5 all [upgradable from: 0.43+rpi4]
fontconfig/stable 2.11.0-6.3+deb8u1 armhf [upgradable from: 2.11.0-6.3]
fontconfig-config/stable 2.11.0-6.3+deb8u1 all [upgradable from: 2.11.0-6.3]
fonts-opensymbol/stable 2:102.6+LibO4.3.3-2+rpi4 all [upgradable from: 2:102.6+LibO4.3.3-2+rpi3]
gdb/stable 7.7.1+dfsg-5+rpi1 armhf [upgradable from: 7.7.1+dfsg-5]
gdbserver/stable 7.7.1+dfsg-5+rpi1 armhf [upgradable from: 7.7.1+dfsg-5]
gir1.2-gdkpixbuf-2.0/stable 2.31.1-2+deb8u5 armhf [upgradable from: 2.31.1-2+deb8u4]
git/stable 1:2.1.4-2.1+deb8u3 armhf [upgradable from: 1:2.1.4-2.1+deb8u1]
git-core/stable 1:2.1.4-2.1+deb8u3 all [upgradable from: 1:2.1.4-2.1+deb8u1]
git-man/stable 1:2.1.4-2.1+deb8u3 all [upgradable from: 1:2.1.4-2.1+deb8u1]
gnupg/stable 1.4.18-7+deb8u3 armhf [upgradable from: 1.4.18-7]
gnupg-agent/stable 2.0.26-6+deb8u1 armhf [upgradable from: 2.0.26-6]
gnupg2/stable 2.0.26-6+deb8u1 armhf [upgradable from: 2.0.26-6]
gpgv/stable 1.4.18-7+deb8u3 armhf [upgradable from: 1.4.18-7]
greenfoot/stable 2.4.2a all [upgradable from: 2.4.2]
gstreamer1.0-alsa/stable 1.4.4-2+deb8u1 armhf [upgradable from: 1.4.4-2]
gstreamer1.0-plugins-bad/stable 1.4.4-2.1+deb8u2 armhf [upgradable from: 1.4.4-2.1rpi4rpi1g]
gstreamer1.0-plugins-base/stable 1.4.4-2+deb8u1 armhf [upgradable from: 1.4.4-2]
gstreamer1.0-plugins-good/stable 1.4.4-2+deb8u3 armhf [upgradable from: 1.4.4-2]
gstreamer1.0-x/stable 1.4.4-2+deb8u1 armhf [upgradable from: 1.4.4-2]
gtk2-engines-clearlookspix/stable 1:2.20.4 armhf [upgradable from: 1:2.20.2-3+rpi1]
icu-devtools/stable 52.1-8+deb8u5 armhf [upgradable from: 52.1-8+deb8u3]
idle-python2.7/stable 2.7.9-2+deb8u1 all [upgradable from: 2.7.9-2]
initramfs-tools/stable 0.120+deb8u3 all [upgradable from: 0.120]
libapt-inst1.5/stable 1.0.9.8.4 armhf [upgradable from: 1.0.9.8.2]
libapt-pkg4.12/stable 1.0.9.8.4 armhf [upgradable from: 1.0.9.8.2]
libarchive13/stable 3.1.2-11+deb8u3 armhf [upgradable from: 3.1.2-11]
libasound2/stable 1.0.28-1+rpi2 armhf [upgradable from: 1.0.28-1+rpi1]
libasound2-data/stable 1.0.28-1+rpi2 all [upgradable from: 1.0.28-1+rpi1]
libaudiofile1/stable 0.3.6-2+deb8u2 armhf [upgradable from: 0.3.6-2]
libavcodec56/stable 6:11.9-1~deb8u1+rpi1 armhf [upgradable from: 6:11.4-1~deb8u1+rpi1]
libavformat56/stable 6:11.9-1~deb8u1+rpi1 armhf [upgradable from: 6:11.4-1~deb8u1+rpi1]
libavresample2/stable 6:11.9-1~deb8u1+rpi1 armhf [upgradable from: 6:11.4-1~deb8u1+rpi1]
libavutil54/stable 6:11.9-1~deb8u1+rpi1 armhf [upgradable from: 6:11.4-1~deb8u1+rpi1]
libbind9-90/stable 1:9.9.5.dfsg-9+deb8u11 armhf [upgradable from: 1:9.9.5.dfsg-9+deb8u5]
libblas-common/stable 3.7.0-1~bpo8+1 armhf [upgradable from: 1.2.20110419-10]
libblas3/stable 3.7.0-1~bpo8+1 armhf [upgradable from: 1.2.20110419-10]
libbluetooth3/stable 5.23-2+rpi2 armhf [upgradable from: 5.23-2+rpi1]
libbluray1/stable 1:0.9.3-3 armhf [upgradable from: 1:0.6.2-1]
libc-bin/stable 2.19-18+deb8u9 armhf [upgradable from: 2.19-18+deb8u3]
libc-dev-bin/stable 2.19-18+deb8u9 armhf [upgradable from: 2.19-18+deb8u3]
[... 224 lines cut ...]
udev/stable 215-17+deb8u7 armhf [upgradable from: 215-17+deb8u3]
uno-libs3/stable 4.3.3-2+rpi4 armhf [upgradable from: 4.3.3-2+rpi3]
unzip/stable 6.0-16+deb8u3 armhf [upgradable from: 6.0-16+deb8u2]
ure/stable 4.3.3-2+rpi4 armhf [upgradable from: 4.3.3-2+rpi3]
va-driver-all/stable 1.7.3-2 armhf [upgradable from: 1.4.1-1]
vim-common/stable 2:7.4.488-7+deb8u3 armhf [upgradable from: 2:7.4.488-7]
vim-tiny/stable 2:7.4.488-7+deb8u3 armhf [upgradable from: 2:7.4.488-7]
wget/stable 1.16-1+deb8u2 armhf [upgradable from: 1.16-1]
[... cut ...]
xserver-xorg-core/stable 2:1.18.4-2+rpi1 armhf [upgradable from: 2:1.17.2-1+rpi1]
xserver-xorg-input-all/stable 1:7.7+16 armhf [upgradable from: 1:7.7+7+b1]
xserver-xorg-input-evdev/stable 1:2.10.3-1 armhf [upgradable from: 1:2.9.2-1~bpo8+1]
xserver-xorg-input-synaptics/stable 1.8.3-2 armhf [upgradable from: 1.8.2-1~bpo8+1]
xserver-xorg-video-fbdev/stable 1:0.4.4-1+rpi2 armhf [upgradable from: 1:0.4.4-1+rpi1]
xserver-xorg-video-fbturbo/stable 1.20161111~122359 armhf [upgradable from: 1.20150305~205709]

GemmaMayLatham, 2017-06-01:
sorry for all the copy and pasting, I'm just not really sure what to do with the information it gives back

GemmaMayLatham, 2017-06-01:
and David yes, I can confirm I ran update before each install

carlosperate, 2017-06-01:
Did you try to install mu again after all this?

It looks like there is a problem with `pt-aliases` (a dependency from `pt-speaker`). Might be a good idea to report that to the PiTop issue tracker as well.

GemmaMayLatham, 2017-06-01:
I have mu installed on this pitop but it wont let me install the pitop speaker but on another pitop I install the speaker first and now cant install mu

GemmaMayLatham, 2017-06-01:
I will see if I can find out where to report to pitop

carlosperate, 2017-06-01:
The copy-paste you did before for the `sudo apt-get -f install` command it is trying to fix the `pt-aliases` packages, which looks to be a dependency from `pt-speaker`. So I would assume at least that command was run in a PiTop that has had the `pt-speaker` packaged installed, and then I assume it failed to install Mu afterwards, no?
So in that PiTop, have you tried to install Mu again after running all these commands?

GemmaMayLatham, 2017-06-01:
no that last screen shot is from the pitop which has mu installed but not the speaker. any chance they use the same pt-aliases? I have re-run install and rebooted and still not installing

GemmaMayLatham, 2017-06-01:
I dont have the other pitop with me today to test it the other way round. How would I uninstall mu?

carlosperate, 2017-06-01:
It's going to be hard to track the problem if we have reports from different machines with different issues, one failing to install Mu after pt-speaker, and the other failing to install pt-speaker after Mu.
You can use `sudo apt-get remove xxx` to uninstall the packages and then try again.

Btw, are you using normal Raspbian, or the PiTop OS (is based on raspbian, but I believe they add package repositories, and maybe that could be causing issues or conflicts).

GemmaMayLatham, 2017-06-01:
when its trying to fix the pt-aliases after running sudo apt-get -f install it appears to be telling me that it requires openbox to be reconfigured. Tried running 'openbox --reconfigure@ as it says whihc does nothing and have also rest
arted the pi.

GemmaMayLatham, 2017-06-01:
its the pitop os

GemmaMayLatham, 2017-06-01:
But as PiTops are becoming so common place surely this needs to be something all are aware of if the given online install instructions for mu is not working

carlosperate, 2017-06-01:
And just to be clear, Mu does install and run correctly if pt-speaker is not installed first?

GemmaMayLatham, 2017-06-01:
yes and vice versa, the speaker will install if mu has not been installed first and its the same error either way round

GemmaMayLatham, 2017-06-01:
It won't let me remove mu....says pt-speaker depends on pt-aliases and to run apt-get -f install. Seems I'm stuck between the two now

GemmaMayLatham, 2017-06-01:
this is the error when trying to remove mu:

pi@pi-top:~ $ sudo apt-get remove mu
Reading package lists... Done
Building dependency tree       
Reading state information... Done
You might want to run 'apt-get -f install' to correct these:
The following packages have unmet dependencies:
 pt-speaker : Depends: pt-aliases (>= 1.0.0-1) but it is not going to be installed
E: Unmet dependencies. Try 'apt-get -f install' with no packages (or specify a solution).

ntoll, 2017-06-01:
From the original screenshot it appears that apt is complaining that `python3-pyqt5*` packages are "not going to be installed". This is peculiar and makes me wonder why. Do the various PiTop package rely on a different and incompatible `python3-pyqt5` version of things..?

Truth be told, if this is a vanilla Raspbian based distro "it should just work", however I can't see why there should be unmet dependencies errors unless PiTop packages use something that interferes with PyQt.

Or am I barking up the wrong tree..?

In any case, this is something about which we should liaise with the PiTop people.

Thoughts..?

carlosperate, 2017-06-01:
Okay, so I am very far from an expert in this packaging stuff, but looking at the `sudo apt install -f` output from @GemmaMayLatham previous post, we have some clues:

```
pi@pi-top:~ $ sudo apt install -f
Reading package lists... Done
Building dependency tree
Reading state information... Done
Correcting dependencies... Done
The following packages were automatically installed and are no longer required:
libdrm-freedreno1 libdrm-nouveau2 libdrm-radeon1 libelf1 libffi-dev
libllvm3.5 nodejs rlwrap
Use 'apt-get autoremove' to remove them.
The following extra packages will be installed:
pt-aliases
The following NEW packages will be installed:
pt-aliases
0 upgraded, 1 newly installed, 0 to remove and 316 not upgraded.
2 not fully installed or removed.
Need to get 0 B/2,278 B of archives.
After this operation, 2,048 B of additional disk space will be used.
Do you want to continue? [Y/n] y
```

Up to this point we are not sure what is causing the problem exactly, but there is an unmet dependency, `pt-aliases` that has to be installed to fix the issue.
Based on our previous conversation I assume in this Pi-Top Mu was installed successfully first, and then you tried to install `pt-speaker` and it failed.
___

Then this is the main problem:
```
(Reading database ... 139423 files and directories currently installed.)
Preparing to unpack .../pt-aliases_1.1.0-1_armhf.deb ...
Unpacking pt-aliases (1.1.0-1) ...
dpkg: error processing archive /var/cache/apt/archives/pt-aliases_1.1.0-1_armhf.deb (--unpack):
trying to overwrite '/usr/bin/rpi-vol', which is also in package rpi-vol 1.0.0-1
```

This directory `/usr/bin/rpi-vol` has already been installed by the package `rpi-vol 1.0.0-1`.  How and when this packaged was installed to begin with is unknown to me, it would be weird if it was installed with Mu, as it is not a direct dependency, and being raspberry pi specific it doesn't sound like it would an indirect dependency either.
___

Then:
```
Removing volume keyboard shortcuts from Openbox
````
I am not really sure why this is printed, don't really know where it is coming from.
___

Then it just undoes what it tried to install with `pt-aliases`, so it tells you to reconfigure openbox since it is possible that it modified it during the (failed) installation.

```
Package 'pt-aliases' has been uninstalled, but requires openbox to be reconfigured. Please reconfigure openbox by running 'openbox --reconfigure' or restarting.
Errors were encountered while processing:
/var/cache/apt/archives/pt-aliases_1.1.0-1_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
___

So, one thing you could do is to manually install `pt-aliases` with the option to overwrite:

```
sudo dpkg -i --force-overwrite /var/cache/apt/archives/pt-aliases_1.1.0-1_armhf.deb
sudo apt-get -f install
```

But I cannot guarantee that will fix it, and if it does, I cannot guarante is not going to break whatever the `rpi-vol 1.0.0-1` package did.

GemmaMayLatham, 2017-06-01:
As someone who runs a lot of workshops and is hoping to do more with the microbits (but my kit consists of all piTops) I think it would def be worth you guys chatting to PiTop about this. I think I probably represent a good proportion of people who don't really know anything about the more complex stuff on the pi. I just want to be able to install something and it work if that makes sense.

GemmaMayLatham, 2017-06-01:
ok, let me try the manual install for pt-aliases (bare with me)

GemmaMayLatham, 2017-06-01:
after running the first one, it says "Errors were encountered while processing: pt-aliases"

I have logged the issues with PiTop support on their forum and will let you know if they come back with anything

carlosperate, 2017-06-01:
@ntoll 

> From the original screenshot it appears that apt is complaining that python3-pyqt5* packages are "not going to be installed". This is peculiar and makes me wonder why. Do the various PiTop package rely on a different and incompatible python3-pyqt5 version of things..?

This was my first thought as well, maybe PiTopOs has repositories with conflicting packages.

>Truth be told, if this is a vanilla Raspbian based distro "it should just work", however I can't see why there should be unmet dependencies errors unless PiTop packages use something that interferes with PyQt.

That was my though as well, however now I am thinking that maybe if installing pt-speakers has a dependency problem, then whatever you try to apt-get install will give you a similar error message. It just so happens that with Mu the only dependencies not already installed in Raspbian are the PtQt ones, and so the error only mentions those.

@GemmaMayLatham It would be quite useful if you could get a PiTop with nothing installer, or just pt-speaker, and then tried to install any other raspberry pi package using `apt-get`.

> In any case, this is something about which we should liaise with the PiTop people.
>
>Thoughts..?

Completely agree, do you know anybody from the team? Otherwise I'm sure Ben would be able to put us in contact.

GemmaMayLatham, 2017-06-01:
@carlosperate Happy to try with a fresh SD card and install something. Any suggestions for other packages I could test? Probably not got time until next week now though

ntoll, 2017-06-01:
@GemmaMayLatham if / when you reach out to PiTop, please point them to this issue so they're able to take part in the discussion. :-)

In any case, hurrah for open source collaboration for making stuff better for people. :tada:

GemmaMayLatham, 2017-06-01:
haha, totally. So nice to have this kind of support. I'll post a link to here on the PiTop forum. Thanks

GemmaMayLatham, 2017-06-01:
http://support.pi-top.com/support/tickets/14013

carlosperate, 2017-06-01:
@GemmaMayLatham about what else to install, I don't have any suggestions from the top of my head. Ideally we should try a package with dependencies not already installed by default, I know it's hard to figure out what might fall in that category, but any "big" application should do the trick (and editor, browser, remote desktop application, etc).

> http://support.pi-top.com/support/tickets/14013

:weary: a bit contra productive to have a support forum that requires registration to find solutions to problems that could have already been resolved...

DavidWhaleMEF, 2017-06-01:
@carlosperate PiTop use freshdesk same as micro:bit foundation - that is a support ticket that Gemma has logged, which for data protection reasons is a private ticket between Gemma and PiTop as it may have personal details in it.

Curated solutions in any freshdesk installation (which if they work the same as we do, will be curated from existing tickets once they have reached a sensible resolution), are here: http://support.pi-top.com/support/solutions

There is a solution search at the top of that solutions page that will search all items that are in their knowledge base. If this is a new issue, it might not appear in their knowledge base yet.

DavidWhaleMEF, 2017-06-01:
Link to lots of excellent community support, on twitter: https://twitter.com/whaleygeek/status/870334208073498624

carlosperate, 2017-06-01:
Thanks David.

For anybody looking at this issue, it'd be great if somebody could test a "clean" install of `pt-speaker` on pitopOS and see if the `pt-aliases` dependency is installed correctly.
My assumption here is that there is a problem with the current version of `pt-aliases` and the `rpi-vol` package, and this issue blocks installing anything else with apt-get until it is resolved.

GemmaMayLatham, 2017-06-01:
Thanks all. Should be able to try the fresh install tomorrow night (not got any SD cards with me)

m-roberts, 2017-06-02:
Hi - pi-top engineer here. Is there a reason that you are running this specific version of pi-topOS? It's very old!

The latest version of pi-topOS has `pt-aliases`, `pt-speaker`, etc. all included and at the latest versions, and I have managed to install `mu` successfully.

I'm not sure why you're experiencing these problems, but this is not something that I can replicate with the latest version, and so, if there's no specific reason that you have not updated, I would recommend burning the latest OS and installing it on that.

DavidWhaleMEF, 2017-06-02:
@m-roberts It was apparently a new PiTop taken out of the box, booted up, and then it did auto updates. So it will be whatever version of the OS was on the card in the box that you delivered to the user. Looks like something broke in your upgrade process?

GemmaMayLatham, 2017-06-02:
Hi @m-roberts As David points out this was a PiTop CEED, preloaded SD card and speaker all out of the same box bought only last month from RS online. I ran the updates once logged in so seems a shame that I have to reload the SD card.  

I will do a fresh load on the SD as soon as I can and let you know how that goes.

Can you confirm that the PiTop speaker needs to be installed as someone on twitter mentioned it is now preloaded?

Thanks

DavidWhaleMEF, 2017-06-02:
It's worth noting that stock held by larger organisations like RS and Farnell, might have a long pipeline from the card first being imaged, to it being delivered to a customer and powered up. This means that it's never going to be the latest OS when customer receives it, so the robustness of the auto upgrade process is critical here for good user experience.

carlosperate, 2017-06-02:
> The latest version of pi-topOS has pt-aliases, pt-speaker, etc. all included and at the latest versions, and I have managed to install mu successfully.
>
> I'm not sure why you're experiencing these [...]

Hi @m-roberts, would it be possible to list what are the latest versions of `pt-speaker`, `pt-aliases`, and `rpi-vol` in pitopOS?
If you could uninstall these three packages (and dependencies) and then install `rpi-vol` and `pt-speaker` in that order, then it should help figure out if there is a conflict with both packages trying to overwrite '/usr/bin/rpi-vol'.

m-roberts, 2017-06-02:
Hi all,

After some more investigating, it seems as though the control file for `pt-aliases` is not declaring that it is not compatible with `rpi-vol`, which has been deprecated.
I am in the process of rolling out an updated version of `pt-aliases` (`1.1.0-2`) which specifically declares that it is to replace `rpi-vol`.

In general, updates have been smooth for most users, and I have no idea how this has only just been discovered, so please forgive the oversight.

We are working hard to ensure that the update system works for everyone, so please be sure to let me know if there are further issues, and I will do my best to handle them :)

m-roberts, 2017-06-02:
I have rolled out the update - please run `sudo apt-get update`, and see how you get on now!

GemmaMayLatham, 2017-06-02:
Hi @m-roberts  Thanks for the info. Glad I wasn't just imagining something. I am genera one of those people that you can rely on to happen upon an issue if there is one to be found . I am sure any others who may have come across this before will have known enough to work around it. 

I'll run the update when I get home and let you know how I get on. Thanks so much

Thanks all for your help, advise and support with this
Gemma

GemmaMayLatham, 2017-06-02:
Just tried sudo-apt-get update and then install again and still get an error. Will reflash SD card

GemmaMayLatham, 2017-06-02:
All installed after re-flashing latest OS.

carlosperate, 2017-06-22:
Okay, so since it looks like everything as been resolved I'll close this issue. Please do reopen if there is still a problem.

closed 2017-06-22 by carlosperate