mu-editor/mu #272: adding Adafruit circuitpy boards

ladyada, 2017-06-06:
hiya, we've launched a bunch of circuitpy boards, so if 
https://github.com/mu-editor/mu/blob/b16479ff55b1c4c7ef9a0ba054e96168ec7a4270/mu/logic.py#L45
could be updated to remove (0x239A, 0x8013), (0x239A, 0x8015) (they weren't used in the final python dist)
and add
```
 (0x239A, 0x8014) # metro m0 PID
 (0x239A, 0x8019) # circuitplayground m0 PID
 (0x239A, 0x801B) # feather m0 express PID
```
that would be lovely!

ntoll, 2017-06-07:
Ack

I'm in the middle of a bunch of Mu related work and I'll get this update in with the currently pending changes for the next release. I think you're going to like it. :-)

ladyada, 2017-06-07:
huzzah! :)

ladyada, 2017-07-24:
hiya checkin' in on this one. we also have a gemma m0 we're releasing with VID/PID 0x239A 0x801D

ntoll, 2017-07-24:
Cool. It's going into a branch I'm working on right now that'll turn into Mu 1.0 in the not-too-distant future.

closed 2017-11-28 by ladyada