mu-editor/mu #273: Segmentation fault (Linux binary on Ubuntu 16.04)

hackscribble, 2017-06-18:
I've being using `mu-0.9.13.linux.bin` from https://codewith.mu/ successfully on 64-bit Ubuntu 16.04.

Decided to try the latest Linux build from http://ardublockly-builds.s3-website-us-west-2.amazonaws.com/?prefix=microbit/linux/ as per the README.

When I execute the binary, I get the following error :

```
$ ./mu_2017-05-03_15_50_master_37f4c7e.bin
Segmentation fault (core dumped)
```

I've tried a couple of the other builds going back a few months and get the same error.  Is something else required apart from downloading the binary and making it executable?

carlosperate, 2017-06-22:
Oh wow, that looks like a tough one to debug.
The builds are done in Travis, and sometimes OS updates, or dependency updates can cause some inadvertent errors with the executables.

Did you at any point find a version from http://ardublockly-builds.s3-website-us-west-2.amazonaws.com/?prefix=microbit/linux/ that still worked?

carlosperate, 2017-06-22:
I've cannot replicate with https://s3-us-west-2.amazonaws.com/ardublockly-builds/microbit/linux/mu_2017-06-21_18_07_master_3202867.bin in 64bit 14.04.
I'll see if I can get a 16.04 VM going, I assume you are using default Ubuntu with Unity?

hackscribble, 2017-06-22:
> I'll see if I can get a 16.04 VM going, I assume you are using default Ubuntu with Unity?

That's right.  

I have narrowed down where the problem came in:
- `mu-2016-11-14_08_54_05` works OK
- `mu-2016-11-19_12_17_38` gives the segmentation fault

I haven't tried all the files, but I can't find any earlier ones with the fault, nor any later ones that work.

carlosperate, 2017-06-22:
Okay, so based on the dates these should be the logs, which list the versions of everything used to build Mu:

- `mu-2016-11-14_08_54_05` https://travis-ci.org/mu-editor/mu/builds/175643592 (or maybe https://travis-ci.org/mu-editor/mu/builds/175643562)
- `mu-2016-11-19_12_17_38` https://travis-ci.org/mu-editor/mu/builds/177250885

However there is a problem with travis and the logs cannot be fetched, I'll see if I can open an issue in the Travis tracker in case they can still be recovered.

carlosperate, 2017-06-22:
Yep, I can confirm the same seg fault in Ubuntu 16.04 64-bit Unity desktop 😕 
mu-2016-11-14_08_54_05 does work for me as well, although gtk complains about Glibversion being too old, I wonder if the problem is running in GTK3, my 14.04 setup uses xfce which is still GTK2.

ntoll, 2017-06-22:
Remember that Unity is going to be "retired" / dropped / killed off from the next version of Ubuntu.

ntoll, 2017-06-22:
I can confirm (unsurprisingly) I get this behaviour with Gnome on 16.04.

carlosperate, 2017-06-22:
Yes, but that doesn't mean is going away any time soon, and Gnome is likely to have the same issue?
A lot of the laptops I've seen running ubuntu for workshops and things like that usually stick with an LTS for quite a while. Although in those cases they can run Mu from source, so it shouldn't be much of a problem

carlosperate, 2017-06-22:
Okay, so I don't have a PyInstaller issue to link to (and there is no point really to spend the time looking for it now), but I remembered that during the past few months a few PyQt issues have been popping up in the repository tracker.

So I've updated the PyInstaller version installed in Travis to the latest version that passes their CI server tests, and seems to work on my environment, hooray! 

Could somebody try it as well to confirm? 
https://s3-eu-west-2.amazonaws.com/mu-builds/linux/mu_2017-06-22_16_44_linux_build_4c24a8d.bin

Also, it'd be great if somebody could also check if the OS X version still works. The build still hasn't started yet, and it's likely to fail for unrelated reasons, but it should end up here eventually: mu-builds.s3-website.eu-west-2.amazonaws.com/?prefix=osx/
Called something like `mu_2017-06-22_xx_xx_linux_build_4c24a8d.bin`

hackscribble, 2017-06-22:
Thanks @carlosperate :smile: your new build works for me.  

It does send a long list of warnings to the console: [mu-errors.txt](https://github.com/mu-editor/mu/files/1095614/mu-errors.txt)

carlosperate, 2017-06-22:
The error seems related to the css styling:
```
Gtk-WARNING **: Theme parsing error: gtk-widgets.css:1792:25: Missing name of pseudo-class
```
@ntoll I believe you'd done some work in this area recently, were you aware of this?

ZanderBrown, 2017-06-23:
Hmm.

Thing to note here is that it is a Gtk warning from a Qt application.

On the assumption your using Unity 7 on Ubuntu you are using a Gtk3 desktop probably with the Ambiance theme.

On such a setup Qt attempts to blend in and appear as if a Gtk3 application.

To achieve this effect it uses a set of special Gtk APIs to load the current Gtk theme.

So returning to that warning I would suggest this is a fault with the Ambiance theme and therefore a issue to raise with Canonical. Perhaps try switching to a different Gtk theme and checking for updates.

Anyway as the CSS parser in Gtk is pretty solid I would suggest the line regarding libstdc++ is the more likely cause of the segfault

ntoll, 2018-06-04:
OK... since we're moving away from .bin Linux executables for 1.0 I guess this can be marked as closed.

closed 2018-06-04 by ntoll

VoxelTek, 2022-07-13:
This issue occurs with the Linux AppImage package, in the latest version downloaded from the mu website, on Kubuntu 22.04. I believe this issue should be re-investigated and re-opened.