mu-editor/mu #276: Single instance application

carlosperate, 2017-06-22:
We need to consider if this is something we should enforce with Mu. As discussed in https://github.com/mu-editor/mu/issues/268 several instances of the editor would be using the same config file and log.

The easiest solution would be to set the application to be single-instance only, so that would be my vote. But it'd be good to check if this is also the best solution for the Mu target audience, does anybody have an insight in this area?

ntoll, 2018-06-04:
I have absolutely no insight into this, nor do I know how to enforce single instance. Any idea how this would work? (I believe single instance is probably a helpful thing to do given the context you describe @carlosperate).

ZanderBrown, 2018-06-04:
I think we already have this?

ZanderBrown, 2018-06-04:
So in the Qt world the answer seems to be that it's doable but harder than it should be

tjguk, 2018-06-05:
The conventional approach is to attempt to acquire a mutex of some sort and, on failure, switch focus to the already-activated version of the app.

Trying to activate an existing app can be tricky, so an alternative to transparent activation of the existing instance is to force failure by putting up a modal dialog warning that an instance is already running.

The second is the less complex option, but even so I don't know whether it's necessary in comparison to the small risk of a session file being unexpectedly overwritten.

tjguk, 2018-06-05:
Alternatively, we can ignore the single-instance notion and work harder to *merge* session files when saving. Again, I don't know if this complexity outweighs the risk.

ZanderBrown, 2018-06-05:
This is why GApplication is so nice, I'd have thought everyone would have these niceties by now but it seems QApplication is relatively "dumb" in this area

Either solution is going to be complex but the current state of affairs isn't great either (especially as we are an open with the target now)

carlosperate, 2018-06-05:
Maybe we can use this: https://github.com/IARI/alsa_jack_gui/blob/master/qtsingleapplication.py

tjguk, 2018-06-05:
That's basically technique (1) for my post above. And it's good to have something already written and working. But it does complicate the startup of the app and leaves the door open for odd problems.

ntoll, 2018-06-05:
OK... so there's a thing in Qt called QLockFile... see: http://blog.aeguana.com/2015/10/15/how-to-run-a-single-app-instance-in-qt/ and https://doc.qt.io/qt-5/qlockfile.html

We'd just need to do:

`from PyQt5.QtCore import QLockFile`

Check/create lock file and if there's a problem complain with a modal error box. This feels the simplest solution.

carlosperate, 2018-06-05:
> There is no danger when the app crashes, because the lock will be destroyed with the app instance.

Okay, that's good. Main reason I've never suggested lock files is because I cannot count the number of times I had to manually remove these when apps crash (I'm looking at you Eclipse....)

ntoll, 2018-06-05:
Quite... it appears the QLockFile is exactly what we're looking for.

carlosperate, 2021-08-17:
This is also relevant: https://stackoverflow.com/questions/8786136/pyqt-how-to-detect-and-close-ui-if-its-already-running

haikusw, 2022-12-31:
> > There is no danger when the app crashes, because the lock will be destroyed with the app instance.
> 
> Okay, that's good. Main reason I've never suggested lock files is because I cannot count the number of times I had to manually remove these when apps crash (I'm looking at you Eclipse....)

I was checking out the 1.2.1 milestones of open tasks to see if there were any I could contribute to and ran across this.
Saw this comment about the crashed-app destroying the lock.  That doesn't sound like it's accurate on unix platforms (macOS) according to the documentation I found while tracking down and fixing #2375.

From the Qt 5.7 docs (emphasis on the last sentence mine):  [https://stuff.mit.edu/afs/athena/software/texmaker_v5.0.2/qt57/doc/qtcore/qsharedmemory.html](https://stuff.mit.edu/afs/athena/software/texmaker_v5.0.2/qt57/doc/qtcore/qsharedmemory.html)

> Unix: [QSharedMemory](https://stuff.mit.edu/afs/athena/software/texmaker_v5.0.2/qt57/doc/qtcore/qsharedmemory.html) "owns" the shared memory segment. When the last thread or process that has an instance of [QSharedMemory](https://stuff.mit.edu/afs/athena/software/texmaker_v5.0.2/qt57/doc/qtcore/qsharedmemory.html) attached to a particular shared memory segment detaches from the segment by destroying its instance of [QSharedMemory](https://stuff.mit.edu/afs/athena/software/texmaker_v5.0.2/qt57/doc/qtcore/qsharedmemory.html), the Unix kernel release the shared memory segment. **But if that last thread or process crashes without running the [QSharedMemory](https://stuff.mit.edu/afs/athena/software/texmaker_v5.0.2/qt57/doc/qtcore/qsharedmemory.html) destructor, the shared memory segment survives the crash.**

This made me go back and look at the fix I made for #2375 and realize it's not right.  worked on that more

carlosperate, 2023-01-13:
Having a quick look at those docs, this seems like a deal breaker to me:

> Note: On Windows, this class has problems detecting a stale lock if the machine’s hostname contains characters outside the US-ASCII character set.

We've had quite a few issues encoding and decoding paths on Windows when users have different languages configured, so the chances of somebody having a non ASCII hostname are high enough to worry about this.
And compared with the shared memory approach (on which worse-case-scenario a stale lock gets resolved after a restart), a user hitting this issue with the lock file will never be able to recover without manually deleting this file.

haikusw, 2023-01-13:
Hi @carlosperate Thanks for the information.  Very unfortunate Windows behavior for `QLockFile` is so flawed.   Strong reason to not use it (hopefully someone with a Windows dev setup can look into fixing that upstream in PyQt or QT).

Surprising about the anti-virus glitch, but also compelling.

Ok.  Where does that leave us in regards to #2386? 🤔

Seems like that conversation belongs there.

carlosperate, 2024-12-16:
I think we can close this one as completed via the multiple PRs mentioned here + https://github.com/mu-editor/mu/pull/2415.

closed 2024-12-16 by carlosperate