mu-editor/mu #278: Call tips transient bug

kamar535, 2017-08-10:
On my laptop (OS X) call tips works just as expected and pressing the help button takes me to https://codewith.mu/help/0.9.13/

On my department Linux system, call tips sometimes works but most of the times no call tips are shown. I have not yet been able to detect any pattern causing the call tips to work or not to work.  

When pressing the help button on the Linux version no web page is opened. Is there any other way to detect the version of Mu? I've tried both `mu --help` and `mu --version` but both flags seems to have no effect. 

When pressing the help button the following messages appear on STDOUT. 

```
$> uname --all
Linux arrhenius 4.4.0-89-generic #112-Ubuntu SMP Mon Jul 31 19:38:41 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
$> /opt/mu.bin 
Logging to /home/kamar535/.cache/mu/log/mu.log
libGL error: unable to load driver: swrast_dri.so
libGL error: failed to load driver: swrast
/bin/bash: /tmp/_MEIOKbWM2/libtinfo.so.5: no version information available (required by /bin/bash)
/bin/bash: /tmp/_MEIOKbWM2/libtinfo.so.5: no version information available (required by /bin/bash)
/opt/google/chrome/google-chrome: /tmp/_MEIOKbWM2/libdbus-1.so.3: no version information available (required by /usr/lib/x86_64-linux-gnu/libgconf-2.so.4)
/opt/google/chrome/google-chrome: /tmp/_MEIOKbWM2/libxkbcommon.so.0: no version information available (required by /usr/lib/x86_64-linux-gnu/libgdk-3.so.0)
/opt/google/chrome/google-chrome: /tmp/_MEIOKbWM2/libdbus-1.so.3: no version information available (required by /usr/lib/x86_64-linux-gnu/libavahi-client.so.3)
/opt/google/chrome/google-chrome: /tmp/_MEIOKbWM2/libdbus-1.so.3: no version information available (required by /usr/lib/x86_64-linux-gnu/libdbus-glib-1.so.2)
/opt/google/chrome/google-chrome: /tmp/_MEIOKbWM2/libdbus-1.so.3: no version information available (required by /usr/lib/x86_64-linux-gnu/libatk-bridge-2.0.so.0)
/opt/google/chrome/google-chrome: /tmp/_MEIOKbWM2/libxkbcommon.so.0: no version information available (required by /usr/lib/x86_64-linux-gnu/libmirclient.so.9)
/opt/google/chrome/google-chrome: /tmp/_MEIOKbWM2/libstdc++.so.6: version `CXXABI_1.3.8' not found (required by /usr/lib/x86_64-linux-gnu/libmirclient.so.9)
/opt/google/chrome/google-chrome: /tmp/_MEIOKbWM2/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by /usr/lib/x86_64-linux-gnu/libmirclient.so.9)
/opt/google/chrome/google-chrome: /tmp/_MEIOKbWM2/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /usr/lib/x86_64-linux-gnu/libmirclient.so.9)
/opt/google/chrome/google-chrome: /tmp/_MEIOKbWM2/libdbus-1.so.3: no version information available (required by /usr/lib/x86_64-linux-gnu/libatspi.so.0)
/opt/google/chrome/google-chrome: /tmp/_MEIOKbWM2/libstdc++.so.6: version `GLIBCXX_3.4.20' not found (required by /usr/lib/x86_64-linux-gnu/libmircommon.so.7)
/opt/google/chrome/google-chrome: /tmp/_MEIOKbWM2/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by /usr/lib/x86_64-linux-gnu/libmircommon.so.7)
/opt/google/chrome/google-chrome: /tmp/_MEIOKbWM2/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /usr/lib/x86_64-linux-gnu/libmircommon.so.7)
/opt/google/chrome/google-chrome: /tmp/_MEIOKbWM2/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by /usr/lib/x86_64-linux-gnu/libmirprotobuf.so.3)
/opt/google/chrome/google-chrome: /tmp/_MEIOKbWM2/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /usr/lib/x86_64-linux-gnu/libmirprotobuf.so.3)
/opt/google/chrome/google-chrome: /tmp/_MEIOKbWM2/libstdc++.so.6: version `CXXABI_1.3.9' not found (required by /usr/lib/x86_64-linux-gnu/libmircore.so.1)
/opt/google/chrome/google-chrome: /tmp/_MEIOKbWM2/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /usr/lib/x86_64-linux-gnu/libmircore.so.1)
/opt/google/chrome/google-chrome: /tmp/_MEIOKbWM2/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /usr/lib/x86_64-linux-gnu/libboost_system.so.1.58.0)
/opt/google/chrome/google-chrome: /tmp/_MEIOKbWM2/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /usr/lib/x86_64-linux-gnu/libprotobuf-lite.so.9)
/opt/google/chrome/google-chrome: /tmp/_MEIOKbWM2/libstdc++.so.6: version `GLIBCXX_3.4.20' not found (required by /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.58.0)
/opt/google/chrome/google-chrome: /tmp/_MEIOKbWM2/libstdc++.so.6: version `GLIBCXX_3.4.21' not found (required by /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.58.0)
```

The log `/home/kamar535/.cache/mu/log/mu.log` is empty.

ntoll, 2017-08-10:
It looks like you have your default browser sat as Chrome (that's what the OS is telling Python) but not all the dependencies are installed for Chrome to work. Please confirm you're able to use Chrome.

Regarding call tips, I'm an Ubuntu user (as you appear to be) and it works all the time for me. Is there anything about your system that's non-standard Ubuntu that could be causing such a discrepancy in behaviour (a long shot, I know). Again, logging works fine for me on Ubuntu.

kamar535, 2017-08-14:
I can start google chrome without any problems. 

```
$> which google-chrome
/usr/bin/google-chrome
$>
```

A [quick search](https://github.com/mu-editor/mu/search?utf8=%E2%9C%93&q=codewith.mu&type=) shows that Mu uses the webbrowser module to open the help url. 

From the terminal, the following

```
$> python -m webbrowser -t "http://www.python.org"
``` 

, successfully opens the page www.python.org in Firefox which is also the default browser set in `system-settings > Details > Default Applications`.  For some reason, when Mu uses the `webbrowser`module it fails to open a url in the default browser.

kamar535, 2017-08-14:
On OS X a call tip is shown directly after open parenthesis `(` is typed. 

[image]

On Ubuntu I found a pattern. The call tip is not shown after open parenthesis `(` is typed, but is shown after deleting the open parenthesis `(` and retyping it. 

[image]

ntoll, 2017-08-14:
Hi @kamar535, thanks for the further investigations.

Regarding browser opening - this appears to be a problem with Python's `webbrowser` module rather than Mu per-se (although I can't reproduce this behaviour on my Ubuntu machine).

As for call tips, I'm even more confused. On my version of Ubuntu it behaves just like your OSX gif demonstrates (i.e. the tip is shown immediately after the open parenthesis). I presume you're using the binary version of Mu downloaded from the website? If you clone this repository and follow the developer setup instructions in the README to get it running locally, does the behaviour change? If it does, I suspect the version of Scintilla used by the CI/build boxes is different / old compared to the current version packaged in Ubuntu - perhaps that explains the discrepancy in behaviour.

I'm on holiday this next fortnight but will do some experiments upon my return home, unless you manage to get to the bottom of it first.

Thanks!

carlosperate, 2017-08-14:
Hi @kamar535,

The .bin file packs it's own python executable and python modules, so if there is an issue with `webbrowser` it's likely to be a compatibility problem with the modules inside the Mu executable.
If it's complaining about glibc/libstdc++ versions, could you run thse command for me and paste the output here?
```
strings /usr/lib/x86_64-linux-gnu/libstdc++.so.6 | grep GLIBCXX;
```
and
```
/lib/x86_64-linux-gnu/libc.so.6
```

This should be able to list the versions available in your system.
Taking in consideration Chrome launches correctly outside of Mu, I'm assuming you will have version `3.4.21` available, but it's a good first step to see what the problem might be.

kamar535, 2017-08-15:
@ntoll The system admins installed the following Mu binary. 

```
$> file /opt/mu.bin 
/opt/mu.bin: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, BuildID[sha1]=dcb7928ff5bfd3f1de0f2f774950cd22212132c1, stripped
$>
```
I don't have root access on the system. Following the [README](https://github.com/mu-editor/mu)
I've been able to create a virtual environment and execute the following steps: 

```
pip install -r requirements.txt
pip install PyQT5
```

Trying to run Mu. 

```
$> python3 run.py 
Traceback (most recent call last):
  File "run.py", line 2, in <module>
    from mu.app import run
  File "/home/kamar535/mu/mu/app.py", line 28, in <module>
    from mu.interface import Window
  File "/home/kamar535/mu/mu/interface.py", line 33, in <module>
    from PyQt5.Qsci import QsciScintilla, QsciLexerPython, QsciAPIs
ImportError: No module named 'PyQt5.Qsci'
$>
```
I don't know how to install `python3-pyqt5.qsci` and `python3-pyqt5.qtserialport` locally as a none root user. Do I have to ask my system admins to install these packages or is there another way for me to play around with this myself?

kamar535, 2017-08-15:
@carlosperate 

```
>$ uname --all
Linux enequist 4.4.0-89-generic #112-Ubuntu SMP Mon Jul 31 19:38:41 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
$>
```

```
>$ strings /usr/lib/x86_64-linux-gnu/libstdc++.so.6 | grep GLIBCXX;
GLIBCXX_3.4
GLIBCXX_3.4.1
GLIBCXX_3.4.2
GLIBCXX_3.4.3
GLIBCXX_3.4.4
GLIBCXX_3.4.5
GLIBCXX_3.4.6
GLIBCXX_3.4.7
GLIBCXX_3.4.8
GLIBCXX_3.4.9
GLIBCXX_3.4.10
GLIBCXX_3.4.11
GLIBCXX_3.4.12
GLIBCXX_3.4.13
GLIBCXX_3.4.14
GLIBCXX_3.4.15
GLIBCXX_3.4.16
GLIBCXX_3.4.17
GLIBCXX_3.4.18
GLIBCXX_3.4.19
GLIBCXX_3.4.20
GLIBCXX_3.4.21
GLIBCXX_DEBUG_MESSAGE_LENGTH
$>
```

```
>$ /lib/x86_64-linux-gnu/libc.so.6
GNU C Library (Ubuntu GLIBC 2.23-0ubuntu9) stable release version 2.23, by Roland McGrath et al.
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.
Compiled by GNU CC version 5.4.0 20160609.
Available extensions:
	crypt add-on version 2.1 by Michael Glad and others
	GNU Libidn by Simon Josefsson
	Native POSIX Threads Library by Ulrich Drepper et al
	BIND-8.2.3-T5B
libc ABIs: UNIQUE IFUNC
For bug reporting instructions, please see:
<https://bugs.launchpad.net/ubuntu/+source/glibc/+bugs>.
$>
```

carlosperate, 2017-08-17:
>I don't know how to install python3-pyqt5.qsci and python3-pyqt5.qtserialport locally as a none root user. Do I have to ask my system admins to install these packages or is there another way for me to play around with this myself?

You can go to the https://pypi.org and download the python wheels, then use pip to install them in your virtual environment. I'm not quite sure if it'll work that way without root access, but it's worth a try.

carlosperate, 2017-11-03:
hi @kamar535, just wondering if you were able to install the dependencies via Wheels and run Mu from source. Do you need more help with this issue?

kamar535, 2017-11-03:
Sorry for not responding earlier. Since it was a while since I looked into this and there have been some updates to the department Linux system I've started from scratch. 

```
$ uname -av
Linux atterbom 4.4.0-98-generic #121-Ubuntu SMP Tue Oct 10 14:24:03 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
```

```
$ lscpu
Architecture:          x86_64
CPU op-mode(s):        32-bit, 64-bit
Byte Order:            Little Endian
CPU(s):                16
On-line CPU(s) list:   0-15
Thread(s) per core:    2
Core(s) per socket:    4
Socket(s):             2
NUMA node(s):          2
Vendor ID:             GenuineIntel
CPU family:            6
Model:                 26
Model name:            Intel(R) Xeon(R) CPU           E5520  @ 2.27GHz
Stepping:              5
CPU MHz:               2268.000
CPU max MHz:           2268.0000
CPU min MHz:           1600.0000
BogoMIPS:              4533.38
Virtualization:        VT-x
L1d cache:             32K
L1i cache:             32K
L2 cache:              256K
L3 cache:              8192K
NUMA node0 CPU(s):     0-3,8-11
NUMA node1 CPU(s):     4-7,12-15
Flags:                 fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc aperfmperf pni dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm dca sse4_1 sse4_2 popcnt lahf_lm tpr_shadow vnmi flexpriority ept vpid dtherm ida
$ 
```

I've created a new virtual environment named mu. 

```
(mu) $ pip --version
pip 9.0.1 from /home/kamar535/mu/local/lib/python2.7/site-packages (python 2.7)
(mu) $
```

```
(mu) $ pip install -r requirements.txt 
Collecting pytest (from -r requirements.txt (line 1))
  Using cached pytest-3.2.3-py2.py3-none-any.whl
Collecting pycodestyle (from -r requirements.txt (line 2))
  Using cached pycodestyle-2.3.1-py2.py3-none-any.whl
Collecting pyflakes (from -r requirements.txt (line 3))
  Using cached pyflakes-1.6.0-py2.py3-none-any.whl
Collecting coverage (from -r requirements.txt (line 4))
  Using cached coverage-4.4.1-cp27-cp27mu-manylinux1_x86_64.whl
Collecting sphinx (from -r requirements.txt (line 5))
  Using cached Sphinx-1.6.5-py2.py3-none-any.whl
Collecting pytest-cov (from -r requirements.txt (line 6))
  Using cached pytest_cov-2.5.1-py2.py3-none-any.whl
Collecting pyserial (from -r requirements.txt (line 7))
  Using cached pyserial-3.4-py2.py3-none-any.whl
Collecting pyqt5 (from -r requirements.txt (line 8))
  Could not find a version that satisfies the requirement pyqt5 (from -r requirements.txt (line 8)) (from versions: )
No matching distribution found for pyqt5 (from -r requirements.txt (line 8))
(mu) $
```

Tried to download the wheel files for `pyqt5` listed here https://pypi.org/project/PyQt5/#files and install with pip but all three gives the same error: 

```
(mu) $ pip install PyQt5-5.9.1-5.9.2-cp35.cp36.cp37-none-win_amd64.whl 
PyQt5-5.9.1-5.9.2-cp35.cp36.cp37-none-win_amd64.whl is not a supported wheel on this platform.
(mu) $ pip install PyQt5-5.9.1-5.9.2-cp35.cp36.cp37-none-win32.whl 
PyQt5-5.9.1-5.9.2-cp35.cp36.cp37-none-win32.whl is not a supported wheel on this platform.
(mu) $ pip install PyQt5-5.9.1-5.9.2-cp35.cp36.cp37-abi3-macosx_10_6_intel.whl 
PyQt5-5.9.1-5.9.2-cp35.cp36.cp37-abi3-macosx_10_6_intel.whl is not a supported wheel on this platform.
(mu) $ 
```

carlosperate, 2017-11-03:
Thanks for the quick response to my last message @kamar535.
Mu is Python 3 only, do you have Python 3 (ideally 3.5 or newer) installed in those computers? If so check if `pip3 --version` works and points to the main `python3` installation (`which python3` should show you where it is). If you have both, then `pip3 install -r requirements.txt` and the `python3 run.py` should do the trick.

kamar535, 2017-11-03:
```
$ python3 --version
Python 3.5.2
$
```

`pip3` is not installed but I managed to create a `virtualenv` with `pip3`. 

```
(mu) $ pip3 --version
pip 9.0.1 from /home/kamar535/mu/lib/python3.5/site-packages (python 3.5)
(mu) $
```

Now I can start to build Mu. 

```
(mu) $ pip3 install -r requirements.txt 
````
The builds hangs here for a very long time: 

```
  building 'twisted.test.raiser' extension
  creating build/temp.linux-x86_64-3.5
  creating build/temp.linux-x86_64-3.5/src
  creating build/temp.linux-x86_64-3.5/src/twisted
  creating build/temp.linux-x86_64-3.5/src/twisted/test
  x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m -I/home/kamar535/mu/include/python3.5m -c src/twisted/test/raiser.c -o build/temp.linux-x86_64-3.5/src/twisted/test/raiser.o
  src/twisted/test/raiser.c:4:20: fatal error: Python.h: No such file or directory
  compilation terminated.
  error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
  
  ----------------------------------------
  Failed building wheel for Twisted
  Running setup.py clean for Twisted
Failed to build Twisted
Installing collected packages: MarkupSafe, Jinja2, babel, sphinx, pytest-cov, pyserial, sip, pyqt5, qscintilla, ipython-genutils, parso, jedi, decorator, traitlets, simplegeneric, ptyprocess, pexpect, pickleshare, wcwidth, prompt-toolkit, ipython, tornado, jupyter-core, pyzmq, python-dateutil, jupyter-client, ipykernel, qtconsole, numpy, pyparsing, cycler, matplotlib, zope.interface, constantly, incremental, attrs, Automat, hyperlink, Twisted, lxml, cssselect, pycparser, cffi, asn1crypto, cryptography, pyOpenSSL, queuelib, PyDispatcher, pyasn1, pyasn1-modules, service-identity, w3lib, parsel, scrapy, beautifulsoup4, tqdm, requests-toolbelt, pkginfo, twine
```

Finally the builds terminates due to Disk quota exceeded. 

```
Exception:
Traceback (most recent call last):
  File "/home/kamar535/mu/lib/python3.5/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/home/kamar535/mu/lib/python3.5/site-packages/pip/commands/install.py", line 342, in run
    prefix=options.prefix_path,
  File "/home/kamar535/mu/lib/python3.5/site-packages/pip/req/req_set.py", line 784, in install
    **kwargs
  File "/home/kamar535/mu/lib/python3.5/site-packages/pip/req/req_install.py", line 851, in install
    self.move_wheel_files(self.source_dir, root=root, prefix=prefix)
  File "/home/kamar535/mu/lib/python3.5/site-packages/pip/req/req_install.py", line 1064, in move_wheel_files
    isolated=self.isolated,
  File "/home/kamar535/mu/lib/python3.5/site-packages/pip/wheel.py", line 345, in move_wheel_files
    clobber(source, lib_dir, True)
  File "/home/kamar535/mu/lib/python3.5/site-packages/pip/wheel.py", line 323, in clobber
    shutil.copyfile(srcfile, destfile)
  File "/home/kamar535/mu/lib/python3.5/shutil.py", line 116, in copyfile
    copyfileobj(fsrc, fdst)
OSError: [Errno 122] Disk quota exceeded
(mu) $
```

I will request more disk quota and try to build again on Monday.

carlosperate, 2017-11-04:
Thanks for the update @kamar535. Being Python 3.5 I was hoping most dependencies being already precompiled in python wheels, however looks like something within the twisted package needs building on your platform. Well, we are doing some progress, let us know how it goes once you get more disk space.

kamar535, 2017-11-06:
Got more quota, build now completes but with similar errors. 

```
  building 'twisted.test.raiser' extension
  creating build/temp.linux-x86_64-3.5
  creating build/temp.linux-x86_64-3.5/src
  creating build/temp.linux-x86_64-3.5/src/twisted
  creating build/temp.linux-x86_64-3.5/src/twisted/test
  x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m -I/home/kamar535/mu/include/python3.5m -c src/twisted/test/raiser.c -o build/temp.linux-x86_64-3.5/src/twisted/test/raiser.o
  src/twisted/test/raiser.c:4:20: fatal error: Python.h: No such file or directory
  compilation terminated.
  error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

  ----------------------------------------
  Failed building wheel for Twisted
  Running setup.py clean for Twisted
Failed to build Twisted
nstalling collected packages: Twisted, pyasn1, pyasn1-modules, service-identity, scrapy, beautifulsoup4, requests-toolbelt, pkginfo, tqdm, twine
  Running setup.py install for Twisted: started
    Running setup.py install for Twisted: finished with status 'error'
    Complete output from command /home/kamar535/mu/bin/python3.5 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-uuo3giy8/Twisted/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-76a72api-record/install-record.txt --single-version-externally-managed --compile --install-headers /home/kamar535/mu/include/site/python3.5/Twisted:
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-x86_64-3.5
    creating build/lib.linux-x86_64-3.5/twisted
    copying src/twisted/plugin.py -> build/lib.linux-x86_64-3.5/twisted
    copying src/twisted/__init__.py -> build/lib.linux-x86_64-3.5/twisted
    copying src/twisted/__main__.py -> build/lib.linux-x86_64-3.5/twisted
    copying src/twisted/_version.py -> build/lib.linux-x86_64-3.5/twisted
```
I skip to the end of the build output.

```
    building 'twisted.test.raiser' extension
    creating build/temp.linux-x86_64-3.5
    creating build/temp.linux-x86_64-3.5/src
    creating build/temp.linux-x86_64-3.5/src/twisted
    creating build/temp.linux-x86_64-3.5/src/twisted/test
    x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m -I/home/kamar535/mu/include/python3.5m -c src/twisted/test/raiser.c -o build/temp.linux-x86_64-3.5/src/twisted/test/raiser.o
    src/twisted/test/raiser.c:4:20: fatal error: Python.h: No such file or directory
    compilation terminated.
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

    ----------------------------------------
Command "/home/kamar535/mu/bin/python3.5 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-uuo3giy8/Twisted/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-76a72api-record/install-record.txt --single-version-externally-managed --compile --install-headers /home/kamar535/mu/include/site/python3.5/Twisted" failed with error code 1 in /tmp/pip-build-uuo3giy8/Twisted/
```
Complete log of the build: [log.txt](https://github.com/mu-editor/mu/files/1446261/log.txt)

carlosperate, 2017-11-06:
Assuming you are in Ubuntu, you will need to have installed the python dev packages, try:
```
sudo apt-get install python3-dev
```

kamar535, 2017-11-06:
I don't have root privileges on our department Linux/Ubuntu system. I will ask my system admins to install `python3-dev` and try again.

carlosperate, 2017-11-07:
Thanks @kamar535, and sorry this is turning out to be a much arduous process than it should be. I'm taking notes from all this to make sure than once we re document the installation steps we include all these things.

kamar535, 2017-11-07:
The package `python3-dev` is now installed and the build completes without errors. Running `run.py` crashes with the following error: 

```
(mu) $ python3 run.py
Traceback (most recent call last):
  File "run.py", line 2, in <module>
    from mu.app import run
  File "/home/kamar535/mu/mu/mu/app.py", line 40, in <module>
    from PyQt5.QtCore import QTimer
ImportError: /usr/lib/x86_64-linux-gnu/libQt5Core.so.5: version `Qt_5.9' not found (required by /home/kamar535/mu/lib/python3.5/site-packages/PyQt5/QtCore.so)
(mu) $
```

Shouldn't the build have detected that `Qt_5.9` in not available on my system?

The following versions of libQt are installed on my system. 

```
(mu) $ ls /usr/lib/x86_64-linux-gnu/ | grep "libQt[0-9]*Core.so"
libQt5Core.so.5
libQt5Core.so.5.5
libQt5Core.so.5.5.1
libQtCore.so
libQtCore.so.4
libQtCore.so.4.8
libQtCore.so.4.8.7
(mu) $
```
I will ask my system administrator to install `libQt5Core.so.5.9.

carlosperate, 2017-11-07:
Oh boy, now looks like the PyQt5 version installed via wheel is not compatible with the system installed Qt: https://stackoverflow.com/questions/39129615/pyqt5-not-finding-installed-qt5-library

You'll have to pip uninstall PyQt5 and have the following pacakges installed it via apt-get:
python3-pyqt5
python3-pyqt5.qsci
python3-pyqt5.qtserialport

 
pip install would not really detect the local library versions, it basically searched for the python package in the PyPi database (in this case PyQt5) and if there is a python wheel matching your operating system it will download it and install it. The wheel itself is provided by the creators of PyQt, and it is meant to come with the required files already built (the issue we had before was about a different dependency not provided in this format, that had to be built from source code in your computer). It's the first time I see this type of incompatibility, didn't realise the wheel also depends on system install QT libraries.

kamar535, 2017-11-07:
After pip uninstalling `pyqt5`, I also removed `pyqt5` from `requirements.txt`,  run the build again and can now start Mu.  

In the window asking me to select a mode there is no OK or Cancel buttons. 

[image]

carlosperate, 2017-11-07:
Glad you got it running from source.
@ntoll the missing "okay" and "cancel" buttons is a bit odd, any ideas?

kamar535, 2017-11-07:
Note that I've only tested remotely by means of `ssh -Y` to connect to the department Linux system so there might be some differences in the GUI due to that.

kamar535, 2017-11-08:
I missed that I got these errors in the terminal when running Mu. 

```
(mu) $ python run.py
Logging to /home/kamar535/.cache/mu/log/mu.log
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
(mu) $
```

Here is the [mu.log](https://github.com/mu-editor/mu/files/1453743/mu.log)

kamar535, 2017-11-08:
> The window resize should have not hidden the buttons, but just in case, did you try increasing that window size to see if they appear?

Changing the size of the settings window doesn't make the Ok and Cancel buttons appear.

kamar535, 2017-11-08:
Pressing the Help button opens the following web page: https://codewith.mu/en/help/1.0 but this page does not exist yet. 

```
$ wget https://codewith.mu/en/help/1.0
--2017-11-08 13:04:18--  https://codewith.mu/en/help/1.0
Resolving codewith.mu... 104.24.110.89, 104.24.111.89, 2400:cb00:2048:1::6818:6e59, ...
Connecting to codewith.mu|104.24.110.89|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2017-11-08 13:04:18 ERROR 404: Not Found.
```
This might be a known issue but I report anything I find :-)

carlosperate, 2017-11-10:
The libGL error, might be an issue with your graphics drivers: https://askubuntu.com/questions/541343/problems-with-libgl-fbconfigs-swrast-through-each-update

The documentation link I guess it's expected, as master hasn't been released yet so we haven't added a git submodule to https://github.com/mu-editor/mu-editor.github.io/tree/master/help, but thanks for the report, this is something we definitely need to do before the release.

carlosperate, 2017-11-10:
@kamar535 rather than keep adding stuff to this issue (where it's likely to be missed), could we open a new one for the Ok and cancell buttons not appearing?

ntoll, 2017-11-10:
Hi Folks,

Thank you @kamar535 for reporting the "help" bug. It's already on my to-do pre-release check-list. I've been up to my eyes with "real work" these past two weeks (with more to come during November and December) so progress is slowing - but definitely moving forward. Thank you @carlosperate for taking up the "replying to issues and making this a welcoming place to be" role. It's *very much appreciated*.

We'll get there in the end..! :-)

ntoll, 2018-03-21:
Hi, I'm using Ubuntu and can't reproduce the problem with call tips. The missing buttons problem appears to be the result of upgrading Qt (note: to fix, restart with a fresh virtualenv).

Ergo I'll close the issue.

closed 2018-03-21 by ntoll