mu-editor/mu #279: bash: ./mu-0.9.13.linux.bin: cannot execute binary file: Exec format error. is 64bit?

sctv, 2017-08-10:
I'm not sure if this is the right place to raise this. If there is a better place let me know.

I downloaded the mu linux executable from https://codewith.mu/#download
After doing chmod +x mu.bin I tried to execute it.

./mu-0.9.13.linux.bin 
`bash: ./mu-0.9.13.linux.bin: cannot execute binary file: Exec format error
`
I am wondering if I get this error message because I am running 32bit linux and the precompiled binary is for 64bit? 

If this is the case, this will confuse beginners to the website. Is there a way to make it more obvious?

here is output from file /sbin/init
`/sbin/init: ELF 32-bit LSB  shared object, Intel 80386, version 1 (SYSV), dynamically linked (uses shared libs), for GNU/Linux 2.6.24, BuildID[sha1]=c394677bccc720a3bb4f4c42a48e008ff33e39b1, stripped
 `

carlosperate, 2017-08-10:
Hi @sctv,

Yes, the binary is build for 64bit linux and will not work in 32bit distros.
And I think you are right, we should probably indicate in the website that this is for 64bit only.

carlosperate, 2017-08-10:
@ntoll My first though was that the download info box would be the best place for it, but taking having the 32/64 info only in one of the platforms would also get users wondering if the same applies to Windows (macOS has been 64 bit only long enough that I don't think it's relevant).

So we could add the info to the 3 boxes, or maybe just add it to the file itself? Thoughts?

sctv, 2017-08-10:
Hi Carlos,
Thanks for confirming what I thought(32bit vs 64bit).

I did get mu working by downloading the github repository and using
executing run.py.

gbaman, 2017-12-31:
Although a rather old Github issue now, I have just been asked the question as to why Mu only supports 32 bit (specifically on Linux) as their hardware was still 32 bit only. Any particular reason binaries for 32 and 64 bit aren't a doable thing?

carlosperate, 2017-12-31:
The linux version at the moment is 64 bits only, the main reason being that the Linux CI servers we use are 64 bits.
If we there was a great need for 32bit Linux builds we could find a way to do those in cloud CIs, but the amount of work required to achieve that is probably not worth the effort. It would be a complex set-up and the demand for it is very limited, also there is the option to run Mu from source in those environments.

On Windows the application is 32-bits because those work just fine in 64-bit Windows versions as well.

On macOS the application is 64 bits because of the CI server configuration again, and because the OS has been 64 bit only since 10.7, and most used already ditched 32 bits earlier than that.

ntoll, 2018-03-21:
Closing this as a "won't fix" for Linux 32bit. It turns out the upstream PyQt packages are not 32bit anyway.

closed 2018-03-21 by ntoll

sctv, 2018-03-22:
Thanks for the update and info.