mu-editor/mu #280: Mcafee false positive

microbit-carlos, 2017-08-14:
We have a user reporting that Mcafee is in some cases is identifying Mu as a virus and quarantining it.

Something interesting is that in one PC without the serial driver installed it works fine, but it is quarantined in a PC that does have the mbed serial driver. 

@ntoll I believe you had to deal with something similar in the past with Norton? Did you have to send a "Mu sample" for analysis? Or was it a trusted domain distributing executables type of thing?

A quick google search yields this info about submitting an application for analysis:
https://kc.mcafee.com/corporate/index?page=content&id=KB85567

https://service.mcafee.com/webcenter/portal/cp/home/articleview?articleId=TS102053

ntoll, 2017-08-14:
It was a trusted domain problem. We should submit Mu for analysis at McAfee. I'm on holiday for the next two weeks so, unless someone else wants to do this, I won't be doing anything until the end of August. We should probably submit the new (currently work in progress) version of Mu.

microbit-carlos, 2017-08-14:
I'll try to submit the latest release for analysis. Once the new version is released and we have working executables we can just submit it again. 

Enjoy your holiday!

microbit-carlos, 2017-08-17:
Just notice that for both methods listed in the mcafee website there is a file limit of 10 MBs, so need to find a different way to submit the sample.
Apart from that, to register an account we need a "Grant Number", not sure what that is either, probably a type of license key.

ntoll, 2017-08-17:
Hmm... OK, I'll look into this upon my return form Ireland (in just over a week).

ntoll, 2017-08-17:
I notice from the form needs the product, engine and "dat" version. How can we get this information?

microbit-carlos, 2017-08-21:
The product is McAfee LiveSafe, which looks like it's a suit of McAfee products bundled together. 

SecurityCenter:
* version: 16.2
* build: 16.2.142

VirusScan:
* version: 20.2
* build: 20.2.155
* engine version: 3075.0

Personal Firewall:
* version: 17.2
* build: 17.2.143

WebAdvisor:
* version: 4.0
* build: 4.0.6.149

Anti-Spam:
* version: 17.2
* build: 17.2.143
* engine version: 3.0.0.429

QuickClean and Shredder:
* version: 16.2
* build: 16.2.157

Vulnerability scanner:
* version: 6.2
* build: 6.2.145

That is all the Not really sure how to find out the "dat" version, not even sure what it is, probably the virus database?

I've invited to user to join this conversation, but I could also relay any messages through the microbit support system.

Chris-Westlake, 2017-08-21:
Hi, my name is Chris, it was my computer that had the issues with McAfee and the Mu Editor, if I can be of further assistance

microbit-mark, 2017-09-12:
Has this one moved forward any @ntoll? I've got this as pending in our support system. Let me know if I can help.

ntoll, 2017-09-12:
No forward movement yet. Only until we have a 1.0 build that we can submit to McAfee is there any point in getting them involved otherwise we're doing the same work twice in quick succession. (Something I'm keen to avoid ;-)

microbit-mark, 2017-09-12:
I'll note that it's pending a 1.0 release and do a KB article. Since @Chris-Westlake is on the thread, i think it's best to monitor progress here.

ntoll, 2017-09-12:
Great stuff. Sorry I can't be more helpful, but there's a whole bunch of stuff that needs to happen first before we can make this happen (but make this happen we will).

Chris-Westlake, 2017-09-12:
Hi Guys, the “McAfee exempt fix” is still working OK on two separate PCs.

ntoll, 2017-09-12:
Great, so we have a temporary work-around for the time being. :-)