mu-editor/mu #281: numu + circuitpython : REPL doesnt get input?

ladyada, 2017-08-15:
i got the repl working and its printing out all nicely, but if i focus on it and press ^C or ^D it doesn't cancel/restart the python program

[image]

i also can't get this to take input either :/
name = input("What is your name? ")
print("hello "+name)

[image]

running py3.5.2, installed requirements.txt running windows 7 x64 pro with 'python3 run.py'

ladyada, 2017-08-15:
eh? i restarted twice and now it works? is there a preferences or state folder i can delete to refresh my install? 

cause i swear the REPL didnt appear the first time at all, and second time didnt have input. (same thing happened on two win7 computers)

ntoll, 2017-08-15:
If you click on the cog you'll see the logs. The first thing Mu writes to logs are things like details of your system and locations of various things such as a `settings.json` file (usually in the default location for application settings for the given OS). The log window may also reveal what the REPL was up to - I added it because we got lots of feedback from teachers who wanted to work out what their students had done to get themselves into a certain state. ;-) 

I've used the `input` builtin function on other MicroPython boards and "it just works" for me (I realise this isn't helpful to you, but it may indicate it's something to do with your local set-up).

Keep this feedback coming! It's gold-dust for stuff to fix, improve, smooth over when I get back from holiday. Until then, assume alpha quality code. ;-)

ladyada, 2017-11-20:
ok i did more work and i narrowed this down to CircuitPython expecting **DTR** to be set (e.g. terminal open) but qtserial does not actually set DTR, even if you tell it to. which is...weird. but reproducable. here's a simple script that tests DTR setting and data reads. in windows 7, at least, DTR is always 'false' 

> import sys, time
> from PyQt5.QtWidgets import QWidget, QApplication, QPushButton
> from PyQt5.QtCore import QObject, pyqtSlot, QIODevice, pyqtSignal
> from PyQt5.QtSerialPort import QSerialPort
> 
> PORT = "COM119"
> 
> class DVIFlasher(QObject):
>     ser = None
>     flash_done = pyqtSignal(int, int)
> 
>     def __init__(self):
>         super(DVIFlasher, self).__init__()
>         self.ser = QSerialPort(PORT)
>         self.ser.readyRead.connect(self.read)
> 
>     def flash(self):
>         self.ser.setBaudRate(115200)
>         self.ser.setFlowControl(QSerialPort.HardwareControl)
>         print("open success?", self.ser.open(QIODevice.ReadWrite))
>         self.ser.dataTerminalReady = True
>         print("DTR: ", self.ser.isDataTerminalReady())
>     def read(self):
>         print("Got data: ", bytes(self.ser.readAll()).decode("utf-8") )
> 
> 
> class Widget(QWidget):
>     def __init__(self):
>         super(Widget, self).__init__()
>         self.btn = QPushButton("Start", self)
>         self.flasher = DVIFlasher()
>         self.flasher.flash()
>         self.resize(self.btn.sizeHint())
> 
> if __name__ == '__main__':
>     app = QApplication(sys.argv)
>     win = Widget()
>     win.show()
>     sys.exit(app.exec_())

commit 5c70f294 (2017-11-28): Fixes #281 and #302. Thank you @ladyada for your contribution.
closed 2017-11-28 by ntoll