mu-editor/mu #283: Pyinstaller fails on Ubuntu

titimoby, 2017-09-20:
I followed the Readme instructions and I have a working Mu when using `python3 run.py`

I though I will then try to package this version for a conference where I wanted to show the near future Mu but pyinstaller failed.

```
thierry-laptop% pyinstaller package/pyinstaller.spec
56 INFO: PyInstaller: 3.2.1
56 INFO: Python: 3.6.1
57 INFO: Platform: Linux-4.10.0-35-generic-x86_64-with-debian-stretch-sid
58 INFO: UPX is not available.
Spec file resources selected: [('../mu/resources/css/night.css', 'mu/resources/css'), ('../mu/resources/css/day.css', 'mu/resources/css'), ('../mu/resources/css/contrast.css', 'mu/resources/css'), ('../mu/resources/images/flash.png', 'mu/resources/images'), ('../mu/resources/images/stop.png', 'mu/resources/images'), ('../mu/resources/images/zoom-in.png', 'mu/resources/images'), ('../mu/resources/images/run.png', 'mu/resources/images'), ('../mu/resources/images/repl.png', 'mu/resources/images'), ('../mu/resources/images/load.png', 'mu/resources/images'), ('../mu/resources/images/zoom-out.png', 'mu/resources/images'), ('../mu/resources/images/quit.png', 'mu/resources/images'), ('../mu/resources/images/icon.png', 'mu/resources/images'), ('../mu/resources/images/step-out.png', 'mu/resources/images'), ('../mu/resources/images/theme.png', 'mu/resources/images'), ('../mu/resources/images/step-over.png', 'mu/resources/images'), ('../mu/resources/images/theme_day.png', 'mu/resources/images'), ('../mu/resources/images/theme_contrast.png', 'mu/resources/images'), ('../mu/resources/images/step-in.png', 'mu/resources/images'), ('../mu/resources/images/python.png', 'mu/resources/images'), ('../mu/resources/images/logs.png', 'mu/resources/images'), ('../mu/resources/images/new.png', 'mu/resources/images'), ('../mu/resources/images/help.png', 'mu/resources/images'), ('../mu/resources/images/microbit.png', 'mu/resources/images'), ('../mu/resources/images/files.png', 'mu/resources/images'), ('../mu/resources/images/adafruit.png', 'mu/resources/images'), ('../mu/resources/images/check.png', 'mu/resources/images'), ('../mu/resources/images/snippets.png', 'mu/resources/images'), ('../mu/resources/images/splash-screen.png', 'mu/resources/images'), ('../mu/resources/images/save.png', 'mu/resources/images'), ('../mu/resources/fonts/SourceCodePro-It.otf', 'mu/resources/fonts'), ('../mu/resources/fonts/SourceCodePro-Bold.otf', 'mu/resources/fonts'), ('../mu/resources/fonts/SourceCodePro-Regular.otf', 'mu/resources/fonts'), ('../mu/resources/fonts/SourceCodePro-Semibold.otf', 'mu/resources/fonts'), ('../mu/resources/fonts/LICENSE.txt', 'mu/resources/fonts'), ('../mu/resources/fonts/SourceCodePro-SemiboldIt.otf', 'mu/resources/fonts'), ('../mu/resources/fonts/SourceCodePro-BoldIt.otf', 'mu/resources/fonts')]
68 INFO: Extending PYTHONPATH with paths
['/home/thierry/Documents/perso/projects/mu',
 '/home/thierry/Documents/perso/projects',
 '/home/thierry/.pyenv/versions/3.6.1/lib/python3.6/site-packages/PyQt5']
68 INFO: checking Analysis
68 INFO: Building Analysis because out00-Analysis.toc is non existent
68 INFO: Initializing module dependency graph...
69 INFO: Initializing module graph hooks...
70 INFO: Analyzing base_library.zip ...
Traceback (most recent call last):
  File "/home/thierry/.pyenv/versions/3.6.1/bin/pyinstaller", line 9, in <module>
    load_entry_point('PyInstaller==3.2.1', 'console_scripts', 'pyinstaller')()
  File "/home/thierry/.pyenv/versions/3.6.1/lib/python3.6/site-packages/PyInstaller/__main__.py", line 90, in run
    run_build(pyi_config, spec_file, **vars(args))
  File "/home/thierry/.pyenv/versions/3.6.1/lib/python3.6/site-packages/PyInstaller/__main__.py", line 46, in run_build
    PyInstaller.building.build_main.main(pyi_config, spec_file, **kwargs)
  File "/home/thierry/.pyenv/versions/3.6.1/lib/python3.6/site-packages/PyInstaller/building/build_main.py", line 788, in main
    build(specfile, kw.get('distpath'), kw.get('workpath'), kw.get('clean_build'))
  File "/home/thierry/.pyenv/versions/3.6.1/lib/python3.6/site-packages/PyInstaller/building/build_main.py", line 734, in build
    exec(text, spec_namespace)
  File "<string>", line 38, in <module>
  File "/home/thierry/.pyenv/versions/3.6.1/lib/python3.6/site-packages/PyInstaller/building/build_main.py", line 212, in __init__
    self.__postinit__()
  File "/home/thierry/.pyenv/versions/3.6.1/lib/python3.6/site-packages/PyInstaller/building/datastruct.py", line 161, in __postinit__
    self.assemble()
  File "/home/thierry/.pyenv/versions/3.6.1/lib/python3.6/site-packages/PyInstaller/building/build_main.py", line 317, in assemble
    excludes=self.excludes, user_hook_dirs=self.hookspath)
  File "/home/thierry/.pyenv/versions/3.6.1/lib/python3.6/site-packages/PyInstaller/depend/analysis.py", line 560, in initialize_modgraph
    graph.import_hook(m)
  File "/home/thierry/.pyenv/versions/3.6.1/lib/python3.6/site-packages/PyInstaller/lib/modulegraph/modulegraph.py", line 1509, in import_hook
    source_package, target_module_partname, level)
  File "/home/thierry/.pyenv/versions/3.6.1/lib/python3.6/site-packages/PyInstaller/lib/modulegraph/modulegraph.py", line 1661, in _find_head_package
    target_module_headname, target_package_name, source_package)
  File "/home/thierry/.pyenv/versions/3.6.1/lib/python3.6/site-packages/PyInstaller/depend/analysis.py", line 209, in _safe_import_module
    module_basename, module_name, parent_package)
  File "/home/thierry/.pyenv/versions/3.6.1/lib/python3.6/site-packages/PyInstaller/lib/modulegraph/modulegraph.py", line 2077, in _safe_import_module
    module_name, file_handle, pathname, metadata)
  File "/home/thierry/.pyenv/versions/3.6.1/lib/python3.6/site-packages/PyInstaller/lib/modulegraph/modulegraph.py", line 2167, in _load_module
    self._scan_code(m, co, co_ast)
  File "/home/thierry/.pyenv/versions/3.6.1/lib/python3.6/site-packages/PyInstaller/lib/modulegraph/modulegraph.py", line 2585, in _scan_code
    module, module_code_object, is_scanning_imports=False)
  File "/home/thierry/.pyenv/versions/3.6.1/lib/python3.6/site-packages/PyInstaller/lib/modulegraph/modulegraph.py", line 2831, in _scan_bytecode
    global_attr_name = get_operation_arg_name()
  File "/home/thierry/.pyenv/versions/3.6.1/lib/python3.6/site-packages/PyInstaller/lib/modulegraph/modulegraph.py", line 2731, in get_operation_arg_name
    return module_code_object.co_names[co_names_index]
IndexError: tuple index out of range

```

Done from my Ubuntu 17.04 machine.
I can provide more details if it's useful, I just don't know what to add.

carlosperate, 2017-09-20:
Pyinstaller 3.2.1 does not support Python 3.6, I think your options would be to either use python 3.5, or use the latest pyinstaller from master https://github.com/pyinstaller/pyinstaller

carlosperate, 2017-09-22:
Pyinstaller 3.3 was released today, so `pip install pyinstaller` should now work with Python 3.6 as well.

titimoby, 2017-09-27:
I upgraded pyinstaller to get 3.3
Everything went fine and I have a package version.
My first talk ever was done right from the dev version, next one in end of october will be done with a package version! ;)

closed 2017-09-27 by titimoby