mu-editor/mu #284: Update mu/package/icons with new logo

ntoll, 2017-09-20:
The icons found in `mu/package/icons` all use the old logo. Please update these to the new logo (see attached).
[image]

ZanderBrown, 2017-09-20:
Looks nice, not 'mu' as such but at least it's distinct from IDLE

ntoll, 2017-09-20:
I wanted a snake to represent Python, but not the blue or yellow snakes from the official logo. "Mu" as in the Greek letter would be lost on most people. 

Also, I think she deserves a name! Pynelope perhaps..? ;-)

ZanderBrown, 2017-09-20:
I like it but maybe we can come up with something that's got an obscure reference in it? (unless i'm missing something other than the py)

ZanderBrown, 2017-09-20:
Neidr? Not so much a python reference but is apparently the welsh for snake (seems appropriate for a PyCon in Cardiff

ntoll, 2017-09-20:
Penelope is the name of my daughter... :-P

Pynelope seems an appropriate name for another thing I've "fathered" into existence. Is that obscure enough. :-D

ZanderBrown, 2017-09-20:
I'd say that's a good name then.

ntoll, 2017-09-20:
Hehehehe... it's official then. My daughter will be both amused, embarrassed and slightly annoyed. 

I'm not going to mention it until PyCon UK (she's coming along to the kids' day). :-D

carlosperate, 2017-09-20:
Awesome, which branch should I PR against? numu?

ntoll, 2017-09-20:
Do it against master. I merged numu into that when I delivered the beta version (i.e. feature complete, needs testing / debugging) for RPi last week.

carlosperate, 2017-09-20:
There will be no code updates, just icon files, so I'll commit directly to master if that's okay with you

ntoll, 2017-09-20:
+1 and please feel free to close this issue when it's pushed. Thanks! :-)

carlosperate, 2017-09-20:
Implemented in adef1daa756817d532edbfa36f0b3eddbfe0a07b :)

closed 2017-09-20 by carlosperate