mu-editor/mu #290: CI Servers: Uses AppVeyor/Travis

carlosperate, 2017-09-27:
This would not only speed up builds, but help avoid down time when external servers experience issues (like right now with SourceForge providing PyQT).

https://www.appveyor.com/docs/build-cache/
https://docs.travis-ci.com/user/caching/

ntoll, 2018-06-04:
Closing due to lack of movement. Travis/Appveyor just chug along in the background so I guess this isn't essential.

closed 2018-06-04 by ntoll

carlosperate, 2018-06-05:
We don't really have installables for PyQt to download, everything goes via PyPI which is already cached at the CI level for speed, so this task is now obsolete.

However, taking in consideration how slow is brew to download and compile Python on mac maybe we could have a good speed increase if we downloaded the official installer, cached it and installed it using that.
Very low priority at the moment (until we start testing stuff dependent on Travis, the 20+min builds annoys us again and we end up looking into this again out of frustration)