mu-editor/mu #291: Files button freezes Mu on MacBook

KFW, 2017-10-15:
Just downloaded Mu (0.9.13) on MacBook to use with micro:bit. Whenever I use the "Files" button I get the rainbow pinwheel and have to force quit Mu. If I don't have a micro:bit plugged in, I get a warning message. Curiously, if I unplug my micro:bit, the pinwheel stops - I get the warning about no Micro:Bit detected, but the screen shows the boxes for "Files on your micro:bit" and "Files on your computer". However the display doesn't actually show they python file that I'd saved. If I reattach micro:bit nothing happens. If I click the "Files" button again at that point, I get the pinwheel. I've attached the log file if that helps.
[Mu_error.txt](https://github.com/mu-editor/mu/files/1385605/Mu_error.txt)

ntoll, 2017-10-15:
Hi @KFW - thanks for the bug report. I'll take a look at it just as soon as I can get to my Mac. I may not have the same hardware / OS combination as you, but I'll try to recreate from your description.

KFW, 2017-10-20:
Had same exact same issues on iMac at home.

carlosperate, 2017-11-03:
As a general note for something we need to look into, is that I also had Mu crash a few times if I clicked on the Files button while the micro:bit was scrolling an exception message.

Boundarybreaker, 2017-11-08:
Note, I think this is an issue with APFS, Apple's new file system. It didn't happen before I started using High Sierra/APFS.

carlosperate, 2017-11-08:
That's a very interesting point @Boundarybreaker, thanks! And based on the log file it looks like @KFW is also using sierra.

ZanderBrown, 2017-11-08:
> That's a very interesting point @Boundarybreaker, thanks! And based on the log file it looks like @KFW is also using sierra.

Sierra ≠ High Sierra

Boundarybreaker, 2017-11-20:
I've tried accessing files and looking at the console, and I've found one single console log when it freezes:
`CoreAnimation: failed to receive fence reply: 10004003`
It seems like there's something wrong with the Mu app and CoreAnimation that's making it not like files? I have no idea.

Boundarybreaker, 2017-11-20:
Looking in the generated log file, it seems to crash too fast to actually log anything. I am noticing it is kinda spamming the log saying that it found a device once every second, though.
[mu.log](https://github.com/mu-editor/mu/files/1486987/mu.log)

ntoll, 2017-11-20:
Hi Folks, all this investigating is really wonderful to see. The "files" aspect of Mu needs a lot of love and I'll try to revisit it sometime this week. One immediately obvious problem is that it was hacked together in about 1/2 hour on a train journey into London (such were the constraints on my time back then), and as a result it's the most "quick and dirty" hack you can imagine. I'm pretty confident I can make it work better and fix things up such that the "whole" app won't fail when things go wrong (i.e. we may get to see the hoped for log of the error causing the crash).

ntoll, 2017-12-08:
Hi folks, a quick status update. I've done a bunch of refactoring on a private branch and ensured the `microfs` operations happen in a separate thread. I've also added much more robust error handling too (hopefully, if things go wrong we'll at least see an exception and stack trace in the logs). 

The test suite remains broken by this refactor (I've moved a bunch of code around) so once I've fixed up the tests I'll merge into master and I'd love some feedback.

ntoll, 2017-12-27:
Folks, release beta.13 (use pip to install it) contains the fixes mentioned above. Please test and all feedback most welcome.

Thanks..!

Boundarybreaker, 2018-02-12:
It works!!!

Boundarybreaker, 2018-02-12:
Or at least, it doesn't crash. It sends this error:
```There was a problem getting the list of files on the micro:bit. Please check Mu's logs for technical information. Alternatively, try unplugging/plugging-in your micro:bit and/or restarting Mu.```

ntoll, 2018-02-13:
@Boundarybreaker great stuff!

Could you make it happen again, but this time click on the little cog icon on the bottom right hand side of Mu's window and copy/paste the content into a file that you attach to this issue please? (Those are Mu's logs and will contain the error messages from OSX/Python explaining precisely what the problem is, thus allowing me to debug the issue and [hopefully] fix it).

ntoll, 2018-03-21:
Closing since the fix landed in beta.12

closed 2018-03-21 by ntoll