mu-editor/mu #293: Add pi specific instructions

carlosperate, 2017-11-03:
We currently have `requirements_pi.txt`, which we need for Raspbian Jessie.
It's very likely that with the Stretch release (and also thanks to @bennuttall piwheels https://github.com/bennuttall/piwheels once the new Py3.5 wheels are ready) it will work the same as with any other platform (just pip install).
At that point we should remove the `requirements_pi.txt` and just add a note in the general instructions with steps for the older versions of Raspbian.

bennuttall, 2017-11-03:
Unfortunately, I think there will still be some apt packages required, as pyqt is only distributed as wheels for windows, mac and linux - not arm - and no sdist, so piwheels (or a user) can't build it).

And if it's in requirements.txt this will fail without those lines taken out.

But it's not a huge problem for regular users as it'll be packaged for apt anyway. But annoying for developers who want to hack on Mu on the Pi.

bennuttall, 2017-11-03:
Ah I didn't see the `requirements_pi.txt` - so you've removed the relevant packages to get around this? Excellent!

carlosperate, 2017-11-03:
Yes, at the moment we have to ask to install these packages via `apt-get install`, and then the user could `pip install` from `requirements_pi.txt` instead of `requirements.txt`.

Does that mean that, as it currently stands and based on what's provided by Riverbank Computing, piwheels cannot automatically produce PyQt wheels?
If that's the case then it looks like we will have to continue with these two `requirements.txt` files for the foreseeable future.

bennuttall, 2017-11-03:
Yes, that's true. I've been in touch with Riverbank and they say it's a pain to build, they don't use setup.py, so it's something they don't support others doing.

carlosperate, 2017-11-03:
Thanks Ben, it's a shame they don't produce ARM wheels.
Even though it's likely the installation method is going to stay the same, I'll leave this issue open as we still need to document it.

ntoll, 2017-11-03:
+1 on all the above.

We shouldn't use `setup.py` for development. Rather, `pip install requirements_pi,txt` and `./run.py` should be enough to get Mu running on the Pi.

ntoll, 2018-03-21:
See: https://mu.readthedocs.io/en/latest/setup.html#raspberry-pi

closed 2018-03-21 by ntoll