mu-editor/mu #294: Install Mu from package source

carlosperate, 2017-11-03:
I've noticed we have a few different outstanding issues for installing Mu from source and dealing with its dependencies, so this is a "catch all" issue to collect everything related to this area.

* https://github.com/mu-editor/mu/issues/78: Consider pinning all build dependencies
* https://github.com/mu-editor/mu/issues/202: Outdated setup instructions for Mac
* https://github.com/mu-editor/mu/issues/120: requirements.txt misses some required packages
* https://github.com/mu-editor/mu/issues/266: Make check returns errors
* https://github.com/mu-editor/mu/pull/254: Add python3-pyqt5.qtwebkit dependancy
* https://github.com/mu-editor/mu/issues/142: Track pycodestyle debian package release
* https://github.com/mu-editor/mu/issues/216: setup.py needs to include install_requires
* https://github.com/mu-editor/mu/issues/217: Separate requirements.txt and requirements-dev.txt?
* https://github.com/mu-editor/mu/issues/197: setup.py fails on windows
    * https://github.com/mu-editor/mu/pull/257: Focused fix to allow setup.py intall to succeed on Windows 
* https://github.com/mu-editor/mu/issues/293: Add pi specific instructions

@ntoll are we still going to support `python setup.py install`? It's my impression based on the conversation from https://github.com/mu-editor/mu/commit/81c01f03a444ec7d6a898d0ca2a91569a9fe2c6d#commitcomment-24434748 that Mu will only support `pip install .` If so, is that something that needs to be added to the readmes? I assume some people will just try to do `python setup.py install` as soon as they clone the repo, and while they are likely to be experienced programmer, we should probably have at least one line somewhere indicating if it'll not work and what would work instead.

Things left to do:
* [ ] Add `python setup.py install`/`pip install .` clarification somewhere
* [ ] Clarify dependencies differences from using Python <= 3.4 or => 3.5 
* [ ] Add install instructions (from source) somewhere, including raspberry pi instructions if different from other debian based distros
* [ ] Separate running and development dependencies
* [ ] Test on Windows
* [ ] Test on Linux
* [ ] Test on macOS
* [ ] Test on RaspberryPi

Please keep in mind that some of the tasks above might have already done, or maybe should not be done at all. Ideally once we have all of these boxes ticked, then we can just consider this area done (for the time being).

ntoll, 2017-11-03:
+10000

I'm trying (and failing) to find time to document all the things so this sort of story is properly explained (viz my failed efforts during the sprints at PyCon UK - however, much progress was made in other ways).

carlosperate, 2017-11-03:
Indeed, there has been a lot of progress recently, which is bound to cause some of this things to fall out of sync.
I was able to get a running PyInstaller build with Python 3.6 + wheel dependencies on AppVeyor (where we were standing behind due to a few incompatibilities), so once I have that done, cleaned up and PRed I can look into putting some general instructions together.

carlosperate, 2017-11-04:
Added issue #295 and item to the to-do list about virtualenvs and `python`/`python3`, `pip`/`pip3` .

tjguk, 2018-02-28:
Another couple of notes in this area from the Windows end of things:

* There's talk, here and elsewhere, of specifying "pip3" or "python3" but these are not present on Windows (for the python.org installer; I think ActiveState / Anaconda create them)

* The latest PyQt wheels link against python3.dll which isn't copied over by the currently released virtualenv. There's a fix for that (cf https://github.com/pypa/virtualenv/pull/1083) but there's no released virtualenv which includes it. It *is* copied by the stdlib venv module, so py -3 -mvenv whatever works ok. I've opened #355 to track this, although I'm not quite sure what text should go where.

carlosperate, 2018-02-28:
I've had pip3 available in my path for quite a few releases, although I conciously always install python2, and then python3 with the py launcher, so don't know if this is an installer option or something that was added to the Windows installer at one point.

Neverthless, it is true that python3 is not really available on Windows as such, so we need to keep that in mind for the docs.

ZanderBrown, 2018-04-07:
Status update: #78 #120 #254 #266 #142 #217 #295 are all still open

ZanderBrown, 2018-04-10:
Looks like 'we' are featured here [MicroPython - Fedora Developer Portal](https://developer.fedoraproject.org/tech/languages/python/micropython.html) and yes the repos do contain a version *that old* but there is activity on the Red Hat Bugzilla to track releases

ntoll, 2018-04-10:
Heh... FWIW @kushaldas (a good friend of mine and awesome Mu supporter) is a Fedora developer and originally packaged Mu for that platform. When we release 1.0 I'll let him know. :-)

ntoll, 2018-04-10:
Also, Redhat Linux 5.0 was my first Linux distro back in 1998 (I still have the CD-ROM somewhere and the "Learn Linux in 24hour" book that came with it -- ah the memories). Feels rather satisfying that something I've written is now in Fedora (what RHL morphed into). :-D

ZanderBrown, 2018-04-10:
Originally found the packaged version by accident when PackageKit was trying to be helpful in bash, uflash is there as well

If you hadn't noticed I'm on Fedora 27

ntoll, 2018-04-10:
Heh... I switched to Debian "Potato" back in 2000 and have never looked back. FWIW I'm still trying to learn Linux (the "24 hours" was a lie). ;-)

ZanderBrown, 2018-04-10:
I've come to the conclusion all these "24 hours" things are lying, I started on Debian and eventually moved to Fedora last year and I can't say I miss Debian (although muscle memory still puts `apt` where `dnf` ought to be :slightly_smiling_face: )

ntoll, 2018-06-04:
I'm closing this since the checklist appears to be met by the various instructions and details found in the developer docs at: https://mu.readthedocs.io/en/latest/