mu-editor/mu #295: Not clear that pip3 must be used to build Mu

kamar535, 2017-11-03:
In the instructions https://github.com/mu-editor/mu#development `pip` is used and not `pip3`.

ZanderBrown, 2017-11-04:
> Ensure you have the correct dependencies for development installed by creating a virtualenv and running:

It is expected that you will be using a python3 `virtualenv` in which case `pip` is `pip3` however it would be good to have some details on creating that `virtualenv` and make its use a little more explicit than a passing mention

kamar535, 2017-11-04:
In my attempt at investigating #278 I simply followed the instructions to build Mu and only later found out that I needed to use a Python 3 `virtualenv`. Making this assumption explicit would be great.

carlosperate, 2017-11-04:
It's not required to use a virtualenv, but it is highly recommended. Rather than add instructions about doing virtualenvs, I would much rather find a good comprehensive article and link to it, pointing out that Python 3, and a pip for that python installation, is required.

I'll add this to the list in https://github.com/mu-editor/mu/issues/294.

cefn, 2018-02-09:
+1 for referencing `pip3` not `pip` since this works both in virtualenv AND outside virtualenv so therefore has no cost.

ntoll, 2018-06-04:
The developer docs are now explicit about using virtualenv.

closed 2018-06-04 by ntoll