mu-editor/mu #297: Missing OK and Cancel buttons in select mode dialog

kamar535, 2017-11-11:
As a follow-up to #278 I'm testing on the head of the master branch: 45694343b37ebfd7c25317811fa6a5d5292dfacb. In the window asking me to select a mode there is no OK or Cancel buttons. 

[image]

Changing the size of the settings window doesn't make the Ok and Cancel buttons appear.

I got these errors in the terminal when running Mu. 

```
(mu) $ python run.py
Logging to /home/kamar535/.cache/mu/log/mu.log
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
(mu) $
```

Here is the [mu.log](https://github.com/mu-editor/mu/files/1453743/mu.log)

carlosperate, 2017-11-16:
Thanks for opening the new issue @kamar535.

I assume you can still select one of the options by pressing the "Enter" key on your keyboard, or does that not work either?

BravoDelta151, 2017-12-01:
I ran into the same issue.  It was caused by having installed older PyQt libraries while testing another project.  I resolved the issue by removing PyQt and then reinstalling mu-editor with --upgrade option.  Note that I also removed QScintilla and qtconsole as well so it is possible they were at fault.
To remove:
pip uninstall PyQt5
Then:
pip install mu-editor --upgrade

ntoll, 2017-12-01:
Hi @BravoDelta151 

Awesome investigation of the fix! I'll add this to the README / documentation. Hurrah!

kamar535, 2017-12-05:
@carlosperate Pressing Enter doesn't work to select one of the options. I will now try the method described by @BravoDelta151 .

kamar535, 2017-12-05:
I can confirm that removing `PyQt5` and reinstalling Mu with the `--upgrade`options solved this issue. 

```
(mu) $ pip uninstall PyQt5
(mu) $ pip install mu-editor --upgrade
(mu) $ python3 run.py
```

[image]

carlosperate, 2017-12-05:
Glad you got it working, and thanks @BravoDelta151 for identifying the issue!
Might be a bit late now, but do you know which version of PyQt5 was producing the issue?

ZanderBrown, 2018-01-09:
Just experience the same issue on Fedora 27, resolved with
```
$ pip3 uninstall PyQt5
$ pip3 install --user mu-editor --upgrade
```
Which also cleared up some other Qt internal issues and improved native integration (open dialogue ext)

ntoll, 2018-03-21:
Closing since this is a Qt upgrade related problem.

closed 2018-03-21 by ntoll