mu-editor/mu #298: I'm running on Linux mint and it barely works

rhubarbdog, 2017-11-14:
Hi, after trying a few things I tried a github install of Mu.  The installation is not brilliant, access to 'files' is unreliable

i always get this output:
`Logging to /home/phil/.cache/mu/log/mu.log
QApplication: invalid style override passed, ignoring it.`

i often also get this diagnostic:
`QXcbConnection: XCB error: 3 (BadWindow), sequence: 5405, resource id: 12606807, major code: 40 (TranslateCoords), minor code: 0`

It quite often crashes, sometimes it just dies and i need to kill it with command `kill -term %%`
other times i get this python error output
`Traceback (most recent call last):
  File "/home/usrlocal/local/bin/mu-master/mu/modes/microbit.py", line 169, in toggle_files
    self.add_fs()
  File "/home/usrlocal/local/bin/mu-master/mu/modes/microbit.py", line 189, in add_fs
    self.view.add_filesystem(home=self.workspace_dir())
  File "/home/usrlocal/local/bin/mu-master/mu/interface/main.py", line 301, in add_filesystem
    self.fs_pane = FileSystemPane(home)
  File "/home/usrlocal/local/bin/mu-master/mu/interface/panes.py", line 447, in __init__
    self.ls()
  File "/home/usrlocal/local/bin/mu-master/mu/interface/panes.py", line 457, in ls
    microbit_files = microfs.ls(microfs.get_serial())
  File "/home/usrlocal/local/bin/mu-master/mu/contrib/microfs.py", line 146, in ls
    return ast.literal_eval(out.decode('utf-8'))
  File "/usr/lib/python3.5/ast.py", line 46, in literal_eval
    node_or_string = parse(node_or_string, mode='eval')
  File "/usr/lib/python3.5/ast.py", line 35, in parse
    return compile(source, filename, mode, PyCF_ONLY_AST)
  File "<unknown>", line 1
    croPython v1.9.2-34-gd64154c73 on 2017-09-01; micro:bit with nRF51822
               ^
SyntaxError: invalid syntax`

ntoll, 2017-11-15:
Hi @rhubarbdog thank you for the feedback.

Can you let me know the steps needed to reproduce the problems you've been having.

You say installation is not brilliant - in what way? It should be as simple as typing `pip install mu` in the command line (assuming you have Python3 installed).

Some of the errors output to your logs appear to be Qt related (i.e. the GUI framework upon which Mu is built) rather than problems with Mu itself.  It would be helpful to know how you've installed Mu.

Finally, you say it quite often crashes. This is where it would be *really* useful to have explicit  steps so I can try to reproduce.

Thanks in advance.

rhubarbdog, 2017-11-15:
Hi
Do you want me to send you all of this by email or by updating my github
issue?

Phil

ntoll, 2017-11-15:
It would probably be most useful (for the sake of wider visibility, in case others encounter the same, or similar bugs with Mu) if you attach things to this issue. Then at least others can see how we either fix or code around the problem. It also means other devs involved in Mu get to see the details you provide too.

Thank you!

rhubarbdog, 2017-11-15:
I installed Mu as per these instructions i answered on askubuntu
 https://askubuntu.com/questions/965820

if i flash this program onto the microbit I get the following error:`
`from microbit import *`
``
`#`
`# this works after user_input.py has been uploaded onto the micro:bit`
`# You have to wait for the painc message to end before you can click`
`# 'file' in Mu and transfer the module and reset the microbit`
`import user_input as ui`
``                    
 ``       
`menu = ("Horse","Cat","Exit","Goldfish","Pig")`
``
`if ui.ask_question("Are you happy?"):`
`    display.show(Image.HAPPY)`
`else:`
`    display.show(Image.SAD)`
`sleep(500)`
``    
`while True:`
``
`    item=ui.menu_list(menu)`
`    if item=="Exit":`
`        break`
``
`    value=ui.menu_range(1,6,2)`
``
`    display.scroll(str(value)+" "+item)`
``
``   
`display.clear()`

`line 7 ImportError no module named 'user_input'`
if i click file while this is scrolling Mu crashes dying with a syntax error
The code for user input is in this file here which is actually called user_input.py, but i had to add the .txt to get git to upload it
[user_input.py.txt](https://github.com/mu-editor/mu/files/1474164/user_input.py.txt)

where have you published installing mu with `pip install mu`

rhubarbdog, 2017-11-25:
I am unable to reproduce any running errors, but i did logout / reboot (i.e. took a new terminal) before I got good running.  Initially i hadn't performed step 8 of the install guide i think that missing step was crucial.  Apart from when the microbit is in panic mode i can't make Mu fall over. you know about this last problem in another issue consider this one closed

closed 2017-11-25 by rhubarbdog