mu-editor/mu #299: mu-0.9.13.linux.bin fails to run on Ubuntu 17.10

histoplasmosis, 2017-11-16:
After downloading mu-editor and chmod +x and attempting to run the bin file the following traceback is produced:
`Traceback (most recent call last):
  File "run.py", line 2, in <module>
  File "/usr/local/lib/python3.4/dist-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
  File "mu/app.py", line 25, in <module>
  File "/usr/local/lib/python3.4/dist-packages/PyInstaller/loader/pyimod03_importers.py", line 573, in load_module
ImportError: /usr/lib/x86_64-linux-gnu/mesa/libGL.so.1: undefined symbol: drmGetDevices2
Failed to execute script run
`

carlosperate, 2017-11-16:
This one is going to be a bit difficult to debug at the moment. While we look into it I would recommend to try to run Mu from source.

The recommended way is to crate virtual environment for with Python 3 and run: `pip install mu-editor`

Otherwise you could install Mu with all its dependencies globally by just running in a terminal: `pip3 install mu-editor`

histoplasmosis, 2017-11-16:
Works when installed from pip both in a virtual environment and locally.  The bin however still does not run and produces the aforementioned traceback.

carlosperate, 2017-11-16:
I'm very glad to hear you got it working from pip.

I think the bin file issue is one of those things that might be a bit too time consuming to diagnose properly. Based on some google searches it could be fixed by removing some of the files bundled by PyInstaller (that's what we use to package the Python application in an executable), but in that case the application is likely to stop working on other systems (possibly in older versions of ubuntu), so quite a bit of debugging and testing would be required to find out what the problem is and how to fix it.

We are probably going to drop support for the linux bin file for the next version of Mu, as it can be installed via pip or via deb file (and I think a fedora package is also being provided). So I'll probably won't be able to investigate much further, I hope you don't feel we are ignoring your issue, it's just that we have limited free time to work on this project and the pip alternative does currently work. Thank you very much for the report though! We really do appreciate you took the time to let us know about this and also test if the pip installation worked correctly!

tim1mw, 2017-11-19:
I have the same problem on Mageia 6, it worked ok on Mageia 5 but failed with this error after the upgrade. I think this is hardware dependent, since the failure only occurs on my laptop (which has Intel graphics), my desktop computer has an Nvidia chip.

Unfortunately, I can't get around this by using pip because it fails to install on my disto... (I'll post a separate issue for that)

histoplasmosis, 2017-11-19:
I also have Intel graphics on the machine that errors on the bin.

nthiery, 2018-01-04:
Just as a data point: I got the same failure with the provided binary on Ubuntu 17.10. I tried the pip installation within my usual conda environment but got a sesgfault.
```
mistral-~>pip install mu-editor
...
Successfully installed matplotlib-2.0.2 mu-editor-1.0.0b13 pycodestyle-2.3.1 pyflakes-1.6.0 pyqt5-5.9 pyserial-3.4 qscintilla-2.10.1 sip-4.19.6

mistral-~>mu
Logging to /home/nthiery/.cache/mu/log/mu.log
zsh: segmentation fault (core dumped)  mu
```
```
mistral-~>cat /home/nthiery/.cache/mu/log/mu.log 
2018-01-04 11:38:29,860 - root:101(run) INFO: 
-----------------

Starting Mu 1.0.0.beta.13
2018-01-04 11:38:29,860 - root:102(run) INFO: uname_result(system='Linux', node='mistral', release='4.13.0-22-generic', version='#25-Ubuntu SMP Tue Dec 19 11:13:36 UTC 2017', machine='x86_64', processor='x86_64')
2018-01-04 11:38:29,902 - mu.logic:344(__init__) INFO: Setting up editor.
2018-01-04 11:38:29,902 - mu.logic:353(__init__) DEBUG: Creating directory: /home/nthiery/.local/share/mu
2018-01-04 11:38:29,902 - mu.logic:167(get_settings_path) DEBUG: Creating settings file: /home/nthiery/.local/share/mu/settings.json
2018-01-04 11:38:29,903 - mu.logic:355(__init__) INFO: Settings path: /home/nthiery/.local/share/mu/settings.json
2018-01-04 11:38:29,903 - mu.logic:356(__init__) INFO: Log directory: /home/nthiery/.cache/mu/log
2018-01-04 11:38:29,903 - mu.logic:357(__init__) INFO: Data directory: /home/nthiery/.local/share/mu
2018-01-04 11:38:29,903 - mu.logic:365(setup) INFO: Available modes: python, adafruit, microbit, debugger
2018-01-04 11:38:29,903 - mu.logic:369(setup) DEBUG: Creating directory: /home/nthiery/mu_code
```

My conda environment as quite some stuff in it, so that may be the cause. Ping me if I it could provide useful data to redo this in a fresh conda env.

I like the aims of this project and am looking forward the debian packages :-)

ntoll, 2018-01-04:
@nthiery did you pip install in a virtualenv?

I don't use Conda (although I know what it is) and I wonder if, indeed, there is some sort of conflict (hence my question about virtualenv, which should isolate your Python environment, and install Mu in a sort of safe sandbox).

Any more details would be most welcome!

Alternatively, if you have Python3 debug symbols installed in some way, I guess you could see what went wrong with GDB: 

```
gdb `which python3` core 
```

ntoll, 2018-03-21:
Since the Linux .bin file is going to "go away" for the foreseeable future as we recommend people use `pip` instead and/or native Debian/Fedora packages, as/when they become available, I'll close this bug.

closed 2018-03-21 by ntoll

cefn, 2018-04-20:
Note the workaround mentioned at https://github.com/pyinstaller/pyinstaller/issues/3286