mu-editor/mu #300: pip installation on windows

carlosperate, 2017-11-16:
Opening this one in the "need to further investigate" category.

Tried to install Mu in a virtualenv  on Windows and go this error on app start-up:
```
Traceback (most recent call last):
  File "C:\python35\Lib\runpy.py", line 184, in _run_module_as_main
    "__main__", mod_spec)
  File "C:\python35\Lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\carlo\Envs\hahaha\Scripts\mu.exe\__main__.py", line 5, in <module>
  File "c:\users\carlo\envs\hahaha\lib\site-packages\mu\app.py", line 40, in <module>
    from PyQt5.QtCore import QTimer
ImportError: DLL load failed: The specified module could not be found.
```

Windows 10 x64
Python 3.5.2
PyQt5 5.9
Mu version 1.0.0b12

Pip freeze output:
```
colorama==0.3.9
cycler==0.10.0
decorator==4.1.2
ipykernel==4.6.1
ipython==6.2.1
ipython-genutils==0.2.0
jedi==0.11.0
jupyter-client==5.1.0
jupyter-core==4.4.0
matplotlib==2.0.2
mu-editor==1.0.0b12
numpy==1.13.3
parso==0.1.0
pickleshare==0.7.4
prompt-toolkit==1.0.15
pycodestyle==2.3.1
pyflakes==1.6.0
Pygments==2.2.0
pyparsing==2.2.0
PyQt5==5.9
pyserial==3.4
python-dateutil==2.6.1
pytz==2017.3
pyzmq==16.0.3
QScintilla==2.10.1
qtconsole==4.3.1
simplegeneric==0.8.1
sip==4.19.5
six==1.11.0
tornado==4.5.2
traitlets==4.3.2
wcwidth==0.1.7
win-unicode-console==0.5
```

Pip install output:
```
pip install mu-editor
Collecting mu-editor
  Downloading mu_editor-1.0.0b12-py3-none-any.whl (1.4MB)
    100% |################################| 1.4MB 931kB/s
Collecting pyserial==3.4 (from mu-editor)
  Downloading pyserial-3.4-py2.py3-none-any.whl (193kB)
    100% |################################| 194kB 3.3MB/s
Collecting qtconsole==4.3.1 (from mu-editor)
  Using cached qtconsole-4.3.1-py2.py3-none-any.whl
Collecting pycodestyle==2.3.1 (from mu-editor)
  Using cached pycodestyle-2.3.1-py2.py3-none-any.whl
Collecting qscintilla==2.10.1 (from mu-editor)
  Downloading QScintilla-2.10.1-5.9.1-cp35.cp36.cp37-none-win32.whl (1.6MB)
    100% |################################| 1.6MB 799kB/s
Collecting pyflakes==1.6.0 (from mu-editor)
  Downloading pyflakes-1.6.0-py2.py3-none-any.whl (227kB)
    100% |################################| 235kB 3.8MB/s
Collecting pyqt5==5.9 (from mu-editor)
  Using cached PyQt5-5.9-5.9.1-cp35.cp36.cp37-none-win32.whl
Collecting matplotlib==2.0.2 (from mu-editor)
  Using cached matplotlib-2.0.2-cp35-cp35m-win32.whl
Collecting ipython-genutils (from qtconsole==4.3.1->mu-editor)
  Using cached ipython_genutils-0.2.0-py2.py3-none-any.whl
Collecting ipykernel>=4.1 (from qtconsole==4.3.1->mu-editor)
  Using cached ipykernel-4.6.1-py3-none-any.whl
Collecting jupyter-client>=4.1 (from qtconsole==4.3.1->mu-editor)
  Using cached jupyter_client-5.1.0-py2.py3-none-any.whl
Collecting jupyter-core (from qtconsole==4.3.1->mu-editor)
  Downloading jupyter_core-4.4.0-py2.py3-none-any.whl (126kB)
    100% |################################| 133kB 3.7MB/s
Collecting pygments (from qtconsole==4.3.1->mu-editor)
  Using cached Pygments-2.2.0-py2.py3-none-any.whl
Collecting traitlets (from qtconsole==4.3.1->mu-editor)
  Using cached traitlets-4.3.2-py2.py3-none-any.whl
Collecting sip<4.20,>=4.19.3 (from pyqt5==5.9->mu-editor)
  Downloading sip-4.19.5-cp35-none-win32.whl (42kB)
    100% |################################| 51kB 852kB/s
Collecting numpy>=1.7.1 (from matplotlib==2.0.2->mu-editor)
  Downloading numpy-1.13.3-2-cp35-none-win32.whl (6.8MB)
    100% |################################| 6.8MB 217kB/s
Collecting pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=1.5.6 (from matplotlib==2.0.2->mu-editor)
  Using cached pyparsing-2.2.0-py2.py3-none-any.whl
Collecting python-dateutil (from matplotlib==2.0.2->mu-editor)
  Using cached python_dateutil-2.6.1-py2.py3-none-any.whl
Collecting cycler>=0.10 (from matplotlib==2.0.2->mu-editor)
  Using cached cycler-0.10.0-py2.py3-none-any.whl
Collecting pytz (from matplotlib==2.0.2->mu-editor)
  Downloading pytz-2017.3-py2.py3-none-any.whl (511kB)
    100% |################################| 512kB 2.0MB/s
Collecting six>=1.10 (from matplotlib==2.0.2->mu-editor)
  Using cached six-1.11.0-py2.py3-none-any.whl
Collecting tornado>=4.0 (from ipykernel>=4.1->qtconsole==4.3.1->mu-editor)
  Using cached tornado-4.5.2-cp35-cp35m-win32.whl
Collecting ipython>=4.0.0 (from ipykernel>=4.1->qtconsole==4.3.1->mu-editor)
  Downloading ipython-6.2.1-py3-none-any.whl (745kB)
    100% |################################| 747kB 1.5MB/s
Collecting pyzmq>=13 (from jupyter-client>=4.1->qtconsole==4.3.1->mu-editor)
  Downloading pyzmq-16.0.3-cp35-cp35m-win32.whl (797kB)
    100% |################################| 798kB 1.4MB/s
Collecting decorator (from traitlets->qtconsole==4.3.1->mu-editor)
  Using cached decorator-4.1.2-py2.py3-none-any.whl
Collecting pickleshare (from ipython>=4.0.0->ipykernel>=4.1->qtconsole==4.3.1->mu-editor)
  Using cached pickleshare-0.7.4-py2.py3-none-any.whl
Collecting simplegeneric>0.8 (from ipython>=4.0.0->ipykernel>=4.1->qtconsole==4.3.1->mu-editor)
Collecting prompt-toolkit<2.0.0,>=1.0.4 (from ipython>=4.0.0->ipykernel>=4.1->qtconsole==4.3.1->mu-editor)
  Using cached prompt_toolkit-1.0.15-py3-none-any.whl
Collecting colorama; sys_platform == "win32" (from ipython>=4.0.0->ipykernel>=4.1->qtconsole==4.3.1->mu-editor)
  Using cached colorama-0.3.9-py2.py3-none-any.whl
Collecting jedi>=0.10 (from ipython>=4.0.0->ipykernel>=4.1->qtconsole==4.3.1->mu-editor)
  Using cached jedi-0.11.0-py2.py3-none-any.whl
Requirement already satisfied: setuptools>=18.5 in c:\users\carlo\envs\hahaha\lib\site-packages (from ipython>=4.0.0->ipykernel>=4.1->qtconsole==4.3.1->mu-editor)
Collecting win-unicode-console>=0.5; sys_platform == "win32" and python_version < "3.6" (from ipython>=4.0.0->ipykernel>=4.1->qtconsole==4.3.1->mu-editor)
Collecting wcwidth (from prompt-toolkit<2.0.0,>=1.0.4->ipython>=4.0.0->ipykernel>=4.1->qtconsole==4.3.1->mu-editor)
  Using cached wcwidth-0.1.7-py2.py3-none-any.whl
Collecting parso==0.1.0 (from jedi>=0.10->ipython>=4.0.0->ipykernel>=4.1->qtconsole==4.3.1->mu-editor)
  Using cached parso-0.1.0-py2.py3-none-any.whl
Installing collected packages: pyserial, ipython-genutils, pyzmq, six, python-dateutil, decorator, traitlets, jupyter-core, jupyter-client, tornado, pickleshare, simplegeneric, wcwidth, prompt-toolkit, colorama, parso, jedi, pygments, win-unicode-console, ipython, ipykernel, qtconsole, pycodestyle, sip, pyqt5, qscintilla, pyflakes, numpy, pyparsing, cycler, pytz, matplotlib, mu-editor
Successfully installed colorama-0.3.9 cycler-0.10.0 decorator-4.1.2 ipykernel-4.6.1 ipython-6.2.1 ipython-genutils-0.2.0 jedi-0.11.0 jupyter-client-5.1.0 jupyter-core-4.4.0 matplotlib-2.0.2 mu-editor-1.0.0b12 numpy-1.13.3 parso-0.1.0 pickleshare-0.7.4 prompt-toolkit-1.0.15 pycodestyle-2.3.1 pyflakes-1.6.0 pygments-2.2.0 pyparsing-2.2.0 pyqt5-5.9 pyserial-3.4 python-dateutil-2.6.1 pytz-2017.3 pyzmq-16.0.3 qscintilla-2.10.1 qtconsole-4.3.1 simplegeneric-0.8.1 sip-4.19.5 six-1.11.0 tornado-4.5.2 traitlets-4.3.2 wcwidth-0.1.7 win-unicode-console-0.5
```

tjguk, 2017-11-19:
I can't reproduce them when installing from a local git clone. Steps to not reproduce:

py -3.5 -mvenv mutest
mutest\Scripts\activate.bat
python -mpip install --upgrade pip
pip install c:\users\tim\work-in-progress\microbit\mu
mu
(works fine)

I'll try again with a pip install from PyPI

tjguk, 2017-11-19:
(Just adding pip freeze output from the above test for comparison)

colorama==0.3.9
cycler==0.10.0
decorator==4.1.2
ipykernel==4.6.1
ipython==6.2.1
ipython-genutils==0.2.0
jedi==0.11.0
jupyter-client==5.1.0
jupyter-core==4.4.0
matplotlib==2.0.2
mu-editor==1.0.0b12
numpy==1.13.3
parso==0.1.0
pickleshare==0.7.4
prompt-toolkit==1.0.15
pycodestyle==2.3.1
pyflakes==1.6.0
Pygments==2.2.0
pyparsing==2.2.0
PyQt5==5.9
pyserial==3.4
python-dateutil==2.6.1
pytz==2017.3
pyzmq==16.0.3
QScintilla==2.10.1
qtconsole==4.3.1
simplegeneric==0.8.1
sip==4.19.5
six==1.11.0
tornado==4.5.2
traitlets==4.3.2
wcwidth==0.1.7
win-unicode-console==0.5

tjguk, 2017-11-19:
Can't reproduce with pip install mu-editor either. 

Only obvious environmental difference is WIn 8.1 x64

BravoDelta151, 2017-11-30:
Based on the error message, it is failing to load the Qt Core library.  This can happen if it is missing a dependency or is picking up an incompatible library by mistake.

I ran into a problem a couple days ago with the Qt buttons not appearing.   Suggest you try removing PyQt and reinstall with --upgrade to ensure all packages mu uses are up to the correct version.
To remove:
pip uninstall PyQt5
Then:
pip install mu-editor --upgrade

This should re-install PyQt5 with mu-editor

carlosperate, 2017-12-01:
I've tried again from an empty virtual env and I am experiencing the same issue. This is in a virtual environment so the PyQt5 wheel is being picked from PyPI, not really using any preinstalled package.

Anyway, I tried uninstalling PyQt and installing again, but it just reinstalls the same version and I still have the problem: [uninstalled_pyqt.log.txt](https://github.com/mu-editor/mu/files/1522759/uninstalled_pyqt.log.txt)

As the original error suggests, it looks like I simply cannot import PyQt5.QtCore:
```
from PyQt5 import QtCore
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: DLL load failed: The specified module could not be found.
```

carlosperate, 2017-12-01:
Okay, so it looks like the problem is with my virtualenv creation: https://stackoverflow.com/questions/42863505/dll-load-failed-when-importing-pyqt5

It copies over the `Scripts/python35.dll` file, but not the `Scripts/python3.dll`. If I copy `python3.dll` from the python installation folder to the virtualenv folder then it works.

Yeah, so looks like a possible bug in virtualenv? I'll try to update. Issue can be closed, thanks  @tjguk and @BravoDelta151 for the help.

closed 2017-12-01 by carlosperate