mu-editor/mu #301: Pip installation fails on Magiea Linux

tim1mw, 2017-11-19:
This is what I get on Mageia 6:

-bash-4.3# pip3 install mu-editor
Collecting mu-editor
  Using cached mu_editor-1.0.0b12-py3-none-any.whl
Collecting pyserial==3.4 (from mu-editor)
  Using cached pyserial-3.4-py2.py3-none-any.whl
Collecting pyflakes==1.6.0 (from mu-editor)
  Using cached pyflakes-1.6.0-py2.py3-none-any.whl
Collecting pyqt5==5.9 (from mu-editor)
  Could not find a version that satisfies the requirement pyqt5==5.9 (from mu-editor) (from versions: )
No matching distribution found for pyqt5==5.9 (from mu-editor)
You are using pip version 8.0.2, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

Mageia 6 is bundled with pyqt5.6, so that may or may not be interfering. I also note the pip upgrade suggestion, I'm reluctant to do this since it would replace the packaged version supplied with my disto and potentially create an upgrade conflict down the line.

carlosperate, 2017-11-22:
What version of Python is pip3 pointing to? If it's lower than 3.5 then there won't be compatible PyQt5 wheels.

tim1mw, 2017-11-28:
Python 3.5.3. Mageia has both pip3 and pip3.5 binaries, although they seem to be the same thing underneath. Weirdly if I repeatedly re-run the pip command, sometimes it fails with the error above, other times it gives a different error:

[timw@baa ~]$ pip3.5 install mu-editor
Collecting mu-editor
  Using cached mu_editor-1.0.0b12-py3-none-any.whl
Collecting pycodestyle==2.3.1 (from mu-editor)
  Using cached pycodestyle-2.3.1-py2.py3-none-any.whl
Collecting matplotlib==2.0.2 (from mu-editor)
  Using cached matplotlib-2.0.2.tar.gz
Collecting pyflakes==1.6.0 (from mu-editor)
  Using cached pyflakes-1.6.0-py2.py3-none-any.whl
Collecting pyserial==3.4 (from mu-editor)
  Using cached pyserial-3.4-py2.py3-none-any.whl
Collecting qtconsole==4.3.1 (from mu-editor)
  Using cached qtconsole-4.3.1-py2.py3-none-any.whl
Collecting qscintilla==2.10.1 (from mu-editor)
  Could not find a version that satisfies the requirement qscintilla==2.10.1 (from mu-editor) (from versions: )
No matching distribution found for qscintilla==2.10.1 (from mu-editor)
You are using pip version 8.0.2, however version 9.0.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

I tried installing the Mageia qscintilla packages for python3, but that didn't help, that said, the Mageia repos are on version 2.9, while pip is looking for 2.10.1. The installed python 3 QT5 packages are version 5.6, which might also be having an impact. Can pip install later versions of such packages alongside distro provided versions? I've got dependencies on the some of the core python3 qt5 libs which makes them difficult to remove without breaking everything.

ntoll, 2017-11-28:
@tim1mw can I just check that you're trying to install in a Python virtualenv rather than with system Python.

ntoll, 2017-11-28:
If you're not using a virtualenv, can you please try using pip within such a virtualenv and paste what happens?

Also, just checking this is Linux on a i386 or amd64 right?

carlosperate, 2017-11-28:
Just a general note, since pip will look for packages in PyPI (https://pypi.org/), and Mu has the version hardcoded in the setup.py file this is where pip will be looking for a package compatible with your system:
https://pypi.org/project/QScintilla/2.10.1/#files

In this case this file should have been found as compatible: `QScintilla-2.10.1-5.9.1-cp35.cp36.cp37-abi3-manylinux1_x86_64.whl`
Not quite sure why it is not found in your case:
>Collecting qscintilla==2.10.1 (from mu-editor)
Could not find a version that satisfies the requirement qscintilla==2.10.1 (from mu-editor) (from versions: )
No matching distribution found for qscintilla==2.10.1 (from mu-editor)

As a workaround you could download the wheel and install manually (`pip install mywheel.whl`), but I'd be more interested to find out why it is not automatically detected and downloaded.
Oh and as @ntoll mentioned, a virtualenv would be highly recommended for these tests as well,

tjguk, 2017-11-28:
Could well be the version of pip. The OP is using 8.0.2 but IIRC a lot 
of versioning rules change in 9.x.

ntoll, 2017-11-28:
@carlosperate I wondered if Mageia's pip was looking in the right place. Alternatively, it could be a case of "the network isn't reliable" biting @tim1mw.
@tjguk that's a really good point. @tim1mw can you ensure you're using a recent version of pip?

Interestingly, we've had *zero* reports of this pip related problem from anyone else which suggests to my mind this could be something isolated to Tim's setup (uncommon Linux distro etc). Having said that, pip *should* work across Linux distros and IIRC Mageia is based on Fedora (??) so shouldn't be *that* weird or "out there" in terms of setup.

tim1mw, 2017-11-28:
It's amd64. I was using the system python, but I just tried a virtual environment and that worked, so it's clearly some kind of issue with the older system libs and pip not working together.

I just tried running python3 run.py from the source code checkout and that starts up using the normal system libraries, whether it crashes when you try to use it properly, I can say. So I now have two working options, which solves my immediate problem.

tim1mw, 2017-11-28:
@ntoll I'd prefer not to upgrade the system pip outside of the normal package process, however, I did install the latest pip into the virtual environment test which worked.

ntoll, 2017-11-28:
Aha... OK. so this is a configuration problem rather than Mu specific (i.e. any PyQt5 application may run into this problem). Glad you've got a working version (or two) now.

As a result, I'll mark this issue as closed. Thanks for the feedback!

closed 2017-11-28 by ntoll