mu-editor/mu #304: Mu misbehaves with multiple tabs

rhubarbdog, 2017-11-25:
I had 2 programs open in Mu hello_world.py and quiz.py i edited quiz.py closed it/hello_world.py and some how had the tab label quiz.py * on the source code for hello_world.py.  I'm trying to reproduce this, whist doing so i've found the following:

How ever i can reproduce this 
1) open 2 programs hello_world.py and quiz.py
2) edit quiz.py the tab is now quiz.py *
3) Hover over the cross on hello_world.py and quiz.py autosaves

rhubarbdog, 2017-11-25:
What I am reproducing is Mu autosaves every few seconds i'm still trying to get Mu to change names on tabs. As outlined in the next issue this autosave is rediculous

rhubarbdog, 2017-11-25:
I can reproduce this 
1) open 2 programs hello_world.py and quiz.py
2) edit quiz.py
3) click close tab on hello_world.py

dialogue "There is un-saved work, closing the tab will cause you to lose it." pop up

ntoll, 2017-11-27:
Which version of Mu?

rhubarbdog, 2017-11-27:
Version 1.0.0.beta.12

ntoll, 2018-03-21:
Fixed by 674005ff4f6192448fc7648b8955fd71eb7f92a0

Thanks for reporting the bug!

closed 2018-03-21 by ntoll