mu-editor/mu #305: Edited code is autosaved on Quit or closing the tab

rhubarbdog, 2017-11-25:
When exitting or closing a tab a dialogue should be used saying "program xxx.py has been edited do you want to save these edits?" 3 buttons YES NO CANCEL

rhubarbdog, 2017-11-25:
you do have a dialogue, the problem is Mu autosaves every few seconds.  I use emacs to enter all my coding  tab and backspace automatically change the indent correctly, I only use Mu for REPL and flashing this causes me loads of issues.  This autosave feature is firstly incorrect secondly dangerous as my edits may be rubbish and the autosave will leave me with a previously working program now in a  mess.

ntoll, 2017-11-27:
Hi @rhubarbdog.

Oh boy... I've had so many thoughts about the way this works. Teachers have consistently asked that Mu auto-saves work. Kids are prone to forget to save their work and this causes no end of problems when using the micro:bit.

However, as a developer, I'm with you: I'd rather explicitly save for precisely the reasons you state (avoidance of incomplete or semi-broken code).

Nevertheless, beginner programmers, learners and teachers are Mu's primary audience so I'd be wrong to ignore their (consistent) feedback. How about a flag in the settings file so that "advanced" users could override the autosave function..?

rhubarbdog, 2017-11-27:
A flag or configuration file in a users home folder would be the way forward, im happy to be ignored as i only really us Mu for flashing, file transfer and repl

ntoll, 2018-03-21:
Hi @rhubarbdog, I'm going to close this since more recent teacher-y feedback has reinforced their desire for auto-save. :-)

closed 2018-03-21 by ntoll