mu-editor/mu #306: Help button does not work

rhubarbdog, 2017-11-25:
Help button correctly opens a tab in my current firefox, how ever it does not switch me to firefox when my browser is on another workspace (all ok when Mu and firefox on the same workspace).  Making a workspace switch may be beyond the scope of the editor (impossible/unpractical) but a dialogue/message at the bottom of the window directing to your browser may be beneficial 

How ever in both cases the web page
https://codewith.mu/en/help/1.0
Yeilds
"Sorry this page does not exist =( "

ZanderBrown, 2017-11-26:
What DE are you using?

I know GNOME on Wayland currently has some bugs in this area (due to be fixed in 3.28 IIRC)

As this link contains `1.0` I'm assuming your running master which is unreleased software and as such help hasn't been published yet

rhubarbdog, 2017-11-26:
I'm running Linux mint Mate desktop which is gnome/gtk.
I am running from the repoistory version 1.0.0.beta.12 i couldn't find any working instructions tried codewith.mu and others

ntoll, 2017-11-27:
Hi @rhubarbdog  - indeed, that web page doesn't exist because I'm still in the middle of writing up the documentation and website. This *will* be fixed before the end of the beta period.

closed 2017-11-27 by ntoll