mu-editor/mu #309: REPL issues on Windows

BravoDelta151, 2017-12-01:
This might be two issues.  

On Windows 10 using Adafruits Circuit Playground Express and Trinket M0.  Both were identical.  On initial installation, the REPL was not working at all and would lock up the COM port so that other programs (putty and tera term) could not interact with the repl either.  Tera Term has a reset port function that worked at least one time.  Other times, I had to unplug/plug back in the device and hit its reset button a couple times to even get putty or tera term to work.  Once I did that, I could open mu and the repl worked correctly until I hit reset on the device or unplugged it (see below).

I am able to reproduce this behavior by unplugging the device with mu open or by pressing the reset button on the device.  One of the users on Adafruits discord chat tested this on Linux and Mac and didn't have the same problem.

I suspect it might be related to this Qt issue https://bugreports.qt.io/browse/QTBUG-50052 although I haven't confirmed whether PyQt5.9 would have this or not.

ntoll, 2017-12-01:
Hi @BravoDelta151, thank you for the feedback. I wonder, could it be something fixed by #303, a work around for QtSerial problems on Windows? This has landed in master branch awaiting an upcoming beta.13 release if you want to try it out "early".

BravoDelta151, 2017-12-01:
This might be fixed with issue [https://github.com/mu-editor/mu/issues/281], I did not see that issue before and I know that is not in the code I built to test this.

BravoDelta151, 2017-12-01:
It looks like this is a duplicate of Issue 303/281.  The changes for 281 look to have fixed the issue. 👍

carlosperate, 2018-03-17:
Sounds like the fix is already shipped in the latests betas, so I'll close this one, but feel free to reopen if there still a problem. Thanks for the report!

closed 2018-03-17 by carlosperate