mu-editor/mu #318: Investigate flatpak packaging

ZanderBrown, 2018-01-09:
We have many issues around the Linux binary (#299, #279), perhaps some of these could be resolved by providing a [flatpak](https://flatpak.org/) probably submitted to [flathub](https://www.flathub.org/). This way we could 'lower the barrier of entry' as mu would be available in package managers like gnome-software by simple searching for Mu (or 'python')

Obviously, this would require us to write an appdata.xml file and have a stable release (flathub only accepts stable releases) but combined with #314 this could make it far easier for people to get there hands on Mu

I believe flathub offers a range of build servers allowing us to provide x86/x86_64/ARM/ARM64 flatpaks instead of a single x86_64 binary

carlosperate, 2018-01-10:
The main issues encountered with the Linux build is is that the compatibility depend on the build environment (32/64 bits and glibc versions mostly). Last time I checked flatpak still has the same problems, so I in that respect I don't think it offers any advantages to the current implementation.

That being said, I didn't know flathub offered build servers, that would allow 32 and 64 bit versions of the application (and ARM, which is good, but for the Pi users will be encouraged to apt-get), which is good,  and I assume the OS/libraries in those are probably old enough to make it as compatible as possible.

On the other hand, I was honestly thinking about perhaps retire the linux .bin files. They are barely used and there will be debian packages, and likely rpm packages as well available. Between that and being able to pip install, it should be able to cover the majority of cases.

ZanderBrown, 2018-01-10:
Flatpaks are built against a specific 'runtime' version so other than kernel there are limited variables from one system to the next hopefully removing glibc issues ext

For the Pi deb would definitely be the preferred option but for others (deb or rpm) flatpak would be great to offer

ntoll, 2018-01-10:
Since we're thinking about packaging on Linux... I've been looking at "snap" (https://docs.snapcraft.io/build-snaps/python). Any thoughts on this..?

ZanderBrown, 2018-01-10:
I don't personally know much about snaps but they don't seem to have quite the same benefits as flatpak and are much more linked to Canonical/Ubuntu than flatpak is with Red Hat/Fedora, still they do seem to have a nice tutorial there for creating a snap out of `pip/setup.py/requirements.txt` packages

ZanderBrown, 2018-01-20:
Relevant: [flatpak-builder-tooks](https://github.com/flatpak/flatpak-builder-tools/tree/master/pip)

ntoll, 2018-03-21:
I think this won't be going anywhere unless snap or flatpak are explicitly requested by users (they're not currently).

closed 2018-03-21 by ntoll

RasheedAZ, 2023-11-05:
> I think this won't be going anywhere unless snap or flatpak are explicitly requested by users (they're not currently).

Hi. Can Mu-Editor be packaged as a Flatpak and published on Flathub please? 

Thanks :)