mu-editor/mu #320: Starting new translation language results with Error 127

ricco386, 2018-01-16:
If you're not a technical person, please fill in the different sections of the issue template 
copied below. If you are technical and feel comfortable submitting a useful and comprehensively 
explained issue, feel free to ignore this template.

In any case, thank you for your contribution. :-)

What were you doing?
--------------------

Please list the steps we need to follow to recreate the issue you're reporting.

* I was using Mu version 1.0.0.beta.13 (commit: 51cacf5139b4564b14bf904a4a12c79335415b64) on Linux Fedora 27.
* I have cloned and checkout the repository: `git clone git@github.com:pyconsk/mu.git; cd mu`
* I have created virtual environments and activate: `python3 -m venv envs3; source envs3/bin/activate`
* I have installed requirements.txt: `pip install -r requirements.txt`
* I wanted to create an up-to-date messages.pot file: `make translate`

What did you expect to happen?
------------------------------

I was expecting  messages.pot to be generated by the script

What actually happened?
-----------------------

I got an error message instead:
```
find . \( -name _build -o -name var -o -path ./docs -o -path ./mu/contrib -o -path ./utils -o -path ./mu/modes/api \) -type d -prune -o -name '*.py' -print0 | xargs -0 -r pygettext
xargs: pygettext: No such file or directory
make: *** [Makefile:72: translate] Error 127
```

Why is this problematic?
------------------------

I am not able to create messages.pot file that is required in order to be able to generate the new translation.

Attach Logs!
------------

```
ls -l ~/.cache/mu/log/mu.log 
ls: cannot access '/home/ricco/.cache/mu/log/mu.log': No such file or directory
```

ricco386, 2018-01-16:
I have found a temporary fix:

I had to install package `python3-tools` that provides `pygettext`, but still is not executable after install. Once I have modified Makefile with path `/usr/lib64/python3.6/Tools/i18n/pygettext.py` make command worked and generated pot file.

ntoll, 2018-01-16:
Hi @ricco386 -- good to hear from you and thank you for reporting this bug..! ;-)

As I understand it, this is a problem created by the operating system's Python packaging. The Python docs (see the mention of `pygettext` in this section: https://docs.python.org/3.6/library/gettext.html#internationalizing-your-programs-and-modules) indicates that the `pygettext.py` command is a part of Python.

I guess the only solution is to flag this in the documentation so people don't trip up with this again and/or have some sign-posting to fix their problem.

ntoll, 2018-01-16:
Documentation based fix in 25adb00

closed 2018-01-16 by ntoll