mu-editor/mu #326: No mu executable after pip3 install on (L)ubuntu

cefn, 2018-02-09:
Having apparently installed mu successfully on a standard Lubuntu 17.10 using...

`pip3 install mu_editor`

...following instructions at https://learn.adafruit.com/adafruit-feather-m0-express-designed-for-circuit-python-circuitpython/installing-mu-editor I was expecting there to be an executable `mu` available. However, there was no `mu` on my path after the install.

As a workaround before tracking down the installed binary I found by experiment that `python3 -c "from mu.app import run; run()"` launched the installed mu-editor.

It turns out there was a mu binary added to the path .local/bin/mu (with a very similar invocation embedded in it). However, this wasn't available to my shell PATH by default. See https://bugs.launchpad.net/ubuntu/+source/bash/+bug/1588562 for the upstream issue.

Can I propose that a final 'announcement' line is added to the build to indicate where the mu binary was as I had no idea how to launch it without investigating in a lot of detail. Alternatively provide instructions for how to launch it without relying on tracking down and wiring up a suitable binary PATH.

ntoll, 2018-02-09:
@cefn ouch..! Sorry to hear you've encountered problems.

Hmm... at the moment the recommended way (by me) to install Mu is to do it in a virtualenv as per the instructions here: https://mu.readthedocs.io/en/latest/setup.html

This should work in all cases on Windows, OSX and Linux.

I notice from the Ubuntu bug report, that the actual cause a mis-configuration in the default PATH (which is something I have no control over) and such things are in hand to make sure pip will work as expected in future versions of the OS.

I'm interested to know what you mean by an `announcement` line. The file to which you refer is automatically created by pip/Python and configured in the `setup.py` file under `entry_points` (i.e. I just specify something and Python does the rest).

Ultimately, when the Ubuntu based bug is fixed, you'll never see this problem again.

ntoll, 2018-02-09:
Apologies... I linked you to the developer setup instructions. The (far simpler) way to install Mu is to create a virtualenv and `pip install mu-editor`. The `mu` executable will definitely be on your path in that case.

cefn, 2018-02-10:
I take your point that it's an Ubuntu bug. Astonishingly it looks like issue affects the previous two versions of Ubuntu! 

However it's a moot point since Mu is targeting non-expert users, and the .bin intended for users to install Mu has started failing on Ubuntu and apparently can't be fixed (my experience of https://github.com/mu-editor/mu/issues/299 is how I arrived here in the first place).

> virtualenv should work in all cases on Windows, OSX and Linux.

Of course, you have a chicken and egg situation here (assuming you install virtualenv through pip3 on Ubuntu 17.04 or 17.10). It installs a binary at .local/bin/virtualenv which is not on the path, so you (if you are a novice user) can't run virtualenv after installing it. 

Possibly you could install via `sudo apt install virtualenv` but I am superstitious about combining apt and pip to manage python packages and I am uncertain whether this should be advised if it can conflict later with pip3 installs.

If you felt it was reasonable to add to your documentation a fallback of running `python3 -c "from mu.app import run; run()"` from the console I think that should indeed work in all cases since pip3 ensures that its packages are on the PYTHONPATH. 

For the long run, this console invocation could be made less complicated by adding a package main, ( https://docs.python.org/3/library/__main__.html ) so that mu could be invoked like...

`python -m mu`

The idea of the announcement was to report something like this to the console during installation...

`mu is now installed at .local/bin/mu`

cefn, 2018-02-16:
Thanks for https://github.com/mu-editor/mu/pull/333 ! 

Only remaining part of issue is to document the console launch via `python3 -m mu` as a launch option in the getting started documentation.

ntoll, 2018-02-17:
Docs added in 8eeed96f32f8858164345abf106120871df02a32

closed 2018-02-17 by ntoll

PTS93, 2018-06-20:
same issue here and I'm not entirely sure what the official way of installing under linux is now. Should I create this virtual environment myself or should `pip3 install mu_editor` be enough?
using pip3 I definitely don't get any binary in my path

ntoll, 2018-06-20:
Hi @PTS93 the official version is described here: https://codewith.mu/en/howto/install_with_python (but I notice I don't mention virtualenv -- I'll fix that).

Yes, please use a virtualenv..! :-)

ntoll, 2018-06-20:
@PTS93 BTW, Linux packaging is described here (it's a work in progress): https://mu.readthedocs.io/en/latest/packaging.html#linux-packages

cefn, 2018-06-20:
There's a utility called pybin which is potentially useful to fix this narrow issue (no binaries on your path) if you don't want a virtualenv.

It's generally a useful thing anyway, as you'll encounter it with any other pip-installed binaries too, including virtualenv!

See https://vgkits.org/blog/pip3-config-howto/

dhalbert, 2018-06-20:
I have been use `pip3 install --user`.

PTS93, 2018-06-20:
Following the guide on the how to install with python page lands me in the situation of not having a binary in my path. I tried `pip3 install --user` now as well but same outcome unfortunately.
I'm a bit confused about why this is the case though, what is the need for the virtualenv if apparently all that is needed is that there is a symlink to a place like `/usr/bin`? I would do that myself but I have the same issue as op that I can't find where the binary is being placed, on Ubuntu 17.10 it doesn't seem to land in `~/.local/bin/`

ntoll, 2018-06-21:
Glad you found it.