mu-editor/mu #327: Random (randint) without being seeded always returns 0

MrYsLab, 2018-02-10:
This occurs with 1.0.0.beta.14.
This does not occur with https://python.microbit.org/v/1.
It is related to https://github.com/bbcmicrobit/micropython/issues/491

If you run this code using mu, randint always returns 0.
If you uncomment out the seed, randint functions properly.

```
from microbit import *
from random import randint, seed

# seed(3)
for x in range(10):
    print(randint(0, 60))
```

MrYsLab, 2018-02-12:
Just want to make sure that it is tested everywhere when the MicroPython updates are integrated. Thanks.

ntoll, 2018-03-21:
This has been fixed in upstream MicroPython, and the change has landed in Mu's `master` branch. It'll be available in beta.16 (coming in the next two days).

closed 2018-03-21 by ntoll