mu-editor/mu #328: why not using system colors?

reithi, 2018-02-11:
I am a visually impaired user and I am using different color settings then the default user. I need always  to code in night mode (dark background and light text) because the other ways round I see nearly nothing.
Meanwhile many Smartphone Apps implement that feature because its much easier to see even for normal users. 
I wonder why the application isn't using my qtsettings which all the other programs are using.
Would be very helpful if it would.
Thx a lot!!

ZanderBrown, 2018-07-16:
Unfortunately as we have fairly extensive theming to achieve a consistent look across platforms (which I think is important for a learners tool) I'm not sure this will be practical.

Would love some feedback on our high-contrast theme though

ntoll, 2018-07-16:
@reithi what @ZanderBrown says.

To put this into context... the feedback we got from teachers is that they use single resources over several platforms, ergo the consistency of the look is important for them (beginners, who may be lacking in confidence, may not know what to do as a result of a trivial [to us] platform specific difference in layout).

Does this make sense? Please do let us know your thoughts, ideas and feedback.

tmontes, 2019-01-27:
Possible idea, dependent on #693:
* At least on Windows (and maybe Linux? not sure) two "start menu" entries could be created: one "default mu" that starts Mu with the default day theme, and a second one "dark mu" which people like @reithi would have probably noticed and used from the start.

dybber, 2021-02-19:
I like the idea of having two different start-menu entries, one just called "Mu", and another called "Mu (High contrast)" or similar. I guess it's the High contrast theme that would be helpful for users like @reithi.

dybber, 2021-02-19:
Oh, good find! That should be doable, I can make a PR and we can get people to test it on various systems. I guess if we can get it to work on Mac, Windows and Gnome that will be good enough for a "best effort", everyone else will have to switch manually. As Mu remembers the theme settings, this will only be necessary to do on the first run of Mu.

ZanderBrown, 2021-02-28:
> or "Gnome"

Oh no

dybber, 2021-03-01:
> > or "Gnome"
> 
> Oh no

Woops! Edited :-)