mu-editor/mu #329: Raspberry pi - development build

martinohanlon, 2018-02-12:
I am trying to install Mu from source on a Raspberry Pi following the instructions on [https://mu.readthedocs.io/en/latest/setup.html#raspberry-pi](https://mu.readthedocs.io/en/latest/setup.html#raspberry-pi)

The following runs successfully:

```
sudo apt-get install python3-pyqt5 python3-pyqt5.qsci python3-pyqt5.qtserialport python3-dev
sudo pip3 install -r requirements_pi.txt
```

When I try and run mu I receive the following error:

ntoll, 2018-02-12:
You've not allowed for site packages to be available in your virtualenv. ;-) (See step two here: https://mu.readthedocs.io/en/latest/setup.html#raspberry-pi)

martinohanlon, 2018-02-12:
I'm not using a virtual environment. Do I need too?

ntoll, 2018-02-12:
Of course..! :-) That's what the instructions say...! :-P

carlosperate, 2018-02-12:
This is likely to be related to https://github.com/mu-editor/mu/pull/312.
Try installing python3-pyqt5.qtsvg as well.

martinohanlon, 2018-02-12:
This issue was a result of problems with the Raspberry Pi setup and a missing dependency `python3-pyqt5.qtsvg` 

PR #331 submitted.

martinohanlon, 2018-03-02:
Resolved.

closed 2018-03-02 by martinohanlon