mu-editor/mu #330: Windows install cant use 'installed' packages

martinohanlon, 2018-02-12:
OS - Windows 10
Installed using Windows 64 bit installer. https://s3-eu-west-2.amazonaws.com/mu-builds/windows/mu_2018-02-10_19_12_master_3893ae7_64bit.exe

When I try to import packages I have installed via pip (e.g. guizero), it cant find them.

I know they are installed as the Python REPL will happily import them (see below).

[image]

ntoll, 2018-02-12:
Doh... my fault for not explaining things clearly enough. As per the packaging docs: the version of Python used by Mu and the "system" Python are different. That's why you're not seeing them from Mu.

ntoll, 2018-02-12:
It's for this reason that people can install Mu and NOT have Python installed on Windows.... Mu comes with its own version that is "sandboxed" from everywhere else.

The good news is that we can tell the Mu installer to include the Windows version of GPIOZero so it's available to Mu's version of Python.

Does that make sense?

martinohanlon, 2018-02-12:
Ok how do I get Mu to use other packages?

ntoll, 2018-02-12:
We agree before I cut a release, which packages should be included in the Mu version of Python. Does this make sense..?

ntoll, 2018-02-15:
As discussed, this is by design.

closed 2018-02-15 by ntoll