mu-editor/mu #334: Create a PyGame Zero mode

bennuttall, 2018-02-13:
How easy would it be to create a PyGame Zero mode? I'd be happy to have a go with some guidance.

The way PyGame Zero works is you write your code in an editor, and run it from the command line with `pgzrun`.

Things to consider:

- Assuming it doesn't get bundled with Mu, pgzero may not be installed (it is pre-installed on Raspbian)
- Since a pgzero script does not have to be a valid Python file (i.e. pgzero imports are not necessary) the checker would not be accurate without knowledge of the pgzero globals (MVP could remove the checker, but a fully featured version could add it back in with a sophisticated pgzero aware checker)

FAO @lordmauve

tjguk, 2018-02-13:
@bennuttall are you suggesting that mu actually *run* the pgzero (ie by invoking pgzrun)? Or simply that it not be confused by the lack of imports etc.

bennuttall, 2018-02-13:
Run!

ntoll, 2018-02-13:
@bennuttall this is on my radar for this week. Great minds think alike (or fools seldom differ). ;-)

ntoll, 2018-02-15:
See the work taking place on the `pygamezero` branch.

bennuttall, 2018-02-15:
This is great! It works very well 👍

ntoll, 2018-02-15:
Great stuff... it was hacked together in about 45mins last night. 

ToDo:

- [ ] Tests
- [ ] Docs
- [ ] Developer docs description for how I implemented it (so others can follow the process)
- [ ] Add buttons to show images and sounds used by PyGameZero (so users can just copy their own game assets over to the right place)
- [ ] Create the default images and sounds from the PyGameZero tutorial in the expected location on the file system.
- [ ] Add API autocomplete / call tips.
- [ ] Make sure code checking works as expected for implied names in PyGameZero namespace.

ntoll, 2018-02-15:
First working Windows installer here: https://s3-eu-west-2.amazonaws.com/mu-builds/windows/mu_2018-02-15_20_49_pygamezero_5861bdc_64bit.exe

ntoll, 2018-02-16:
Landed in master 53cd0fc6d67921103429f4185835dcb828174850

Updated docs here: https://mu.readthedocs.io/en/latest/index.html (including using this work as an example for creating a new mode in Mu).

64bit Windows build here: https://s3-eu-west-2.amazonaws.com/mu-builds/windows/mu_2018-02-16_20_02_master_53cd0fc_64bit.exe

YouTube video of a quick demo: https://www.youtube.com/watch?v=jmIK3XfSvNI

closed 2018-02-16 by ntoll

lordmauve, 2018-02-17:
This is super exciting! All this Pygame and Mu development is making me want to get a new release of Pygame Zero out.

bennuttall, 2018-02-20:
:1st_place_medal:

yonghuming, 2018-03-09:
@lordmauve hmmm, i think pyzero need some ui for sudents to input some info, eg: input textbox, pygzero is a game engine and it need ui framework to make it more comfortable to use