mu-editor/mu #336: mu-0.9.13.exe fails to run on Windows 10

xiebochao, 2018-02-14:
msvcp100.dll could not be extracted

ntoll, 2018-02-14:
Hi,

Could you supply a bit more information? I just Googled your error message and the single result (!) is to a web page that returns a 500 error for me.

Is the installer able to copy over any other files..? Could this be a file-system permissions based issue? Are you trying to install something without the required privileges? Do you have a special setup of Windows 10 (or perhaps a virus checker/scanner) that means such installations are managed or interfered with in some way (for example, some corporate controlled versions of Windows limit what can be installed)..?

Best wishes,

N.

carlosperate, 2018-02-14:
@ntoll I think the issue is about v0.9.13, so I assume @xiebochao has tried to use this executable: https://github.com/mu-editor/mu/releases/download/v0.9.13/mu-0.9.13.win.exe
@xiebochao Could you confirm if that is the program you were trying to execute, and if it isn't could you try to execute this one?

The listed missing msvcp100.dll is part of a Microsoft Visual C++ 2010 Redistributable Package. You can try to install it in your system from the links included by Microsoft here, but it shoudn't be neccesary: https://answers.microsoft.com/en-us/windows/forum/windows_other-performance/msvcp100dll-missing/9a687c31-0619-4ee9-b511-020985e29b5f

This DLL file should be included with Mu.exe. I looked at the uncompressed temporary folder created when Mu.exe is executed and it is there:

```
C:\Users\carlos\AppData\Local\Temp\_MEI27962>tree /f
Folder PATH listing for volume BOOTCAMP
Volume serial number is 0246-9FF7
C:.
│   base_library.zip
│   icudt53.dll
│   icuin53.dll
│   icuuc53.dll
│   mfc100u.dll
│   MSVCP100.dll
│   MSVCR100.dll
│   mu.exe.manifest
│   pyexpat.pyd
│   PyQt5.Qsci.pyd
│   PyQt5.Qt.pyd
│   PyQt5.QtCore.pyd
│   PyQt5.QtGui.pyd
│   PyQt5.QtPrintSupport.pyd
│   PyQt5.QtSerialPort.pyd
│   PyQt5.QtWidgets.pyd
│   python34.dll
│   pythoncom34.dll
│   pywintypes34.dll
│   qscintilla2.dll
│   Qt5Core.dll
│   Qt5Gui.dll
│   Qt5PrintSupport.dll
│   Qt5SerialPort.dll
│   Qt5Svg.dll
│   Qt5Widgets.dll
│   select.pyd
│   sip.pyd
│   unicodedata.pyd
│   win32api.pyd
│   win32com.shell.shell.pyd
│   win32trace.pyd
│   win32ui.pyd
│   _bz2.pyd
│   _ctypes.pyd
│   _hashlib.pyd
│   _lzma.pyd
│   _multiprocessing.pyd
│   _socket.pyd
│   _ssl.pyd
│   _win32sysloader.pyd
│
├───Include
│       pyconfig.h
│
├───mu
│   └───resources
│       ├───css
│       │       day.css
│       │       night.css
│       │
│       ├───fonts
│       │       LICENSE.txt
│       │       SourceCodePro-Bold.otf
│       │       SourceCodePro-BoldIt.otf
│       │       SourceCodePro-It.otf
│       │       SourceCodePro-Regular.otf
│       │       SourceCodePro-Semibold.otf
│       │       SourceCodePro-SemiboldIt.otf
│       │
│       └───images
│               check.png
│               files.png
│               flash.png
│               help.png
│               icon.png
│               load.png
│               new.png
│               quit.png
│               repl.png
│               save.png
│               snippets.png
│               theme.png
│               theme_day.png
│               zoom-in.png
│               zoom-out.png
│
└───qt5_plugins
    ├───iconengines
    │       qsvgicon.dll
    │
    ├───imageformats
    │       qdds.dll
    │       qgif.dll
    │       qicns.dll
    │       qico.dll
    │       qjp2.dll
    │       qjpeg.dll
    │       qmng.dll
    │       qsvg.dll
    │       qtga.dll
    │       qtiff.dll
    │       qwbmp.dll
    │       qwebp.dll
    │
    └───platforms
            qminimal.dll
            qoffscreen.dll
            qwindows.dll
```

So maybe the file had some corruption and could not uncompressed all the included files?

ntoll, 2018-02-14:
Ugh... apologies... I saw the digits "13" at the end of the version number and misread beta 13. Ignore everything I just said. That'll teach me to stay up until 3am watching the Olympics. :-)

carlosperate, 2018-03-17:
Hi @xiebochao, were you able to download the executable and try again?

ntoll, 2018-03-21:
Hi, I'm closing this because Windows users should use the Windows installer and the 0.9.13 exe is a dead end.

closed 2018-03-21 by ntoll