mu-editor/mu #340: Windows installer - what python libraries to 'package'

martinohanlon, 2018-02-14:
A new windows installer has been created for Mu which packages up Python and create a single installer to allow those in restricted environments (i.e. schools) to "get started with python programming using mu". All good stuff.

A constraint of taking this approach is that you also need to package the Python libraries that a school might want to use into the installer as well as the self contained Mu cant use system installed packages.

_Its worth noting that the preferred method of installing mu is to use pip so you get the full power of python and all of the flexibility but it is simply not achievable for many._

So... What libraries to package?  Please insert your views here.

martinohanlon, 2018-02-14:
To kick us off:

`tkinter`
`turtle`
`gpiozero`
`guizero`
`pygame`
`pygamezero`
`networkzero`

martinohanlon, 2018-02-14:
@whaleygeek @bennuttall @waveform80 @davidhoness @MarcScott @rikcross @jrobinson-uk @mrichardson23 @tracygardner @caitlynraspberrypi @topshed @lhyams 

tagging you in, in an attempt to get a wider set of views.

DavidWhaleMEF, 2018-02-14:
```
bitio
mcpi
bottle
py-sonicpy
```

bennuttall, 2018-02-14:
@DavidWhaleMEF What is `py-sonicpy`? Do you mean `python-sonic`? https://pypi.python.org/pypi/python-sonic

I think just using `python-osc` is preferred now it's fully supported in Sonic Pi.

bennuttall, 2018-02-14:
Suggestions:

- `flask`
- `scratch`
- `sense_emu`
- `pigpio` (to accompany `gpiozero`)
- `bluedot`

mrichardson23, 2018-02-14:
I'm a big fan of and use `requests` a ton for projects. Would it be helpful here as well?

martinohanlon, 2018-02-14:
@bennuttall does the `sense_emu` run on Windows?

MarcScott, 2018-02-14:
guizero

<https://github.com/mu-editor/mu/issues/340#issuecomment-365659380>, or mute
the thread

bennuttall, 2018-02-14:
> @bennuttall does the sense_emu run on Windows?

I believe so. You're in a better position than me to try it :)

martinohanlon, 2018-02-14:
> I believe so. You're in a better position than me to try it :)

I am, and it doesn't!

ZanderBrown, 2018-02-14:
sense_emu uses pygobject I believe and should run on Windows but as mu is Qt based (rather than Gtk) any installer would be rather large to include the required DLLs

waveform80, 2018-02-15:
sense_emu *does* run on windows (I tested on linux and windows while writing it) but installing GTK3 is an absolute pig and requires very specific versions of several things (also, I've no idea how easy it would be to integrate its installer with other things). Unfortunately Windows support in GTK has always felt very "oh, if we really have to" (in rather stark contrast to Qt).

On the more general topic, do you want some more "common" libraries included?

* PIL/Pillow for image loading/saving/manipulating
* numpy is practically universal these days (and plays nicely with Pillow for performing simple mathematical operations on images)
* matplotlib (for graphing anything mathematical from numpy, or even for certain types of image manipulation)
* requests (if you want to enable kids to easily HTTP-GET web-pages or interact with web services)

P.S. I thought tkinter was included in the base python package (and turtle)? (I know Debian splits out tk for dependency reasons, but I was fairly sure it was included by default on other platforms).

ntoll, 2018-02-15:
Hi Folks,

This is great and thank you for all the suggestions. 

However, there's a danger of including everything but the kitchen sink. Just so we have some sense of scope -- we're looking for Python modules that can be installed on Windows 32 and 64 bit via pip (i.e. there are existing wheels in PyPI).

When I saw the mention of GTK on Windows I snorted my tea over my laptop. There's definitely a cost/benefit analysis to do here ;-) Basically, let's go for the easy low-hanging fruit and anything that requires more onerous dependencies (such as a whole new GUI framework) we can debate once we've a "first cut". ;-)

I'm very enthusiastic to use Mu's new Windows installer as a way to "liberally" piggy-back other useful Python based modules, especially if they're the basis of Raspberry Pi learning resources, but I also want to get it right for the easy stuff before we tackle more complicated nice-to-have packages.

Does this make sense? Comments welcome! :-)

N.

waveform80, 2018-02-15:
Yup, perfect sense, and I'd second that GTK on Windows is likely more
trouble than it's worth. I'd suggest numpy and Pillow are likely used in
several learning resources, not sure about requests or matplotlib (and the
latter can get horribly large with its GUI options) so they could be chopped

Dave.

ZanderBrown, 2018-02-15:
It'd be great if Gtk3 was included but it would take a reasonable amount of work, perhaps somebody from the Gtk team could advise

ntoll, 2018-02-15:
@ZanderBrown so you inadvertently touch on an important point.... which version..? Raspbian uses GTK2 yet the latest is GTK3. I've never used them above and beyond "hello world" type essays in the art just to get a feel for how it works, but I assume versions 2 and 3 have different APIs (otherwise, why the difference in version number?).

Right now, I'm erring on not touching GTK with a barge-pole simply because it's too complicated with very little in return IYSWIM. ;-)

ntoll, 2018-02-16:
After discussions with @martinohanlon, these will be the initial Python packages included in the Mu installer:

* Tk
* Turtle
* Pygame
* GPIOZero
* PiGPIO
* GuiZero
* Pillow
* Requests
* PyGameZero

martinohanlon, 2018-02-16:
For clarity, we should note that:

- matplotlib
- numpy

Are already included as there are requirements for Mu itself.

martinohanlon, 2018-02-16:
Those which are not included are done so for a number of reasons, mainly due to:
- them not being packaged as wheels
- requiring additional resources (installed exe, webservers, massive gui libraries)
- being out of date / obsolete

ZanderBrown, 2018-02-17:
Unfortunately from the python perspective 2 Vs 3 is very different but as far as they are native 2&2 and 3&3 have inadvertently become locked so to speak (at the C level they are far more API compatible but different ABI and other internals)

It's true the Raspbian LXDE DE is largely Gtk2 but things like the Add / Remove Software,  sense_emu & Calculator are Gtk3 as are most other DE for the Pi

From the wider Linux perspective Gtk3 is Wayland native and the binding is achieved through PyGObject giving support for various other libs 'for free' so in that sense is the obvious candidate

Is there any possibility of having an 'extras' pack for these GObject based things? Things like sense_emu would be great to have available but the extra deps would make the install quite large when it's not needed by many. This would also allow us to make it available at a later date

carlosperate, 2018-02-20:
Out of curiosity, what's the plan for these dependencies with the deb installer for Raspbian? Are they all going to be installed from apt packages in the global python3 environment?

ntoll, 2018-02-20:
@carlosperate that would be the most sensible thing to do AFAICT.

jrobinson-uk, 2018-02-20:
Ditto, python Osc should be used

carlosperate, 2018-02-20:
@ntoll as this issue is for the Windows packages, and my previous question is likely to divert into a parallel conversation, I've opened https://github.com/mu-editor/mu/issues/348.

ntoll, 2018-02-20:
@carlosperate +1 ;-)

yonghuming, 2018-03-07:
@ntoll i have read all the souce of python turtle and in it it say you can use any GUI framework replace tkinter, thus mu is writed using pyqt, why not relace the GUI of python turtle use pyqt?

waveform80, 2018-03-07:
> @ntoll i have read all the souce of python turtle and in it it say you can use any GUI framework replace tkinter, thus mu is writed using pyqt, why not relace the GUI of python turtle use pyqt?

Because, whilst it is possible, it's a lot more work than it looks. The GUI interface stuff doesn't look terribly long in `turtle.py` but it's relying on a fair bit of inherited functionality (and some wonderfully dodgy hacks: take a look at `__forwardmethods`). Porting it to another framework basically means writing a compatibility layer for Tk's Canvas and Frame classes (and the main loop) in Qt. It can be done, but this isn't a quick "do it in a day" job.

ntoll, 2018-03-07:
Heh... *anything* is possible (given enough time and money). :-)

So, while @yonghuming's suggestion is actually VERY interesting, for the purposes of Mu *at this moment in time* the cost of time / effort isn't viable. However, I'd love to be able to embed `turtle` into a Qt application. Perhaps this is a new module in the waiting (so others can re-use the effort).

tjguk, 2018-03-07:
It occurred to me that *someone* must have done this before:

https://github.com/Transpyler/qturtle

(Haven't tried it or anything...)

waveform80, 2018-03-07:
Afraid I already had a grep through that earlier (same thought occurred :) - it's a re-implementation of turtle (doesn't use the built-in library at all, so not quite the same thing as is suggested above). It might be reasonably complete, but I haven't dug that far into it.

martinohanlon, 2018-03-16:
Legend.... I will have a go later.

MarcScott, 2018-03-21:
That's awesome @ntoll 
Quick question - do you need to set the env variables to use remote pins. Thought you could state the IP directly in the file.
```python
from gpiozero.pins.pigpio import PiGPIOFactory
factory = PiGPIOFactory(host="192.168.1.79")
```

ntoll, 2018-03-21:
@MarcScott I'm working on a way to configure envars in Python 3 mode and it should land later today or tomorrow.

waveform80, 2018-03-21:
> Quick question - do you need to set the env variables to use remote pins. Thought you could state the IP directly in the file.

No, you don't *have* to set the env variables. As you note it's possible to specify hosts manually (which also means one script can control multiple Pis by constructing multiple factories); the env vars simply set the default. If the goal is to keep things as simple as possible for the user's script then it's wise to go the env vars route - but they can always be overridden within the script. For full details see [environment variables](http://gpiozero.readthedocs.io/en/stable/remote_gpio.html#environment-variables) and onwards in the GPIO Zero docs.

bennuttall, 2018-03-21:
Exactly. The end goal is that someone on a PC (with Pi Zero attached) can type code that would work on a Pi, without having to change it, e.g.

```python
from gpiozero import LED

led = LED(2)

led.blink()
```

rather than:

```python
from gpiozero import LED
from gpiozero.pins.pigpio import PiGPIOFactory

pizero = PiGPIOFactory('fe80::1%usb0')

led = LED(2, pin_factory=pizero)

led.blink()
```

Also they wouldn't have to know what IP address to use - this would be the default.