mu-editor/mu #341: Pygame [+Zero] naming consistency?

tjguk, 2018-02-17:
I'm not actually sure which option Daniel Pope prefers (or which the Pygame devs prefer) but we have a mishmash of:

"Pygamezero" (the mode as displayed in the bottom right of the screen)
"Pygame Zero" (the mode as displayed when you *click* that name)
and even the list of modes seems conflicted, with the header using "Pygame Zero / PyGame Mode" and then going on to say that it should be "pygame". [https://mu.readthedocs.io/en/latest/modes.html#pygame-zero-pygame-mode]

Even the pygame website seems unsure with, eg, the announcement of pygame 1.9,3 headed "pygame 1.9.3 released" while two lines further down, the download link has "Pygame 1.9.3 released"

FWIW I feel we should just find one and stick to it regardless. The common denominator seems to be "Pygame" and "Pygame Zero"

(tagged as "trivial")

ntoll, 2018-02-17:
Heh... I'm aware of this simply because it bugs me and bugged me even more as I was writing up the docs and realised how inconsistent everything is.

Ergo, I'll decide by fiat that we call it "Pygame" and "Pygame Zero". Why, they're proper nouns (and should probably be capitalized as a result. This also follows at least one of the many naming conventions used by each project. ;-)

I'll clean up the mess next week.

bennuttall, 2018-02-20:
Fwiw I agree. Similarly, I tend to refer to my GPIO library in copy as "GPIO Zero" but the module name is (as per Python convention) `gpiozero`. I would use the same logic for PyGame and PyGame Zero.

*realises there's no upper case G in "Pygame" but lets it slide...*

illume, 2018-02-21:
pep8 ;)

tldr; Please call it whatever you like (or `pygame`).

I've been trying to make it all lower case on the pygame website 'recently', and asking people to just call it `pygame` after the pygame 1.9.3 release. Perhaps I'll do another pass of the pygame docs and website. There were long discussions on this topic, and I think making it slightly easier on newbies wins in the end. Even though it's slightly abhorrent to marketers and pedants (oh wait... double win).

Case matters in python, and that is weird to people who are used to case not mattering. To me, that was worth pointing out in an introduction to newbies. But it's mainly just a `colorful` touch and a bit of fun. `pip install PyGame` works now, and there are a few newbies who do `import Pygame`.

Not that people will listen, and will write it how they like. Then there's the small matter of having to correct 18 years of writing on the internet with pyGame, PyGame, Pygame, pyGAME, py-game, python-game, Py Game... (not to mention the naming confusion from a certain fruit named company funding a pygame fork, and using pygame in their forks package name. _ahem_). Then there is the small topic of languages other than English and how they do capitals and spaces differently.

And don't we have name spaces in python? And a cheese shop(sorry pypi)? So the 'py' prefix isn't really that important these days.

In the late naughties there was this whole pig naming craze that went down. Then people starting punning about with berries & pies... so now pigame is also a thing. Sorry, PiGame (pronounced pee game?).

- `pygame at the start of a sentence feels weird`.
- Putting `pygame at the start of a sentence feels weird`. However, it's sort-of-fun thinking of ways to remove it from the start. Sort of like a game of poetry.

But yeah, sticking with the pep8 way of spelling it is perhaps easier on newbies.

`pip install oo;python -moo`

ps. mu-editor ?
pps. not sure I wrote all this. I find the topic sort of funny I guess.

ZanderBrown, 2018-08-18:
Did this get cleaned up?

commit 001b397f (2020-12-10): Merge pull request #1163 from dybber/pygame_name_cleanup

Clean up Pygame Zero naming in a few places, fixes #341
commit 7a777bcf (2020-12-10): Clean up Pygame Zero naming in a few places, fixes #341
closed 2020-12-10 by dybber