mu-editor/mu #342: Pygame Zero mode doesn't finish running when the program finishes

tjguk, 2018-02-17:
Switch to Pygame Zero mode
Run some code (even an empty file)
Let it finish (eg Ctrl-Q or your game dies or whatever)

mu is still in "Running" mode and won't offer the "Play" button until the "Stop" button is pressed

ntoll, 2018-02-17:
That's correct. The stdin/out panel at the bottom of the screen should show a message saying the process is finished.

This is so any output from the finished program is still visible to the end user, until they want to explicitly dismiss it by clicking "stop".

Does this make sense? Any way to make it clearer? Happy to have suggestions.

ntoll, 2018-03-21:
This appears to be the correct behaviour (as explained above) so closing.

closed 2018-03-21 by ntoll