mu-editor/mu #343: Pygame Zero mode is offered when Pygame Zero is not installed

tjguk, 2018-02-17:
(Unsure whether this is a bug or by design)

If Pygame Zero (or Pygame) is not installed, the Pygame Zero mode is still offered and the [Play] button is still active; but it does nothing and gives no evident error

I would guess that the intention with modes is that you're supposed to know what you're doing when you go in. I'm running in a dev environment, so I only have the requirements.txt installed; presumably a downloaded binary will always have Pygame Zero?

It would be easy enough to detect the absence of Pygame[Zero] and offer a helpful dialog when [Play] is pressed. Happy to submit a PR if that seems a useful approach.

ntoll, 2018-02-17:
Yeah... on my ToDo list is to only make Pygame Zero mode available if Pygame Zero is installed. It should be a relatively simple variation of something like:

```
PGZ_FLAG = True
try:
    import pgzero
except ImportError:
    PGZ_FLAG = False
```

illume, 2018-02-22:
If you don't want to load the module, and see if it's there...

3.4+ (and probably already imported?)
```
import importlib
importlib.util.find_spec("pgzero") is not None
```

Requires pip (and loading pip)... but works on older pythons.
```
import pip
"pgzero" in [x.project_name for x in pip.get_installed_distributions()]
```

ntoll, 2018-02-22:
This has been fixed in beta.15 release. :-)

ntoll, 2018-02-22:
@illume ahh.. that's an interesting way to do it. You're right, I didn't want to try to import all the things to check it was available, so took a slightly different approach to yours (and doesn't introduce a dependency on pip): https://github.com/mu-editor/mu/blob/master/mu/app.py#L91

closed 2018-02-22 by ntoll