mu-editor/mu #349: mac pgzero

illume, 2018-02-21:
Exhaustive details here. 
https://bitbucket.org/lordmauve/pgzero/issues/63

Is it supposed to be calling `pgzrun`?

ukBaz, 2018-02-21:
I am seeing a similar issue. These are the steps I followed.

I created a venv to install mu:
```linux
eddies-computer:~ barry$ python3 -m venv venvs/mu_beta
eddies-computer:~ barry$ cd venvs/mu_beta/
eddies-computer:mu_beta barry$ ll
total 8
drwxr-xr-x   6 barry  502   204B 21 Feb 20:56 .
drwxr-xr-x   3 barry  502   102B 21 Feb 20:56 ..
drwxr-xr-x  12 barry  502   408B 21 Feb 20:57 bin
drwxr-xr-x   2 barry  502    68B 21 Feb 20:56 include
drwxr-xr-x   3 barry  502   102B 21 Feb 20:56 lib
-rw-r--r--   1 barry  502   114B 21 Feb 20:56 pyvenv.cfg
eddies-computer:mu_beta barry$ source bin/activate
(mu_beta) eddies-computer:mu_beta barry$ pip install mu-editor
Collecting mu-editor
  Downloading mu_editor-1.0.0b15-py3-none-any.whl (1.5MB)
    100% |████████████████████████████████| 1.6MB 401kB/s 
Collecting pyserial==3.4 (from mu-editor)
  Using cached pyserial-3.4-py2.py3-none-any.whl
Collecting PyQtChart==5.10 (from mu-editor)
  Downloading PyQtChart-5.10-5.10.0-cp35.cp36.cp37-abi3-macosx_10_6_intel.whl (771kB)
    100% |████████████████████████████████| 778kB 301kB/s 
Collecting pgzero==1.1 (from mu-editor)
  Downloading pgzero-1.1.tar.gz (1.1MB)
    100% |████████████████████████████████| 1.1MB 472kB/s 
Collecting pycodestyle==2.3.1 (from mu-editor)
  Downloading pycodestyle-2.3.1-py2.py3-none-any.whl (45kB)
    100% |████████████████████████████████| 51kB 570kB/s 
Collecting qtconsole==4.3.1 (from mu-editor)
  Downloading qtconsole-4.3.1-py2.py3-none-any.whl (108kB)
    100% |████████████████████████████████| 112kB 903kB/s 
Collecting pyqt5==5.10 (from mu-editor)
  Downloading PyQt5-5.10-5.10.0-cp35.cp36.cp37-abi3-macosx_10_6_intel.whl (86.1MB)
    100% |████████████████████████████████| 86.1MB 13kB/s 
Collecting qscintilla==2.10.2 (from mu-editor)
  Downloading QScintilla-2.10.2-5.9.3-cp35.cp36.cp37-abi3-macosx_10_6_intel.whl (2.0MB)
    100% |████████████████████████████████| 2.0MB 418kB/s 
Collecting matplotlib==2.1.2 (from mu-editor)
  Downloading matplotlib-2.1.2-cp36-cp36m-macosx_10_6_intel.macosx_10_9_intel.macosx_10_9_x86_64.macosx_10_10_intel.macosx_10_10_x86_64.whl (13.2MB)
    100% |████████████████████████████████| 13.2MB 92kB/s 
Collecting pyflakes==1.6.0 (from mu-editor)
  Downloading pyflakes-1.6.0-py2.py3-none-any.whl (227kB)
    100% |████████████████████████████████| 235kB 665kB/s 
Collecting pygame>=1.9 (from pgzero==1.1->mu-editor)
  Downloading pygame-1.9.3-cp36-cp36m-macosx_10_9_intel.whl (4.8MB)
    100% |████████████████████████████████| 4.8MB 212kB/s 
Collecting ipython-genutils (from qtconsole==4.3.1->mu-editor)
  Downloading ipython_genutils-0.2.0-py2.py3-none-any.whl
Collecting jupyter-client>=4.1 (from qtconsole==4.3.1->mu-editor)
  Downloading jupyter_client-5.2.2-py2.py3-none-any.whl (88kB)
    100% |████████████████████████████████| 92kB 851kB/s 
Collecting ipykernel>=4.1 (from qtconsole==4.3.1->mu-editor)
  Downloading ipykernel-4.8.2-py3-none-any.whl (108kB)
    100% |████████████████████████████████| 112kB 883kB/s 
Collecting pygments (from qtconsole==4.3.1->mu-editor)
  Downloading Pygments-2.2.0-py2.py3-none-any.whl (841kB)
    100% |████████████████████████████████| 849kB 397kB/s 
Collecting traitlets (from qtconsole==4.3.1->mu-editor)
  Downloading traitlets-4.3.2-py2.py3-none-any.whl (74kB)
    100% |████████████████████████████████| 81kB 639kB/s 
Collecting jupyter-core (from qtconsole==4.3.1->mu-editor)
  Downloading jupyter_core-4.4.0-py2.py3-none-any.whl (126kB)
    100% |████████████████████████████████| 133kB 701kB/s 
Collecting sip<4.20,>=4.19.4 (from pyqt5==5.10->mu-editor)
  Downloading sip-4.19.7-cp36-cp36m-macosx_10_6_intel.whl (51kB)
    100% |████████████████████████████████| 51kB 1.2MB/s 
Collecting six>=1.10 (from matplotlib==2.1.2->mu-editor)
  Downloading six-1.11.0-py2.py3-none-any.whl
[... 97 lines cut ...]

def draw():
    screen.fill((128, 0, 0))

2018-02-21 21:04:57,401 - mu.interface.panes:602(start_process) INFO: Running script: /Users/barry/mu_code/test_pygame.py
2018-02-21 21:04:57,402 - mu.interface.panes:607(start_process) INFO: Command args: []
2018-02-21 21:04:57,402 - mu.interface.panes:613(start_process) INFO: Working directory: /Users/barry/mu_code
2018-02-21 21:04:57,402 - mu.interface.panes:618(start_process) INFO: Python path: ['', '/Library/Frameworks/Python.framework/Versions/3.6/lib/python36.zip', '/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6', '/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/lib-dynload', '/Users/barry/venvs/mu_beta/lib/python3.6/site-packages', '/Users/barry/venvs/mu_beta/lib/python3.6/site-packages/IPython/extensions']
2018-02-21 21:05:36,687 - mu.logic:618(show_help) INFO: Showing help.
2018-02-21 21:23:05,268 - mu.modes.pygamezero:129(stop_game) DEBUG: Stopping script.
2018-02-21 21:23:06,955 - mu.modes.pygamezero:118(run_game) DEBUG: 
WIDTH = 300
HEIGHT = 300

def draw():
    screen.fill((128, 0, 0))

2018-02-21 21:23:06,968 - mu.interface.panes:602(start_process) INFO: Running script: /Users/barry/mu_code/test_pygame.py
2018-02-21 21:23:06,968 - mu.interface.panes:607(start_process) INFO: Command args: []
2018-02-21 21:23:06,968 - mu.interface.panes:613(start_process) INFO: Working directory: /Users/barry/mu_code
2018-02-21 21:23:06,968 - mu.interface.panes:618(start_process) INFO: Python path: ['', '/Library/Frameworks/Python.framework/Versions/3.6/lib/python36.zip', '/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6', '/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/lib-dynload', '/Users/barry/venvs/mu_beta/lib/python3.6/site-packages', '/Users/barry/venvs/mu_beta/lib/python3.6/site-packages/IPython/extensions']
```

These are the details of the system I am running on:
[image]

ukBaz, 2018-02-21:
I was working off the tweet from earlier today. I have now gone and read the _Developer Setup_ instructions for mu and noticed the requirement for the `--system-site-packages` flag.

If I now install with:
```
eddies-computer:~ barry$ python3 -m venv --system-site-packages venvs/mu_beta2
eddies-computer:~ barry$ cd venvs/mu_beta2/
eddies-computer:mu_beta2 barry$ source bin/activate
(mu_beta2) eddies-computer:mu_beta2 barry$ pip3 install mu-editor
```
All is working fine for me.

carlosperate, 2018-02-21:
That's odd, you shouldn't need to use the `--system-site-packages` flag, as all the packages should have been installed inside the virtual environment. Usually that flag is necessary when some package are added via installer, like older versions of PyQt, or PyQt in Raspbian as it is installed via `apt-get`.

Could you run the virtualenv without the flag and see what this line throws? (Ideally with the test program in a `myfile.py`)
```
python -m pgzero myfile.py
```

ukBaz, 2018-02-22:
Run as requested in the virtualenv that was created without the flag

```
(mu_beta) eddies-computer:mu_code barry$ python -m pgzero myfile.py 
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/Users/barry/venvs/mu_beta/lib/python3.6/site-packages/pgzero/__main__.py", line 3, in <module>
    main()
  File "/Users/barry/venvs/mu_beta/lib/python3.6/site-packages/pgzero/runner.py", line 62, in main
    _substitute_full_framework_python()
  File "/Users/barry/venvs/mu_beta/lib/python3.6/site-packages/pgzero/runner.py", line 55, in _substitute_full_framework_python
    os.execv(framework_python, ['python', '-m', 'pgzero'] + sys.argv[1:])
FileNotFoundError: [Errno 2] No such file or directory
```

ntoll, 2018-05-03:
I cannot reproduce... so closing. :-/

Please feel free to re-open if you try the steps I describe above, but the problem persists.

Thanks!

closed 2018-05-03 by ntoll